MarlinFirmware/Marlin #3026: E-Motor Reversed After Cancel Print Job Command

sarf2k4, 2016-02-24:
Hi,

I had problem where the e-motor reversed on its own after I hit the cancel print job either through octoprint or lcd panel, I'm not sure if anyone having the same issue or not, I'm using slic3r with marlin rc3 currently and the last time I got this problem was about 4 days ago when I tried to find my z offset on auto bed leveling feature.

At first I thought it was my start of g-code to prime the hot end but it is actually a positive value, when cancelling the print job, it starts to reverse the e-motor and furthermore, the reversed travel length seems to be accumulate, e.g, extrude 5mm at g-code start print, cancel because layer too high, start print again with same script, the motor starts to reverse its direction, as more and more canceled job issued, the 5mm extrude will be accumulated to be reversed, 5 times of print job cancelled would mean reversing the e-motor travel by 25mm

I'm not sure if it's only me or anyone having the same problem though

whosawhatsis, 2016-02-24:
E values are generally absolute, which means that if you want to extrude 5mm after you've already extruded 10mm, you have to tell the extruder to go to 15mm. It sounds like your start code doesn't have a "G92 E0" line to reset the current position to 0mm before it primes. Add that line and it should fix the problem, unless there's something unusual going on.

sarf2k4, 2016-02-24:
I didn't have that line, the strange reverse behavior just hit on me after about 3 times of print job cancel, but for first-third print it does move to the right direction (gcode without the g92 e0 line)

sarf2k4, 2016-02-27:
Thank you, that really helped

closed 2016-02-27 by sarf2k4