MarlinFirmware/Marlin #3029: Problem uploading firmware

slarrabee99, 2016-02-25:
Hi,  I finished building my 3d printer yesterday and im having issues with getting it working.  I went through the tutorial to upload marlin to my ardunio and have yet to get it sucessfully working.  So far I can only read temperature of the extruder (using pronterface) but have not been able to get the printer to move or heat up at all.  please help me i would very very appreciate it  Error from arduino below v

In file included from sketch\Marlin.h:23:0,

                 from sketch\thermistortables.h:4,

                 from sketch\Configuration.h:811,

                 from C:\Users\Steve\Desktop\gMax 1-5 PLUS SINGLE firmware\Marlin\Marlin.ino:33:

sketch\pins.h:667:0: warning: "FAN_PIN" redefined [enabled by default]

     #define FAN_PIN            8

 ^

sketch\pins.h:663:0: note: this is the location of the previous definition

     #define FAN_PIN            4 // IO pin. Buffer needed

 ^

Blue-Marlin, 2016-02-25:
Our current pin.h files have not so large line numbers.
You are using a at least one year old version of MarlinFirmware/Marlin or a completely different fork.
To help you, we need a __link__ to your actual (throwing this error) Configuration.h Configuration_adv.h and pins.h.
We'd be happy if you could change to one of our recent versions. (RCBugFix)

paulusjacobus, 2016-02-25:
also a current first mistake is that people think the motors directly run
when you push the arrow buttons in Pronterface. You need to home the
printer first so it knows its coordinates. Without that it refuses to move.
In pronterface you can push the home buttons or issue a home command "G28".

closed 2016-05-04 by jbrazio