MarlinFirmware/Marlin #3032: Watchdog function on GT2560

valdemar4eg, 2016-02-25:
Dear all,
Could somebody, please, explain in more details how does WATCHDOG works on geeetech gt2560 (mix on Andruino Mega 2560)? My printer quite often crushes in the middle of the print with all heaters on. So I would like to use all safety features available on Marlin.

These comments in the code are not clear to me.
#ifdef USE_WATCHDOG
// If you have a watchdog reboot in an ArduinoMega2560 then the device will hang forever, as a watchdog reset will leave the watchdog on.
// The "WATCHDOG_RESET_MANUAL" goes around this by not using the hardware reset.
// However, THIS FEATURE IS UNSAFE!, as it will only work if interrupts are disabled. And the code could hang in an interrupt routine with interrupts disabled.
//#define WATCHDOG_RESET_MANUAL
#endif

Thank you in advance

thinkyhead, 2016-03-11:
I think the comments try to explain too much, and aren't very helpful.

I suggest using `USE_WATCHDOG` as a general safety feature, so Marlin can reset the machine if the Arduino hangs for any reason. `WATCHDOG_RESET_MANUAL` is an alternative method to catch a hang condition, that won't reset the machine but just leave it frozen until manually reset.

The AVR manuals state that the built-in watchdog timer is not necessarily reset when you reboot the board. For this reason, under some conditions the Arduino could keep rebooting repeatedly. This really depends on the code that the Arduino is running. Marlin calls `wdt_reset` frequently, so there's no reason for Marlin to keep rebooting due to this feature unless something is very wrong.

The alternative method, `WATCHDOG_RESET_MANUAL`, uses a custom interrupt routine to catch a hung condition. As long as Marlin keeps calling `wdt_reset` the interrupt will never fire. But if anything goes wrong, after 4 seconds Marlin will try to invoke `kill` and turn off the heaters.

thinkyhead, 2016-03-11:
P.S., I suggest you try `RCBugFix` to get the most reliable performance right now. Perhaps you can try doing a long print and see how it does. The Arduino shouldn't hang at all, of course, so that's what we want to mainly avoid. Check for loose connections, as always. If the board does hang, hopefully in the latest code it can at least turn off the heaters before it gives up the ghost.

Blue-Marlin, 2016-03-11:
> The alternative method, WATCHDOG_RESET_MANUAL, uses a custom interrupt routine to catch a hung condition. As long as Marlin keeps calling wdt_reset the interrupt will never fire. But if anything goes wrong, after 4 seconds Marlin will try to invoke kill and turn off the heaters.

If the hung is not in an interrupt. What is the problem with WATCHDOG_RESET_MANUAL.
Never. Never! Never define WATCHDOG_RESET_MANUAL!

closed 2016-05-04 by jbrazio