MarlinFirmware/Marlin #3033: can't get the correct file name of SD card

WillSmithInChina, 2016-02-26:
for example, the correct file name of SD card is "quick-test.gcode" ,but i get the file name is "QUICK-~1.GCO" , so what it happened , do you meet the same problem ?

WillSmithInChina, 2016-02-26:
the answer is to update the Marlin fireware and use the command of "M33"

for example:
     *   M33 miscel~1/armchair/armcha~1.gco
     *
     * Output:
     *   /Miscellaneous/Armchair/Armchair.gcode
     */

jbrazio, 2016-02-27:
Could this be related with #3039 ?

HedronUser, 2016-02-28:
When I update the firmware I lose advanced delta geometry control....

thinkyhead, 2016-02-28:
> the answer is to update the Marlin fireware and use the command of "M33"

Yes, long filename support was added through a new code `M33`, which I implemented (yes, it's my fault) as a way to request a long name if you needed one. I assumed host software would begin to use it for fancy file listings of SD cards, but so far none has.

If you're going to get one of the new release candidates, be sure to get `RCBugFix` so you can avoid some filename bugs that have recently been patched.

> When I update the firmware I lose advanced delta geometry control

@HedronUser Have you opened an issue about that subject? Delta features are a little in flux right now also.

closed 2016-02-28 by thinkyhead

WillSmithInChina, 2016-02-29:
when i enable "LONG_FILENAME_HOST_SUPPORT" and  enter "M33 QUICK-~1.GCO" , but i get the answer is "/???" ; is there anyone to help me ?

thinkyhead, 2016-03-01:
@WillSmithInChina When you use `M33` I believe you must use lowercase for the file name. `M33 quick-~1.gco` This will be patched in the future.

WillSmithInChina, 2016-03-02:
@thinkyhead  thanks, it solved my problem as the firmware neees to update