MarlinFirmware/Marlin #3034: 1.1 rc3 raise after probing not working

jfpion, 2016-02-26:
hello 
prusa 2
x/y endstop switch 
z probe manualy activated and retracted switch
autobel leveling activated, autogrid
no z_probe_end_script enabled

the start gcode:
M190 S110 ; set bed temperature
M109 S270 ; set temperature
G90 ; Passage coordonnees absolues
G1 Z30 F2000
;pause
G1 x10 Y10
G28 X0 Y0 ; Home X Y
G1 X10 Y10
G28 Z0 ; home Z
G29
G1 Z30 F4000
;@pause

after the g29 probing the head go back to x10 y10  whithout going up as it is supposed to do as per:
 #define Z_RAISE_AFTER_PROBING 15, and dragging my poor switch all across the bed

there is a work around using the z_probe_end_script but any way it's look like there is a small bug here.

and why the head is going to the point it was before the g29 ? I think it's useless as you can do what you want in your start gcode and any way having some movement you're not able to avoid is confusing (specialy when it is not documented.)

Blue-Marlin, 2016-02-26:
Z_RAISE_AFTER_PROBING is only applied when the system knowa you have a probe.
We don't have a probe type 'manual'. So the system does not know you have a probe and what to do with it.
Z_PROBE_END_SCRIPT seems not to be only a workaround but duty if your manual probe needs an extra handling.

With G29 I have never observed returning to the point before. Please give us a link to your Configuration.h .

jfpion, 2016-02-26:
I thought as I ask for an autolevel it was obvious there was as probe and if not otherwise specified it must be a manual one by default.
so nevermind the probe système when it is spécified Z_RAISE_AFTER_PROBING it must raise of the value nevermind of the kind of probe.
if i dont want a raise, I have only to specify 0

that what seems logical to me 

and i'm uploading a zip file withe the video of my test gcode, the gcode and the configuration.h I will post the link asap

thank you

jfpion, 2016-02-29:
as promised

[http://jean.francois.pion.free.fr/pb autoleveling.zip](url)
you can see the head rise after the last probe, (because of the z_probe_endscript ) and after that going to x10 y10 quite slowly before raising againg as per the gcode

thank you

jbrazio, 2016-02-29:
I do confirm this behavior, after probing the Z axis is not lift by `Z_RAISE_AFTER_PROBING` has it was supposed to be. I propose PR #3057 to fix it.

jfpion, 2016-02-29:
and there is the pb of the head going back to x10 y10 (the place it was when the g29 was called) as the video show it 

thank you

Blue-Marlin, 2016-02-29:
@jfpion 
```
#define Z_PROBE_END_SCRIPT "G1 Z10 F1000\nG1 X10 Y10"
```
If you are telling it to go there don't wonder it does.
Or maybe it's the negative gravity in your workshop. :-)

jfpion, 2016-03-01:
hoops...
an another "between chair and keyboard" bug ! silly me !

thank you

thinkyhead, 2016-03-03:
#3057 is pretty comprehensive. Thought it would be best to get the code into `RCBugFix` now, let people encounter it in `RC4`, and then we'll know how truly robust it is!

closed 2016-03-03 by thinkyhead