MarlinFirmware/Marlin #3035: need to add comment in the code 1.1 rc3

jfpion, 2016-02-26:
Hello,
I know it is obvious for the advanced user but i banged my head for a long time before finding the solution in the issues answer.

I suggest to add " // if you are not using max endstop you MUST uncomment this line " just before : 
#define DISABLE_MAX_ENDSTOPS

because without the autolevel it works fine, so no need to bother about it, when you try to add the autoleving feature nothing works !

and I suggest this line should be uncomented by default so the only one who comment it will be the one using z max endstop, and then avoiding head scratching time in front of the computer

thank you

Blue-Marlin, 2016-02-26:
That not a good idea because there are lots off machines out there with x/y-max endstops.
All their owner would be a bit sad if we'd crash their machines by default.

jfpion, 2016-02-26:
OK then, so adding then warning in the code seems the better solution

closed 2016-05-04 by jbrazio