MarlinFirmware/Marlin #3036: panic button ? is there a way to have one ?

jfpion, 2016-02-26:
hello, 

During my try of the autolevel, i got multiple occasion to have to switch off the printer to avoid to crash the head on the bed .

I think it could be useful to have a panic button stopping all the movement and ending the execution of the gcode immediatly not waiting for the buffer to empty. just not a way to stop a print to be able to resume after, but a kill switch to abort to preserve the printer. (sometime you have other ting connected to the power supply so you can't simply switch off the main power

may be it is already existing but i have'nt found it anywhere

thank you

Blue-Marlin, 2016-02-26:
For many displays/panels __the__ extra button is implemented as kill-switch.
Almost all boars have a reset button. Just solder a extra switch in parallel.
I have big red panic buttons to switch off 12V (simple hardware).

jfpion, 2016-02-26:
the pb with the reset button is it is a reset, and when the bord reboot it get other gcode and go on...
switching off the power need a switch with a lot current capacity and I will like to avoid invisible bad contacts , a sure recipe for headhache in case of pb 
and I hanv' nt a panel, 

is the M112 gcode buffered ? if not it will do otherwise it is useless

thank you

brainscan, 2016-02-26:
I've always just pressed reset on the board without any problems.

Blue-Marlin, 2016-02-26:
Look for KILL_PIN in your pin file. Define one outside a panel/display. Connect a closing button switch to GND. Be happy.

It's not obvious but i mend switching off the primary side of the 12V power supply.

M112 is applied as soon it leaves the RX buffer. If that is blocked, for example by a M190, this can take a while.

jfpion, 2016-02-29:
OK  thank you

thinkyhead, 2016-03-02:
@Blue-Marlin It's "meant" not "mend." :grinning:

Blue-Marlin, 2016-03-02:
Ups.
Learning Marlin is by far easier than learning correct English for me. ;-)

closed 2016-05-04 by jbrazio