MarlinFirmware/Marlin #3038: Who assigns board and thermistor IDs ?

jbrazio, 2016-02-26:
This is not a issue but actually it is a question, what is the rule to assign a new board ID so a new pinout file can be created ? The same question applies to a new thermistor compensation table ID.
If there is not a rule defined, who is the "authority" to assign the IDs ?
Or can I just select the next "free" number ?

Thanks
-jbo

Blue-Marlin, 2016-02-26:
For temperature measurement devices the rule is to use analogue devices with a positive number and others with negative. PTs with high numbers. For a new therminstor continuing with 14 is not a bad idea. A alternate table for a already existent or very similar thermistor could have the old number postfixed by 1 (2,3,...)

The boards used to be grouped by 'family's'. But as this is already a bit messed up the numbering is a bit random. 

There is no authority to assign the numbers.
thermistortables.h is authoritative for the thermistors (temperature measurement devices).
boards.h is authoritative for the boards.

If you are polite you add your changes to the Configuration.h's or the Makefile.

jbrazio, 2016-02-27:
Great, thanks.

closed 2016-02-27 by jbrazio