MarlinFirmware/Marlin #3039: SD card support broken in RCBugFix (Fine in RC3). Marlin_main.cpp bug?

filipgoc, 2016-02-26:
Hi, 
I have flashed the latest RCBugFix (Feb 24th) and I cannot print from SD card anymore. The card gets recognized, I can see and browse the files, but selecting a file does not instruct the printer to do anything. The temperature stays at zero; nothing happens. 

It seems to be something in Main_main.cpp, because
even with all my configuration.h tweaks, RCBugFix works fine if I paste in the old RC3 Marlin_main.cpp

jbrazio, 2016-02-27:
I do confirm, I'm having the same issue with RC Bugfix, it seems to be related with the 8.3 file names per the error in the serial line after selecting the file from the LCD menu. I do have the long file names option enabled in advance config.

jbrazio, 2016-02-27:
The solution has been found by @CONSULitAS and fixed with #3027, in a nutshell the g-code command (M23) was being feed into current_command_args thus producing an invalid file name.

I would suggest to close this issue.

HedronUser, 2016-02-28:
Hmm, My marlin_main.cpp does not have current_command_args at all. @jbrazio

jbrazio, 2016-02-28:
Are you sure that it the RCBugFix branch you checked out ?
I would do a fresh clone of Marlin and checkout to the RCBugFix branch before doing any changes.

HedronUser, 2016-02-28:
I'm using a deprecated version of Marlin that has more manual delta geometry adjustments. I'll delete my comment.

thinkyhead, 2016-02-28:
RCBugFix now has all the GCode parser patches that a GCode parser patcher can patch. It should now get arguments correctly, including the all-important filename argument for SD printing. If you can confirm then I will mark RCBugFix and close this issue.

WillSmithInChina, 2016-03-01:
@jbrazio      whether i should update firmware to fix the problem that  the g-code command (M23) was being feed into current_command_args thus producing an invalid file name ?

jbrazio, 2016-03-01:
RCBugFix branch has all the required patches to fix the SD filename issue so I would say start fresh..
```
git clone https://github.com/MarlinFirmware/Marlin.git marlin-fresh
cd marlin-fresh
git branch RCBugFix
```
Copy your configuration files (`Configuration.h` and `Configuration_adv.h`) to the `marlin-fresh/Marlin` folder and open the `Marlin.ino` file using the Arduino IDE. From the menu Tools > Board select `Arduino/Genuino Mega or Mega 2560`, Tools > Processor and select `ATmega2560 (Mega 2560)`.

Make sure that on the `Configuration_adv.h` you have the option `LONG_FILENAME_HOST_SUPPORT` enabled.

Compile and upload to your board.

WillSmithInChina, 2016-03-02:
@jbrazio  it solved my problem, thanks for you regards

closed 2016-03-02 by thinkyhead