MarlinFirmware/Marlin #3041: On LCD panel "Bed" jumps to "Flow"

gddeen, 2016-02-26:
- I follow "Tune" "Bed" and "Flow" appears. code in ultralcd.cpp
- I follow "Tune" "fan speed" and "Flow" appears. code in ultralcd.cpp

Is this recreatable? 

Just a nit: 
I'm running MKS_BASE_BOARD  V1.3 (not 1.1 or 1.2, which is a RAMPS14... not ramps13 based board

in pins_MKS_BASE.h I added at the bottom
#undef SERVO0_PIN
#define SERVO0_PIN 11 // GDD Labeled D11 on board
...
It includes pins_RAMPS_13.h which sets it to pin 7. So, I set it back. Also, someone added "pin 7"
to the heater (Thanks) but left it attached as the servo0.

This board has 4 servo spots, A11, A12, D11, D12

AnHardt, 2016-02-27:
Do you really think another issue for your Flow problem is needed? (https://github.com/MarlinFirmware/Marlin/issues?utf8=%E2%9C%93&q=is%3Aissue+Flow+)

Obviously You have a MKS board to test and the knowledge what is needed in the pin file. How about a PR?
It would be better to define IS_RAMPS_14 before including pins_RAMPS_13_EFB.h. That would solve the SERVO0 problem. (Discovery of the difference at SERVO0 between 1.3 and 1.4 is relative new)

gddeen, 2016-02-27:
No. I do not think pin needs an issue. So should I go look to fix the menu item? Or the january guy is going too?

Sorry, I am not experienced with github.

gddeen, 2016-02-27:
It looks like the macro is skipping the substitution because not everything is "parameterized like a function." Essentially, it doesn't like an argument on the invocation of menu_edit_item? Macros can get kinda difficult to follow quickly..

Was this item already fixed and are a part of RC3?

I suggested the SERVO0 fix because this would mean people with MKS boards (HICTop Prusa i3) would see rather quickly how to add a servo for auto bed leveling. Without the explanation "this label" servo0_PIN has "11" and is labeled on your board as "D11," most would give up what is a trivial addition to their 3d printer.

thinkyhead, 2016-02-28:
The issue of the menu clicks taking you to the wrong place has been patched in `RCBugFix` along with some other errors that slipped into RC3. I recommend using `RCBugFix` until `RC4` is ready, which will patch some other things. You can check the Pull Requests list to see if any of the outstanding bugs may affect you.

thinkyhead, 2016-02-28:
@AnHardt Do you think that `IS_RAMPS_14` needs to be set in `pins_MKS_BASE.h`?

closed 2016-02-28 by thinkyhead

AnHardt, 2016-02-28:
Yes.
The difference of 1.3 and 1.4 was discovered recently.
The difference is only at SERVO0.
pins_RAMPS_13.h used to use the pinning of 1.4
All the pin files for other boards assumed 1.3 and 1.4 to be identical.

-> We have to define `IS_RAMPS_14` in every pin file including pins_RAMPS_13.h.

Alternatively we could make a real pins_RAMPS_14.h
```
#define IS_RAMPS_14
#include pins_RAMPS_13.h
```
and include that. (Or the other way around - rename pins_RAMPS_13.h to 1.4 - change condition to IS_RAMPS_13 - Make a new pins_RAMPS_13.h including 1.4 (1.4 is the general case and 1.3 the exception then - like in the real world))

thinkyhead, 2016-02-29:
A `pins_RAMPS_14.h` file seems like the clearest approach. If 1.3 is now the exception, then sure, I can switch the flag to `IS_RAMPS_13` instead.

AnHardt, 2016-03-01:
As far i can see all is well but the Travis check. That still tries to find `BOARD_RAMPS_13_EEB` with sed in the Configuration.h and fails.

thinkyhead, 2016-03-01:
Ah yes, I always forget to update the `.travis.yml` file.