MarlinFirmware/Marlin #3046: M145 format in initial output is wrong

nebbian, 2016-02-27:
The M145 command (Set the heatup state for Ultimaker panel) when shown in the initial output of the serial console is wrong.  It currently shows:
echo:  M145 M0 H180 B65 F255
echo:  M145 M1 H240 B105 F255

This should be:
echo:  M145 S0 H180 B65 F255
echo:  M145 S1 H240 B105 F255

This is going by the code that handles the M145 command, which states:
`  /**
   * M145: Set the heatup state for a material in the LCD menu
   *   S<material> (0=PLA, 1=ABS)
   *   H<hotend temp>
   *   B<bed temp>
   *   F<fan speed>
   */
`

thinkyhead, 2016-02-28:
Thanks! I've created patches, so this will be fixed up in the next release candidate.

CONSULitAS, 2016-03-01:
btw:
Reprap Wiki (http://reprap.org/wiki/Gcode#M144:_Stand_By_Your_Bed) doesn't know M145.

Hmmm. :disappointed:

thinkyhead, 2016-03-08:
`M145` is a Marlin extension that I added to accompany some items written to EEPROM. So it's not yet finalized or formalized. We might replace the parameter letters, since it's not very GCode-like to use `H` and `B`, while `F` is usually reserved for feedrate. There's nothing especially bad about using unique letter parameters in a GCode, but some sticklers don't like it.

closed 2016-03-08 by thinkyhead