MarlinFirmware/Marlin #3048: E0 and E1 different FR

hexxter, 2016-02-27:
Fist HW: RAMPS 1.4 / DRV8825
Board settings EEB

The E0 has the right speed from config but E1 drive too fast.
If i change the HW (E0 <> E1) the Problem is only on E1.
If i change the PINs in Marlin  E0 is after this change too fast and E1 is good.

Blue-Marlin, 2016-02-27:
If your extruders need different steps/unit, change the tool (T0/T1) and set the needed steps/unit explicitly with M92.

hexxter, 2016-02-27:
no my tool need the same steps/unit. this is the problem they are different.

Blue-Marlin, 2016-02-27:
Are the micro step jumpers set identically?

hexxter, 2016-02-28:
yes the HW jumper/DRV8825/cable/motors all the same

hexxter, 2016-02-28:
here my config.

[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/149860/Configuration.h.zip)

Blue-Marlin, 2016-02-28:
If not all jumpers are set, one of the open could have a short.

Blue-Marlin, 2016-02-28:
```
#define DEFAULT_AXIS_STEPS_PER_UNIT   {172,172,1600,164,164}
#define DEFAULT_MAX_FEEDRATE          {300, 300, 5, 25, 25}
#define DEFAULT_MAX_ACCELERATION      {3000,3000,100,1000,1000} 
```
5? Should not harm (but maybe), but does not help anyway.

hexxter, 2016-02-29:
yes i know its was only for my documentation ;)

thinkyhead, 2016-03-02:
> `#define DEFAULT_AXIS_STEPS_PER_UNIT   {172,172,1600,164,164}`

Yeah, that won't work. Marlin doesn't currently support differing E steppers. They all have to be identical. Making this change is not trivial. So we have to leave this as a feature request for now and it will not be in Marlin until version 1.2.0 and 1.1.1 at the earliest.

closed 2016-05-04 by jbrazio