MarlinFirmware/Marlin #3054: arbitrary reset the printer when printing more than 1 hour

AnatoliyKube, 2016-02-28:
print the SD card:not not connected to PC
If printing takes longer than 50 minutes,printer stops, and resets
if you set the temperature bed  end nuzzel  with LSD (to warm up  before printing ) -several time occurred printer reset

AnatoliyKube, 2016-02-29:
guys, check the firmware printersbrasuetsya 52-59 minutes and when printing, but the item is designed for 2 hours

AnatoliyKube, 2016-02-29:
He returned to the firmware   https://github.com/ErikZalm/Marlin - no such problems
all the RC version has a problem

brainscan, 2016-02-29:
Sorry but I can't make any sense of that. None.

AnatoliyKube, 2016-02-29:
all versions of RC has a problem
The printer prints 52-59 minutes , after - STOPS END RESETS
(version 1.1.0-rc3 ) , very unstable work, reset occur spontaneously ,

jbrazio, 2016-02-29:
Let me try to make sense of the comments..
1) Your printer resets itself when printing after 50m into the job
2) If you set pre-heat for nozzle and bed somethings it resets itself
3) This only occurs with Marlin RC version(s)

I have a couple of questions..
1) What is your hotend heater cartridge power ?
2) What is your hotbed power ?
3) What is your PSU power ?

I have the feeling you may be running close to the PSU power ratting..

brainscan, 2016-02-29:
I use rc3 and have never had this problem. Can you give more idea about your particular set up as it may be unique to you.

AnatoliyKube, 2016-02-29:
I have two printers
Power Supplies 12.3v 20 A (peak 30 А ), 5V 1 A ,specialized,stabilized voltage, air blower(fen)
bed - 1.8 OM - 6,8 A (peak)
extruder heater 4 om - 3 A
engines - 1,2 А *4 =5A
15 A peak
arduino tnd lcd -separately 5V
on the firmware version https://github.com/ErikZalm/Marlin~~pobj  -There are 10 hours a day - everything is fine
on the firmware version RC 1.1.0 - RC3 -malfunction even without heating (periodically),
printing 50-60 minutes-simply reset the printer

jbrazio, 2016-02-29:
Do you see any messages in the serial console prior to the board reset ?

AnatoliyKube, 2016-02-29:
the printer is not connected to PC , in LCD - ready , x ---- y--- z---

AnatoliyKube, 2016-02-29:
again I remind -on the firmware version https://github.com/ErikZalm/Marlin~~pobj -everything is fine

jbrazio, 2016-02-29:
Could you connect the printer to the PC just to have the debug log available ? You should still start the print job using the LCD/SD card but having a serial terminal open from the PC to store the log file.
The error you're having could be related with #2963 as @thinkyhead pointed out, the latest RC branch there are some changes related with thermal stuff.

Let's have the logs first.

AnatoliyKube, 2016-02-29:
1 -3 miin  start - error: Heating failed ,system stopped heatr id  :0

AnatoliyKube, 2016-02-29:
2 - Print time: 1 hour 06 minutes - - error: Heating failed ,system stopped heatr id :0

Blue-Marlin, 2016-02-29:
Relax
```
  /**
   * Whenever an M104 or M109 increases the target temperature the firmware will wait for the
   * WATCH_TEMP_PERIOD to expire, and if the temperature hasn't increased by WATCH_TEMP_INCREASE
   * degrees, the machine is halted, requiring a hard reset. This test restarts with any M104/M109,
   * but only if the current temperature is far enough below the target for a reliable test.
   */
  #define WATCH_TEMP_PERIOD 30                // Seconds
  #define WATCH_TEMP_INCREASE 4               // Degrees Celsius```

Blue-Marlin, 2016-02-29:
Maybe its a good idea to protect your nozzle from the fans.

AnatoliyKube, 2016-03-01:
WHERE TO FIND THIS CODE ?

  #define WATCH_TEMP_PERIOD 30 // Секунды
  #define WATCH_TEMP_INCREASE 4 // Градусы Celsius```

jbrazio, 2016-03-01:
`Configuration_adv.h`

thinkyhead, 2016-03-02:
> WHERE TO FIND THIS CODE ?

We have a search box! :grinning:

thinkyhead, 2016-03-02:
@AnatoliyKube Have you tested yet with `RCBugFix`? We've made a lot of patches since RC3, some of which may have an effect on your issue.

AnatoliyKube, 2016-03-02:
If you have changed RCBugFix for the last 48 hours - NO
Today again there was a failure (full stop) when printing on 42 minutes, the item is flawed

question is not on- when driving range (radius of less than 5 mm) pull is enough going on, the camera goes in spurts

thinkyhead, 2016-03-03:
> If you have changed RCBugFix for the last 48 hours - NO

Excellent. This will help narrow it.

> question is not on- when driving range (radius of less than 5 mm) pull is enough going on, the camera goes in spurts

This makes little sense. You should simply post in your native language and we will translate it ourselves.

In the meantime, through some careful questioning and experimentation we should be able to track down the problem.

- First question: Have you run the `M303` tuning procedure to calibrate your nozzle heater?
- Have you checked your thermistor to make sure it's reading the correct temperature? That's actually quite simple. When the printer is first turned on, the temperature should match room temperature.

AnatoliyKube, 2016-03-03:
термистор работает в норме (раннее устанавливал прошивку
https://github.com/ErikZalm/Marlin~~pobj  -всё работало без проблем)
м303 e0 s230 c8 - проверял , но каждый раз выходили разные значения (11
+-20%, 0,56 +-20% 54+-20%)
пока установил те что идут по умолчанию (22 ........), разницы в работе не
увидел,
при нагреве (240 гр) температура доходит до 230 и плавно увеличивается , не
больше 242 гр но в низ 235, хотя при печати колебания температуры 238-241 гр
 произвольный сброс может возникнуть в любое время - нагрев принтера ,
простой принтера ,печать , в начале я это заметил при печати детали (время
более 1 часа ) - 50-59 минут , принтер просто сбрасывался  (состояние как
при включении) естественно деталь  пришлось выкинуть , потом я это заметил
при нагревании экструдера и стола - это происходило не однократно ,
блок питания 12,3 V  20A  (30A в пике  ) , 5 v 1 A , с воздушным
охлаждением , ещё раз напоминаю что на прошивке
https://github.com/ErikZalm/Marlin~~pobj  -всё работало без проблем , с
таким блоком питания (есть второй принтер )

AnatoliyKube, 2016-03-03:
сейчас заменю блок питания 12 V29,9A с вентилятором

jbrazio, 2016-03-03:
Anyone knows Russian ?

AnatoliyKube, 2016-03-04:
replaced by a more powerful power supply
printed 2 days
print time of 1.2 hours on the item
Today, when you print the item (1.5 hours) for 5 minutes full stop, not pninter riagiruet on the encoder, the inscription on LСD: THEREMAL  RUNAVAY

Blue-Marlin, 2016-03-04:
Against THEREMAL RUNAVAY tune

```
/**
 * Thermal Protection parameters
 */
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius
```

```
/**
 * Thermal Runaway Protection protects your printer from damage and fire if a
 * thermistor falls out or temperature sensors fail in any way.
 *
 * The issue: If a thermistor falls out or a temperature sensor fails,
 * Marlin can no longer sense the actual temperature. Since a disconnected
 * thermistor reads as a low temperature, the firmware will keep the heater on.
 *
 * The solution: Once the temperature reaches the target, start observing.
 * If the temperature stays too far below the target (hysteresis) for too long,
 * the firmware will halt as a safety precaution.
 */

#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders
#define THERMAL_PROTECTION_BED     // Enable thermal protection for the heated bed
```

AnatoliyKube, 2016-03-04:
Yes, in my firmware so exposed 
/**
 * Thermal Protection parameters
 */
#if ENABLED(THERMAL_PROTECTION_HOTENDS)
 #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius

and
#define THERMAL_PROTECTION_HOTENDS // Enable thermal protection for all extruders
#define THERMAL_PROTECTION_BED     // Enable thermal protection for the heated bed
also

AnatoliyKube, 2016-03-04:
Again there was such a failure, the item has not been published, the temperature was normal 240gr and 95gr
do something
firmware has errors
once again I remind you that in (https://github.com/ErikZalm/Marlin~~pobj )  there is no such error

Blue-Marlin, 2016-03-04:
I remind you that relaxe/tune means
a. to change the values, 
b. flash the firmware again, 
c. test
d. begin with a. again, until it works.

If you can't make it work that way you can deactivate THERMAL_PROTECTION_HOTENDS on your own risk.

AnatoliyKube, 2016-03-04:
так я же написал что выставил свё и перешил !!! и опять отказ

AnatoliyKube, 2016-03-04:
so I wrote it and put his alter !!! and failure again

Blue-Marlin, 2016-03-04:
Every system is different.
Do not take the default values. You have to __find__ values suitable for your system

AnatoliyKube, 2016-03-04:
but how can I know which option to change? for this it is necessary to learn the ropes algorithm, and have a detailed description of each function in your code, and this appeal to you. Yes ,RC firmware better, but so many nuances in it, for the simple people gathered mechanics and electronics, to deal with the firmware without a sensible description is very difficult,
I want to note:
better graphics on LSD
layers are applied evenly
improved parking
but
circular movements became jerky , 
greater slowdown in the movement , 
and this is only the first impression

What about parking -nuzhno leave for the team g28 only parking at 0 and add a new command (control LSD) -Search 0 s in the center of the table and poluchaetsya that if I turn ENABLE_AUTO_BED_LEVELING, the SLIC3R  takes for 0 wheelbarrow exposed to define Z_SAFE_HOMING_X_POINT ( X_MAX_LENGTH / 2)
#define Z_SAFE_HOMING_Y_POINT (Y_MAX_LENGTH / 2)
 here extruder should come to a point 0 on all osyat ,

AnatoliyKube, 2016-03-04:
osyat- axes

CONSULitAS, 2016-03-04:
@Blue-Marlin i think you build to complex sentences, he does not understand you by google translate. Let me try a different way...

@AnatoliyKube 

"Thermal runaway" is for safety. It protects your machine from fire. It is needed if hotend or heatbed are damaged.

default ist too hard for your machine:
```
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 4     // Degrees Celsius
```

try:
```
  #define THERMAL_PROTECTION_PERIOD 40        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 6     // Degrees Celsius
```

then try:
```
  #define THERMAL_PROTECTION_PERIOD 60        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 6     // Degrees Celsius
```

then try:
```
  #define THERMAL_PROTECTION_PERIOD 60        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 8     // Degrees Celsius
```

use first good parameters

Bigger values are less safe.
Lower values are safer.

CONSULitAS, 2016-03-04:
@AnHardt 
Hat er einen Delta, wenn er über Performance-Probleme schreibt?

AnatoliyKube, 2016-03-04:
I understood , 
question
in firmware https://github.com/ErikZalm/Marlin~~pobj  no such function ( keine solche Funktion )?

CONSULitAS, 2016-03-04:
@AnatoliyKube 
Great!
Answer: Correct. Because it is rather new function.

Fork of ErikZalm is to old. Made more than one year ago.
You can search: https://github.com/ErikZalm/Marlin/search?utf8=✓&q=THERMAL_PROTECTION
Function not found.

AnatoliyKube, 2016-03-04:
so ,
gibt es eine falsche Entscheidung in dem Algorithmus
nichts für ungut
Neue Lösungen müssen überprüft werden
wenn ich als Designer gearbeitet (Elektronik),Wir luden eine Frau,zum Test.zhenshina weiß nicht, wie das Gerät zu benutzen
sie war auf der Suche nach Fehlern, und wir fixieren

AnatoliyKube, 2016-03-04:
so ,
gibt es eine falsche Entscheidung in dem Algorithmus
nichts für ungut
Neue Lösungen müssen überprüft werden
wenn ich als Designer gearbeitet (Elektronik),Wir luden eine Frau,zum Test. zhenshina weiß nicht, wie das Gerät zu benutzen
sie war auf der Suche nach Fehlern, und wir fixieren

so wird es die richtige Übersetzung sein

CONSULitAS, 2016-03-04:
@AnatoliyKube 
Please translate to english, because the others can follow you.
If you speak german, you are welcome, because it is my native language. But don't use a automatic russian to german translator.

No wrong decision in the algorithm!
The algorithm has to be tuned for every model. Because the physics is different. So you need different parameters.

When you tune it for example for a K8200 it is good for every user with this machine. But we don't have your machine to test.

AnatoliyKube, 2016-03-04:
I have a standard printer - ARDUINO 2650, RAMP 1.4,E3d v6 1.75mm  (I made it myself), there is no difference

AnatoliyKube, 2016-03-04:
CORE XY

CONSULitAS, 2016-03-04:
@AnatoliyKube 
If your problem is solved, please close this issue. Thanks and good luck!

AnatoliyKube, 2016-03-05:
how do you know - what causes the printer to stop -EXTRUDER or bed

AnatoliyKube, 2016-03-05:
by what principle works fire protection

Blue-Marlin, 2016-03-05:
> how do you know - what causes the printer to stop -EXTRUDER or bed

Look at the log
```
12:33:47.892 : Error:Heating failed, system stopped! Heater_ID: 0
12:33:47.893 : Error:Printer halted. kill() called!
```
Heater_ID is ether the extruder id number or "bed".

> by what principle works fire protection

a. read the descriptions.
b. read the source.

AnatoliyKube, 2016-03-05:
I read
I understood-
the system makes a decision - error , 
by condition :(temperature.cpp (638))
 if (degHotend(e) < watch_target_temp[e]) {
          // Stop!
          _temp_error(e, PSTR(MSG_T_HEATING_FAILED), PSTR(MSG_HEATING_FAILED_LCD));
        }

for the time (interval)
#define THERMAL_PROTECTION_PERIOD 60        // Seconds
  #define THERMAL_PROTECTION_HYSTERESIS 8     // Degrees Celsius

how much, measurement provides a basis for a decision ? 
if one - if the measurement is erroneous -

Blue-Marlin, 2016-03-05:
#define OVERSAMPLENR 16

AnatoliyKube, 2016-03-06:
one more problem -
if you move a thread from lcd -prepere-move axis - move 1 mm (0,1mm) - extruder -
when the rotation of the encoder -or suspense, or reset -
-similarly as the axis z

-if you rotate the encoder slowly - in sync with the movement, the problem occurs less frequently

thinkyhead, 2016-03-08:
> if you rotate the encoder slowly - in sync with the movement, the problem occurs less frequently

We have patched this recently. Obtain the latest code for `RCBugFix` to avoid Marlin crashing while moving with the encoder wheel.

AnatoliyKube, 2016-03-08:
I'm going to try now

AnatoliyKube, 2016-03-08:
No error persists
 lcd -prepere-move axis - move 1 mm (0,1mm) - extruder -
when the rotation of the encoder - reset -

AnatoliyKube, 2016-03-08:
extruder  - try to put 300 mm, 100 mm or more there will be reset
Further
twitching occurs characters  X , Y , Z  on the screen

thinkyhead, 2016-03-10:
> twitching occurs characters X , Y , Z on the screen

Does G28 help that?

thinkyhead, 2016-03-10:
> when the rotation of the encoder - reset -

It sounds very strange. Are you ok sticking with Version 1.0.2-1 for the time being? I cannot imagine what is combining on your system to cause such issues. All we can do is compare the old code and the new code, and try to isolate the changes that lead to instability. For my part, I am unable to reproduce any problems using the latest `RCBugFix` code with my hardware.

AnatoliyKube, 2016-03-10:
if you run the g28, twitching  characters disappears,
but ,
after a while begins again(the latest RCBugFix)

Blue-Marlin, 2016-03-10:
That blinking xyz is a wanted change.
In the version before you have seen "---:" insdtead of numbers behind the XYZ when unhomed.
Stepper deactivation time after DEFAULT_STEPPER_DEACTIVE_TIME did not work at all.

Now you see altering XYZ <-> ? before homing and the relative to boot-position coordinate positions.
After homing static XYZ and position.
When DEFAULT_STEPPER_DEACTIVE pased by without a move the steppers are now deactivated.
The display shows then altering XYZ <-> ' '.
You can stop stepper deactivation for all axes by setting DEFAULT_STEPPER_DEACTIVE to 0, or for every single axis with DISABLE_INACTIVE_? false.
If you want the stepper deactivation but not the blinking with ' 'you can define DISABLE_REDUCED_ACCURACY_WARNING.

Blue-Marlin, 2016-03-10:
@thinkyhead 
#3050 fixed xyz but not e and the z-move in manual homing.
Do you want to finalize your #3110 or should i try to extend my #3050?

closed 2016-04-28 by jbrazio