MarlinFirmware/Marlin #3055: Gina and nopherad frustrated about Marlin

AnHardt, 2016-02-28:
https://github.com/foosel/OctoPrint/issues/450#issuecomment-189849243 ++

AnHardt, 2016-02-29:
Some issues:

M110 is not handled at all when send without linenumber/checksum.

M29 does not send a 'ok' when actually closing the file.

thinkyhead, 2016-02-29:
Good to know. I will produce a patch. Both `M110` and `M29` are handled specially, ahead of the regular GCode parsing.

If @foosel and @nophead have any details to add about their needs and observations, this would be the place and time to do it. I would like to get all the bugs reported for RC3 worked out and release RC4 as soon as possible.

nophead, 2016-02-29:
If I am correct in asserting it has always been like this in Marlin then it needs careful consideration of whether it should change because all hosts would need a matching change.

foosel, 2016-02-29:
Sadly I have to agree here. On the other hand, it is broken behaviour as it is now.

I recently ran into another issue with M600, do you want a separate ticket for that?

thinkyhead, 2016-03-02:
> M110 is not handled at all when sent without linenumber/checksum

This is because `M110` requires an `N` argument – and currently if you have `N` then you _must_ also have a checksum. Should we make an exception for `M110`?

> `M29` does not send a 'ok' when actually closing the file.

I see that it _does_ output "**Done saving file.**" Does this mean that `G29` will cause `ADVANCED_OK` to skip a line number, and the host continually asked to re-send the `G29` line again?

> it has always been like this in Marlin

@nophead Perhaps we should test out the changes @AnHardt suggests and see how they _actually do_ affect hosts. (I'm a science-titian!) But maybe you _already know_ how it will affect hosts…?

> `M600`, do you want a separate ticket for that?

That'd be awesome. Please do!

nophead, 2016-03-02:
Currently I fixed my fork of OctoPrint to work with my existing firmware rather than the other way around, see https://github.com/nophead/OctoPrint/commits/sdwrite_fix.

Not sure what the best fix is but on practical level it is easier for me to update OctoPrint instances than it is to re-flash firmware, which is machine specific.

foosel, 2016-03-02:
>> M110 is not handled at all when sent without linenumber/checksum

> This is because `M110` requires an `N` argument – and currently if you have `N` then you must also have a checksum. Should we make an exception for `M110`?

I think you misunderstood @nophead here. If `M110` is sent without checksum and line number, it doesn't have any effect. As in, `N0 M110*<checksum>` resets the line number, `M110` doesn't.

> I see that it does output "`Done saving file.`"

Yes, but *only* that. What @nophead and others observed is that no `ok` followed it. Since the `ok` pretty much tells the host "oh, hi, thanks for your command, I'm done processing it now and you may continue sending more, thanks!", any command that doesn't get properly acknowledged with an `ok` - here `M29` - causes problems since suddenly there is a command which never finished ... so what should the host do?

I just took a look through the current code in this repo though and *from the looks of it* it might be that this is already solved in the current version. I can't test with a current Marlin right now to be absolutely sure, it's an easy test though, open a serial line to a Marlin equipped printer, send this:

    M28 test.gco
    M117 hello world
    M29

See if it outputs

    Writing to file ...
    ok
    ok
    Done saving file.
    ok

or 

    Writing to file ...
    ok
    ok
    Done saving file.

Former is ok, latter is buggy.

> But maybe you already know how it will affect hosts…?

See above. Usually each command is assumed to get a matching ok or the communication cannot legally continue. If hosts have in the past compensated for the wrong behaviour of M29 and injected a fake "ok" in their logic, if it now suddenly shows up that can cause stuff to get out of whack. That can have absolutes no consequences (everything just works fine). Or it can cause interesting effects where the read buffer of the firmware gets spammed by the host and stuff freezes (I saw this behaviour in the past with some printer models when OctoPrint was not yet fully strict regarding the command-ok-command-ok-cycle).

> [ticket about `M600` issue would be awesome]

Ok, I'll open it briefly.

thinkyhead, 2016-03-04:
> I think you misunderstood @nophead here. If M110 is sent without checksum and line number, it doesn't have any effect.

That is what I understood @nophead to mean.

> As in, N0 M110*<checksum> resets the line number, M110 doesn't.

So, are you saying that `M110` with no argument is _specified_ to set the current line to 0?

Are you also saying that if a host (for example) is sending commands, then it may be doing so with no line numbers and no checksums? Under what circumstances will `M110` be received outside the context of a host-controlled print job with line numbers and checksums? I was also under the impression that for `M110` the `N` argument is required. This is why I mentioned that checksums would then also be required.

nophead, 2016-03-04:
I think I am right in saying in all versions of Marlin ```M110 does``` nothing at all. OctoPrint has recently started using it to provoke a response on connection. I mistakenly thought it was to reset the line counter but @foosel informed me it wasn't. I think she intends it as a NOP.

In the version of Marlin I use ```Nn M110*s``` sets the current line number to n. ```M110 Nn``` gives missing checksum. ```Nx M110 Ny*s``` would set it to Nx I think. There is no handler for M110 but the code that handles line numbers will find the first N anywhere on the line, demand a checksum and allow the n to not be last_line + 1 if it finds M110. I think ```M110 Nn*s``` would work.

In the latest version of Marlin this section has been re-written and behaves differently. I think ```M110 Nn``` is a NOP because there is still no handler for M110, just special code that is invoked only when the line starts with N. A difference is it looks for a second N and uses that if found, so ```NxM110 Ny*s``` will set it to Ny instead of Nx.

The specification of M110 says it takes an N parameter and uses that. It should work with or without line numbers and checksums. 

There seems to be some versions in the wild that handle ```NxM110 Ny*s``` differently. Nx has to match the last_line +1 and then it is set to Ny.

foosel, 2016-03-04:
Can we please not talk about a "specification" here? The wiki isn't one, it's a "we wish it were such and such, but if it is you'll have to look up yourself in the source code of your firmware" and the NIST standard doesn't apply either.

I send M110 on printer connect since I have to send something to figure out if I get an ok back ("it's a printer!") or not ("it's not!"). Since there are printers out there which do NOT reset on connect on serial, and if they require linenumbers and checksums to be sent at ALL times (depends on firmware and settings) I have no other clean way to get a communication going in the first place than making M110 my first command to tell the firmware "look, hi, nice to meet you, I have NO idea at which line number you might be right now so let's just start at the beginning, right?".

Now, there are various interpretations of the M110 command in the wild. I saw some that require a real `N` parameter (NOT a line number prefix!) like it's written on the wiki, so `M110 N0` works regardless of checksum or not, some like existing Marlin variants for which `N0 M110` and `M110 N0` are equivalent, others (current Marlin?) where only `N0 M110*<checksum>` would do something, and then yet others which somehow got confused altogether and demand something like `N<n+1> M110 N<new n>`, which of course is completely useless to get comms going if you don't know the current `n`. I do remember about a year or so ago to run into a situation where the both a prefixed and parameter `N` were required and supposed to be equal, but I can't remember what firmware/variant/printer that was. I'm also fairly sure that I saw `M110` without parameter being not a nop but assuming `N0` (so resetting to 0, making the next required line a 1), but I might be mistaken there and misremembering things.

Thing is - like @nophead said - that `M110 N0` used to work, now it doesn't. It works like this with other firmware. There is no specification, no standard which we can depend on here, but at least Marlin IMHO should try not to deviate completely from how others do it, so that host implementations don't have to start implementing the whole comm layer separately for each and every firmware variant.

nophead, 2016-03-04:
```M110 N0``` used to give a missing checksum. ```M110 Nn*s``` used to work but now it doesn't.

foosel, 2016-03-04:
Which in and of itself is an error since it shouldn't interpret a parameter as a line number segment just because they happen to use the same prefix character, but that's some other issue (and I think also present in other firmware).

nophead, 2016-03-04:
Yes I think all versions have one bug or another, like most things in Marlin due to massive re-factoring without regression tests. I have given up on it and don't follow it any more. I only got drawn into this conversation by the mention.

Jnesselr, 2016-03-04:
@nophead That beautiful comment describes the exact state of this project.

nophead, 2016-03-04:
IMHO it should be as follows:

M110 should be handled as a normal command, taking an N parameter which sets the line number. What it should do with a missing N is debatable but probably treating it as NOP is most backwards compatible.

There should also be special case check for M110 in the line number handling that allows it to not have the expected line number at the start. That has the side effect of making ```NnM110*s``` work.

So
```M110``` NOP.
```M110 Nn``` sets last line to n.
```NnM110*s``` sets last line to n, doesn't care what n is.
```NnM110 Nx*s``` sets last line to x, doesn't care what n or x is.

Roxy-3DPrintBoard, 2016-03-04:
> So
> M110 NOP.
> M110 Nn sets last line to n.
> NnM110*s sets last line to n, doesn't care what n is.
> NnM110 Nx*s sets last line to x, doesn't care what n or x is.

Unless somebody objects...   Let's make this happen ASAP.   And lets put comment blocks around the code that it should never be changed and maybe reference this thread.

@foosel  If we get this behavior in the current Release Candidate, and propogate that behaviour forward, will that be good enough?

foosel, 2016-03-04:
@Roxy-3DPrintBoard It should hopefully prevent the current problems from further propagating through the various forks. Especially the change to `M110` hence will stay in the RC versions and hopefully vanish from printers soon.

Sadly, as with every Marlin bug that's been present a while, as the author of a host I'll still have to think of something to work around bugs that have been fixed in Marlin mainline (e.g. `M29` lacking the `ok` if you decide to go that route, personally I'm still leaning towards fixing that, even though it's always been broken in the past), because people don't upgrade their printers, or can't if they depend on forks that are not updated. But that's not a problem you can solve.

Roxy-3DPrintBoard, 2016-03-04:
Would it be best to fix M29 to do the 'OK' ?    The downside for you is that means M29 has two possible responses.   But really...   It feels to me like it should be fixed.    

> Sadly, as with every Marlin bug that's been present a while, as the author of a host I'll still have to think of something to work around bugs that have been fixed in Marlin mainline 

Yes.  But there may be ways we can help you out.   For example, the whole reason you are sending M110 is to help synchronize things.   Would some extra information on the response (such as firmware version number or date) be helpful so you can draw a line in the sand?

foosel, 2016-03-04:
Not on `M110`. An `M115` I can rely on would help. Pretty much all Marlin versions currently present on printers report "1.0" here, with author "default". Making the version non-configurable and part of the  release (last time I looked it was overridable in `Configuration.h`) would already help in that I could at least be fairly certain that a "Marlin 1.2.3-RC4" indeed bears some resemblance to mainline "Marlin 1.2.3-RC4", at least more than to "Marlin 1.0", "Marlin 1.2" or "Marlin 1.3", no matter from which fork it came. Since vendors might want to override the version field with their own firmware version, it might be better to enforce some Marlin version output with `M115`. Machine parseable.

foosel, 2016-03-04:
Oh, sorry, forgot the `M29` thing - I could probably build in another workaround that can manage those two cases, and I'd prefer to rather fix obvious bugs such as that instead of carrying them around endlessly and spreading them further to thousands of forks.

But - I'm not the only host developer. I don't know how Repetier Host handles that, or Printrun, or Cura etc.

Roxy-3DPrintBoard, 2016-03-04:
If that is the way we go, it would feel most correct to change the PROTOCOL_VERSION to something like 1.1     After all, it is the protocol we are trying to change and correct.   Also, the user isn't likely to be changing the #define PROTOCOL_VERSION "1.x" message even if they customize the other information on the machine:
~~~~
#define PROTOCOL_VERSION "1.0"
#define MSG_M115_REPORT "FIRMWARE_NAME:Marlin " DETAILED_BUILD_VERSION " SOURCE_CODE_URL:" SOURCE_CODE_URL " PROTOCOL_VERSION:" PROTOCOL_VERSION " MACHINE_TYPE:" MACHINE_NAME " EXTRUDER_COUNT:" STRINGIFY(EXTRUDERS) " UUID:" MACHINE_UUID "\n"
~~~~

I'm looking right now at the 4 desired responses that NopHead posted to see what it is going to take to get that to happen.

Roxy-3DPrintBoard, 2016-03-04:
We will need to really check this out carefully.  But I think we get NopHead's desired behaviour if we just make this change:
~~~~
      if (npos) {
        boolean M110 = strstr_P(command, PSTR("M110")) != NULL;

        if (M110) {
          char *n2pos = strchr(command + 4, 'N');
		if ( !n2pos && !apos )			// is it just a M110  ?   This should be treated
			return;				// as a NOP
          if (n2pos) npos = n2pos;
        }
~~~~

nophead, 2016-03-04:
To make what I described happen you need to do what I described. That is make a normal M code handler for M110 that does:
```
inline void gcode_M110() {
      if (code_seen('N') 
            gcode_LastN  = code_value();
 }
```

Then reduce the special handling of M110 here: https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L841 to
```
        boolean M110 = strstr_P(command, PSTR("M110")) != NULL;
        if (gcode_N != gcode_LastN + 1 && !M110) {
          gcode_line_error(PSTR(MSG_ERR_LINE_NO));
          return;
        }

```

Roxy-3DPrintBoard, 2016-03-04:
Yeah...   I deleted the previous thought and I was just getting ready to say:

We need a dedicated M110 handler.   It is too complicated to try to leverage all the checksum stuff....

nophead, 2016-03-04:
Exactly because M110 should work without a line number so it can't be handled in the code that is only invoked when you do have a line number.

thinkyhead, 2016-03-08:
Seems @AnHardt is addressing some of these things in #3108. Giving them a test run now…

closed 2016-04-09 by AnHardt