MarlinFirmware/Marlin #3058: Use LCM1602

CoderSquirrel, 2016-02-29:
Is Possible to use lcm1602 with marlin?
It has only 4 pins instead of the 20 pins of the normal 16x2 lcd.

Blue-Marlin, 2016-02-29:
No. At least not out of the box.
There are some I2C panels definable in Marlin - but if one of them is similar enough to drive your display is questionable.
16x2 is supported. But be warned - there is not much room on that displays (https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/ultralcd_implementation_hitachi_HD44780.h#L503).

CoderSquirrel, 2016-03-15:
Hy guys, i managed this by using this way:
i usde the scl and sda pins, 20 and 21.
And in code i used the LiquidCrystal_I2C and de declaration of the lcd was LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE); 
Just for you guys if you want to add there.

closed 2016-03-15 by CoderSquirrel