MarlinFirmware/Marlin #3059: Help with FW e SW Delta printer

mmatt94, 2016-02-29:
Hello
I am Matteo , student of Automation ad Packaging from Bologna, Italy

I need a help regarding the firmware to a related software to be able to use the robot.

I'm doing a Delta Robot, like the one in the picture (link), for use as a 3D printer.

The robot is driven by brushless motors 3 Nema 23 with its driver. This
driver has the direction inputs, Step, etc. and it can optionally be
controlled by USB. (The driver of the seller said that can be controlled
by macro

I think use arduino and RAMPS 1.4 board (and a Repetier SW & FW), but this method is used for linear and non-parallel delta as my robot.

He is able to give me some advice to program Arduino with the features of my robo to put it to use?

Thank you very muchl!

[http://www.elmomc.com/capabilities/6.GMAS%20Multi%20Axis%20Motion/12.GMAS%20Delta%20Robot/picture/6-12-1.png](url)

Blue-Marlin, 2016-02-29:
[image]
Sorry. That's not a kinematic we already support.
This seems to have elements like a SARA machine - calculating angles, mixed with 3 arms - like a DELTA.
So, DELTA and SCARA are the code paths you should read in Marlin.
Likely it's much easier to integrate a new kinematic into RepetierFirmware. Marlin is not very well prepared for that.

whosawhatsis, 2016-02-29:
@Blue-Marlin Actually, outside the 3D printer world, this is what a canonical delta looks like. The linear deltabots that are usually used for 3D printers are almost unheard of elsewhere, and most are controlled by the angles of 3 arms from a fixed base, which is usually above the work area.

That said, @mmatt94, he's right that this type of deltabot is not typically used for 3D printers, and is not what the delta configuration here refers to. There are other bots, like the Firepick Delta, that do work this way though. The Firepick guys have their own fork of Marlin (https://github.com/firepick-delta/Marlin), which is likely to have support for this type of deltabot.

Roxy-3DPrintBoard, 2016-03-02:
@mmatt94  In order to write the software to support this kinematic's, the software developer would need to have a machine to code against.   If this is a commercial product (or going to be a commercial product) you should consider getting a few machines into the 'right' developer's hands.

closed 2016-04-27 by jbrazio