MarlinFirmware/Marlin #3060: Babystepping

ruggb, 2016-03-01:
I have babystepping working on my CoreXY now. Apparently it has been working for a long time.
The key is when it is enabled the XY portion of it must be disabled.

I am proposing a couple of changes:
1. I have no idea why babystepping on X or Y is necessary. I can't think of when I would possibly want to use it on my machine. Maybe for a different machine it is useful. I propose that if babystepping is enabled and corexy is enabled the code just disable XY function - or leave Z & XY disabled by default so when Z is enabled u don't need to disable XY for coreXY - or both. It seems that everyone I asked about this operating on coreXY for Z axis just knew it did not work for coreXY - apparently b/c if it is enabled u get the error since XY is default enabled. AND I think the comment said it did not work for coreXY. That comment is now moved to the XY enable line but could be slightly changed to be clear that it means XY only is not implemented for coreXY.

2. Babystepping is awkward to test and use because it only appears in the menu when a print is running, it is buried in the menu such that it is time consuming to get to, and the time of 15 sec is too short to use this feature. I have moved the routine from the tune menu to the first item on the first menu so it is always quickly available for use and testing purposes and increased the time to 30 sec. to be able to view the result of manipulations before it times out.

Please evaluate these proposals with ur global perspective to see if they make sense.
I have read other post re these same issues so it isn't just me.

thx

thinkyhead, 2016-03-02:
I like these proposals. But I would like to get some feedback from others about how and whether they use `BABYSTEPPING`. Personally, I have never used these options either. I presume they exist to fix an axis getting thrown off without stopping the print and leaving a blob.

ruggb, 2016-03-02:
Well for me, and I think most ppl using it, it makes it MUCH easier to adjust the zero point as the print is starting. I don;t think is is used to fix a problem of axis getting thrown off, which is why I don't think anyone uses the XY part.
I no longer must kill the print and readjust the Z home. I find that the home position is not the same for PLA & ABS which makes it more troublesome. Babystepping on Z is wonderful.
When I was searching for a way to make this work I found a number of posts RE the position of the function on the menus. Some ppl were talented enough to move it, as I did. Some not.
I didn't hear any complaints about the timeout - maybe cause they found it and it was an easy fix.
But 15 sec goes by too quickly when trying to decide if it is right.
I didn't find anyone talking about XY, or if I did they were doubting it was used by anyone.
Having it always available should not be an issue if you are testing it even though it doesn't respect endstops. I think everybody can understand that.

**Also Scott**, when clicking on home all, or home Z, in Repetier it just issues a G28. I don't think there is a way to cause Repetier to drop the Z platform a little before homing. I have scripts that do that, but the built in controls won't.
I know G28 is a std function, but would it be possible to add a user configured switch to make G28 operate per spec or with a drop so I can set it in the firmware? I would want it to always do that.

thinkyhead, 2016-03-08:
@ruggb #3107 proposes to add a Z raise before homing for everybody.

ruggb, 2016-03-10:
**STOP THE PRESSES**
the Z homing change affects babystepping.
when Z is decreased it immediately increase it again. If u turn the knob fast to change it a lot it will some progress toward zero but not without a lot of argument.
If u then try to increase Z it will do the same thing for a few turns the eventually stop arguing so much but not totally.

ruggb, 2016-03-10:
I closed #341 re babystepping
#2935 is still open but is for the same thing as this one
I would like to see this chg in R4, but it seems to keep getting overshadowed with something else.
Fortunately, I am concerned about babystepping and find it doesn't work now with the Z homing chg.

I guess it is my fault for including both in this thread.
Tell me how to fix it so babystepping gets some attention and Z homing gets fixed.
Or since they seem to interact both get done together.

thinkyhead, 2016-03-10:
I will investigate this issue today. I have not used babystepping, nor tested it on my machine.

ruggb, 2016-03-10:
@thinkyhead
There is a possibility it has something to do with the fact that I just tested it with the machine not printing. I will try later with it printing, though it didn't matter b4. It worked either way.
Since urs isn't modified to work with it not printing, it may work. Let me know.

thinkyhead, 2016-03-11:
Since `BABYSTEPPING` is not supported for CoreXY (or delta, either)… is the proposed solution simply to disable babystepping for XY when COREXY is enabled (and to disable XZ babystepping when COREXZ is enabled)?

The reason I haven't messed with `BABYSTEPPING` till now is that I have seen it as a low-level "tweaking" feature without a clear set of specific use cases. And it has always applied directly to moving the steppers rather than referring to the XYZ position (where this is possible, in other words not on Deltas). So it hasn't been a priority to "change the meaning" of `BABYSTEPPING` for CORE machines. However, now that I understand more its specific purpose —to perform small tweaking movements independent of the coordinate space— it seems like it should be pretty straightforward to insert the XY and XY implementations for CORE machines. Basically, `BABYSTEPPING` on a CORE machine will sometimes move two axes instead of one. But they both move in a babystep-style fashion.

thinkyhead, 2016-03-11:
`BABYSTEPPING` on a CoreXY from the LCD menu should do the right thing if articulated like this…
```cpp
switch(axis) {
  case X_AXIS:
    babystepsTodo[A_AXIS] += distance * 2;
    babystepsTodo[B_AXIS] += distance * 2;
    break;
  case Y_AXIS:
    babystepsTodo[A_AXIS] += distance * 2;
    babystepsTodo[B_AXIS] -= distance * 2;
    break;
  case Z_AXIS:
    babystepsTodo[axis] += distance;
    break;
```

thinkyhead, 2016-03-11:
Ok, so #3127 implements `BABYSTEPPING` for CoreXY and CoreXZ so you can use `BABYSTEP_XY` on those machines now. I implemented it as a high-level feature, as specified in the code above. Since paired motors on Core machines are always moved in proportion to each other, I could instead implement it as a low-level feature, but it would need some work.
```cpp
#if ENABLED(BABYSTEPPING)
  for (uint8_t axis = X_AXIS; axis <= Z_AXIS; axis++) {
    int curTodo = babystepsTodo[axis]; //non-volatile for performance
    if (curTodo != 0) {
      #if ENABLED(COREXY) || ENABLED(COREXZ)
        bool dir = (curTodo > 0);
        switch(axis) {
          case X_AXIS: // same direction for X
            babystep(A_AXIS, dir);
            babystep(CORE_AXIS_2, dir);
            babystep(A_AXIS, dir);
            babystep(CORE_AXIS_2, dir);
            break;
          case CORE_AXIS_2: // opposite direction for Y
            babystep(A_AXIS, dir);
            babystep(CORE_AXIS_2, !dir);
            babystep(A_AXIS, dir);
            babystep(CORE_AXIS_2, !dir);
            break;
          case CORE_AXIS_3:
            babystep(CORE_AXIS_3, dir);
            break;
        }
        if (dir) babystepsTodo[axis]--; else babystepsTodo[axis]++; //towards 0
      #else
        if (curTodo > 0) {
          babystep(axis, true); // forward
          babystepsTodo[axis]--; //fewer to do next time
        }
        else {
          babystep(axis, false); // backward
          babystepsTodo[axis]++; //fewer to do next time
        }
      #endif
    }
  }
#endif //BABYSTEPPING
```
You can see that this would technically work, but diagonal movements would waste steps. So it makes more sense to stick with the high-level (and much simpler) version that I posted above.

ruggb, 2016-03-11:
GUYS ........ I hate to get angry but u r frustrating me. Just because u don't have a coreXY or have one and believe everyone b4 u saying it isn't supported so u have not tried it == STOP SAYING IT ISN'T SUPPORTED. WHAT DOES NOT WORK IS THE XY PART OF BABYSTEPPING. BABYSTEPPING Z WORKS PERFECTLY FINE. How many times do I have to post that message b4 u get it??????????
I have proposed that u undefine babystepping_XY as a default. For one, I am betting no one uses the XY function on any machine so few if any would enable it. For another U WILL FIND OUT IT WORKS PERFECT FOR COREXY for the Z axis because when it is enabled it will just work and u don't have to do something else to make it work. So u won't have to keep telling everyone it doesn't work for coreXY. Now if u fix it for XY that is a plus for U as I don't use it so it doesn't matter to me.

1 more time - repeat after me......BABYSTEPPING ABSOLUTELY WORKS JUST FINE FOR THE Z AXIS ON A COREXY MACHINE.

Now, that being said your update "bedd...." broke it. It no longer works as I detailed above.
I am back to 9a3408.... and it is just perfect.
I don't think I messed up in transferring my config files, but I may have so I will recheck them.
And while u r at figuring that out it would be MUCH simpler for u if u moved the routine around as I suggested so u can easily test it without setting up a print and printing. It makes it MUCH more user friendly to use and others besides me are asking/wishing for that change. Others have also implemented partial moves to the top of the TUNE menu but I think a full move to the top of the MAIN menu is the best solution for just this reason - testing - also increase the default timeout to 30000 instead of 15000 as the watching/analyzing of traces takes longer than 15000.

1 last time - BABYSTEPPING_Z WORK PERFECTLY ON A COREXY MACHINE - if babystepping_xy is disabled.

thinkyhead, 2016-03-11:
So a comparison of 9a3408 and current should yield a clue as to the cause of this conflict. I will take a look.

ruggb, 2016-03-11:
AMEN

ruggb, 2016-03-11:
my config and config adv files r good.
I have a clue for u --- when I first config the Z_MIN_HEIGHT_FOR_HOMING I set it = 20
when I checked, I set it back to 4.

At 4 it is ALMOST as quite as the old build but it still sounds like it is arguing with itself.
So set the Height for 20 when u try it.

ruggb, 2016-03-11:
disclaimer - I am not a programmer
 I'm thinking the problem is here: line 2328+ in Marlin_main.cpp
    #elif defined(MIN_Z_HEIGHT_FOR_HOMING) && MIN_Z_HEIGHT_FOR_HOMING > 0

      // Raise Z before homing any other axes and z is not already high enough (never lower z)
      if (current_position[Z_AXIS] <= MIN_Z_HEIGHT_FOR_HOMING) {
        destination[Z_AXIS] = MIN_Z_HEIGHT_FOR_HOMING;
        feedrate = max_feedrate[Z_AXIS] * 60;  // feedrate (mm/m) = max_feedrate (mm/s)
        #if ENABLED(DEBUG_LEVELING_FEATURE)
          if (marlin_debug_flags & DEBUG_LEVELING) {
            SERIAL_ECHOPAIR("Raise Z (before homing) to ", (float)MIN_Z_HEIGHT_FOR_HOMING);
            SERIAL_EOL;
            print_xyz("> (home_all_axis || homeZ) > current_position", current_position);
            print_xyz("> (home_all_axis || homeZ) > destination", destination);
          }

My first clue is If it is = 0, this routine is skipped
my 2nd clue is **(never lower z)** in the 2nd line. and the fact that it makes the most noise when it is trying to move toward the endstop. And it seems like the min noise it makes is when it thinks it is increasing Z from where it originated. IE when I decrease Z it makes the most noise, but it does move some (so that may be another issue for bed leveling). When I move it forward it makes less noise until it gets to some point, I am presuming the point it thinks it started from, then it is quite, but not as quite as if it never went thru this routine in the 1st place. If I only increase Z it makes about the same noise as the min noise above.
good luck

thinkyhead, 2016-03-11:
Under what circumstances are you trying to use BABYSTEPPING?

- What is the expected behavior?

- What is the actual behavior?

- What steps must someone take to reproduce the problem?

Thanks!

thinkyhead, 2016-03-11:
I don't think `BABYSTEPPING` has been directly broken by the introduction of `MIN_Z_HEIGHT_FOR_HOMING`, but the introduction of this feature may be exposing an existing flaw.

As I look at the babystepping feature in isolation, I see that it has no real feed-rate limit. It just does one step per Temperature ISR. I'm not sure how fast that interrupt runs, but perhaps it's trying to step faster than the Z axis can handle. We could try reducing the frequency that the babystep handler runs at and see if it helps.

ruggb, 2016-03-11:
more. with the old 9a3408 I reset then go to BS and turn the knob cw (decrease Z) quickly. I get a loud noise as it move like it jumps all the steps at once. If I turn is slowly it still seems to jump but since it isn't asked to go as far it isn't as noisy. After the first time it does not do that jump. It has not been homed. If I turn CCW it does not jump but it is still noisier.
If I home it first, no jump and it is very quite and smooth both ways.
When I reset after BS it jumps back to someplace, maybe where it thinks it started from after the first reset. Bottom line is that all bets r off until it is homed. U probably don't want to auto home as soon as it is powered on but nothing should work until it is homed.
WAIT A MINUTE!!!.......................

OK - went back to bedd with 20 and APPARENTLY I SCREWED IT UP.
I forgot to home it b4 trying it out.

Bottom line IT DOES APPEAR TO WORK -I'M AN IDIOT

SO MAKE IT BILL PROOF and disable any movement until it is homed. Maybe consider disabling bed and extruders also
IE G28 is the only thing it responds to as a first command - but what do I know.

The home drops 20mm and the BS works just fine and quite for Z.
MY APOLOGIES

ruggb, 2016-03-11:
The problem I was having that I thought was Marlin not being able to process high step rates fast enough was exactly that. Whatever was chgd in 9a3408 fixed that issue.

thinkyhead, 2016-03-11:
Well, at least we got a working `BABYSTEP_XY` for CORE machines out of it!

closed 2016-03-11 by thinkyhead

ruggb, 2016-03-11:
Whether it is needed or not
At least ppl will stop saying it doesn't work for core
I will try it when u build it - or maybe I am too dangerous for that.
Just don't  break the Z function.
Thx Steve

ruggb, 2016-03-11:
@thinkyhead 
Hey Scott - notice I even got ur name right this time--
One thing I notice, the raise Z function is referenced to the home position for Z so that if the position is 4mm off the bed it does not raise it any more.
 Is that what u want?
If the system crashes is it possible to have the reference position wrong?
If the print is stopped the increase Z no longer works if it above 4mm.
so it does not function to raise Z to insure clearance of any obstacles on the bed.
Isn't that part of why u added it?
wouldn't it be better to raise it relative?

thinkyhead, 2016-03-13:
@ruggb As far as I know, `BABYSTEPPING` applies no limits to its use. Except that you must be printing or otherwise already moving the print-head.

Or are you talking about `Z_RAISE_FOR_PROBING` … or `Z_MIN_HEIGHT_FOR_HOMING`?

ruggb, 2016-03-13:
when I am using the bedd... build and min_Z_.... =4, if I issue a g28 it checks to see if it is >=4, if so it does not raise it more. That means if there is something on the bed I won't get any clearance for it. Would it not be better to raise Z relative regardless of where it is so that wherever it is it increase another 4 (or whatever I have set). I don't think it will be a problem for ABL but I don't use it.

RE Babystepping - I don't like the fact that I can only access it when a print is running. Set up that way I can't easily test it or get comfortable with it. Obviously, I have to be aware that I could run it into the bed. I bet a lot of ppl would like to do what I am doing when they finally figure out what it is and how it works, but I guess I can just rearrange mine everytime I update.

Blue-Marlin, 2016-03-13:
@ruggb 
The idea to have a height instead of a raise here is to minimize the risk of crashing at z-max.
This MIN_Z_... is only appied when we home z to the z-min-endstop. Usually that means we have no z-max-endstop what could protect us from going to high.
MIN_Z_... has ever had that risk and still has it - but now a little bit less. As long as you boot the machine with z below Z_MAX -  MIN_Z... we are save now, even if you have raised the z-axis before homing.
If your assumed obstacles are higher than MIN_Z... , just increase the number. If you can boot your machine with z lower than bed-level, just increase the number until you are high enough for an xy-move, every time. But be aware the risk at the top. Higher number -> higher risk. But in every case now the risk is smaller than before.
For a second homing the situation is even better. We already know if we are high enough and only raise if we know we are to low. If you already have parts on your bed and and have to rehome x or y, you're in trouble anyway. But just lifting z by a defined value, homing, returning to the former x/y, and lowering z the same amount, should do the job. Even rehoming z is possible if you can find a way down to the bed.

BABYSTEPPING is crap anyway. Adjusting positions on the fly is completely contradictional to the concept of CNC. Set up you machine properly or use a 3Doodler.

ruggb, 2016-03-14:
I didn't think of it being too high as I have 12" and have yet to get close.
How about comparing the Min_Z_....  with the MAX Z instead of the 
current Z for G28 and if there is enough headroom allowing the Z 
increase up to the MAX Z - If I set it for anything less than wherever 
it is, it won't raise. Setting it to 20 isn't high enough as it can 
easily be over that in the middle of a print crash. And setting it to 20 
greatly increase the home time as it is.

Babystepping - Don't knock my favorite feature.
If everything were perfect babystepping would be useless.
however, it ain't - Beds that are not flat, filament that extrudes 
differently, etc. - CNC is obviously different and I guess it is 
disabled for them, but when I can tweak the nozzle height while it is 
laying down the brim or skirt, I absolutely love it. Aborting and 
starting over is a real PITA.

thinkyhead, 2016-03-16:
I still don't understand the settings you are using for homing. I'm also failing to understand what you mean when you say you can't use the "raise Z" function.

When reporting an issue it helps to give these points in bullet order:

- The behavior you are expecting.
- The behavior you are actually getting.
- Steps that another user must follow to reproduce the behavior you are seeing.
- The relevant configuration options, or the entire configuration if needed.

ruggb, 2016-03-16:
scott
When I killed a print @ Z ~=> 4 and issued G28 there was no increase of Z, I assume b/c Min_Z_for_homing was set at 4 and it was already past 4.

I think that u are looking to see if Z >= min_z_for homing and if it is u don't raise Z - which means u r very likely close enough to a print to interfere. If not, I don't understand why Z didn't increase.

Ur concern that Z might already be @ Z_MAX is obviously valid.
If u looked at MAX Z and increased Z up to min_z_for_homing value then u would probably have a 99.44% chance that it would increase Z and give some clearance.

It may be that all u r thinking about is the ABL operation, which apparently it works for and I can't verify. But why not open ur vision and look at other option. I don't think comparing to Z_max will affect the ABL operation though it may raise it a little more than u designed it for. Is that a problem?

Blue-Marlin, 2016-03-17:
@ruggb 
Would you please be so kind, to try, to use a less sloppy language.
For at least half of your audience her, English is not the preferred language.
We try hard to understand you and really give our best to express our selves.
If you want us to help __you__, or to change things for __you__, or to answer __your__ questions, don't make communication unnecessary difficult for us.

ruggb, 2016-03-17:
@blue-marlin
please explain

Blue-Marlin, 2016-03-17:
"Wenne dat lese kans, bisse mei Held". But probably it will take a while until you find out what I wrote here.
Every word or abbreviation we can't find in a dictionary degrades your chance for an answer.

ruggb, 2016-03-17:
point taken. I thought I was just talking to one person.

thinkyhead, 2016-03-18:
the naming of "min z for homing" might be misleading. Before any G28 (at boot time) the current Z position is "presumed" to be zero. The first thing you should do upon powering up the machine is an initial G28 so that you have a known position. On a Cartesian machine, you are right that 99.9% of the time you have the safety to raise Z probably 100mm without a problem. So it has been proposed to make `Z_SAFE_HOMING` a global option, not just a sub-option for `AUTO_BED_LEVELING_FEATURE`. In the meantime, if 4mm is not enough clearance for you, and you feel safe going higher, then why not just set `MIN_Z_FOR_HOMING` to 10 or 15?

ruggb, 2016-03-18:
Let me try to say it again in a different way...........
It does not matter what MIN_Z_FOR_HOMING is set at, I can set it for 100,

if a print is killed or crashes and I issue a G28, the first thing it does is to look at where Z is.

If Z >= MIN_Z_FOR_HOMING then it proceeds to home X, Y, &Z. If there is a print on the table there is a potential crash coming - especially if the slicer is programmed to finish each piece first. The only situation setting MIN_Z_FOR_HOMING to a greater value would help is if the print happens to be UNDER the set value. If I set MIN_Z_FOR_HOMING to Z_MAX that might work but it would run to that value every time I issued a G28, then back to Z = 0.

If Z< MIN_Z_FOR_HOMING then it increases Z by MIN_Z_FOR_HOMING and proceeds to home X, Y, &Z. The assumption is that since it is close to the bed there is plenty of headroom - unless of course if I set MIN_Z_FOR_HOMING = 100 and my Z_MAX = 100. So you would need to test for this if you want to make it bulletproof, and the above paragraph assumes you do.

I am proposing that the test be
If Z <= Z_MAX - MIN_Z_FOR_HOMING then increase Z by MIN_Z_FOR_HOMING.
If Z = Z_MAX do nothing
else increase Z to Z_MAX

then there should be some clearance for any situation except Z = Z_MAX, and it should not affect the current operation for use with ABL - that I can see.

So technically, there are a few holes in your implementation because apparently you are only thinking about the implication with ABL and no print being on the bed. If the test is as I propose then the only issue is Z = Z_MAX, which cannot be handled but would be extremely rare.

Blue-Marlin, 2016-03-18:
@ruggb 
You always imply z is known. What is homing for? If you need to raise z before you can home x/y, simply do it. If you need to home z there is only one information we can trust - we are somewhere in between z-min and z-max. ...
We are running in circles.

ruggb, 2016-03-18:
no, I think your process of checking where Z is and acting according to that information indicated that I am not implying I know where Z is -- In fact you are.
OK unless I implement ABL, this whole thing is useless to me - or anyone else who doesn't implement ABL - so why bill it as global??

It isn't, since it has nothing to do with homing for any other reason unless you know where Z is, which as you stated you do not, and if you did you would do other things, like implement my proposal.

So it is MIN_Z_FOR_HOMING_ABL

thinkyhead, 2016-03-18:
When there is an object on the bed you should remove it. If you wanted to resume printing on an incomplete object, you would never want to home. Not X, not Y, not Z. Even if the nozzle avoids the object, the gantry will crash into it.

thinkyhead, 2016-03-18:
Please start a new topic. Make it a feature request. Describe what you want, CLEARLY. We will deal with it when we have more time.

thinkyhead, 2016-03-18:
P.S. The min z for homing setting was never conceived to avoid anything except the bed clips, and has never had anything to do with avoiding another print on the bed.

ruggb, 2016-03-18:
Then I submit that you have a problem with semantics.
It should not be called global and should only be referred to and used in the context of ABL.

thinkyhead, 2016-03-24:
> It should not be called global and should only be referred to and used in the context of ABL.

Sure enough. We've been saying the same thing around here for a while. It's not the only name we don't like, either. There are other issues that just keep distracting us. But we'll get around to it.