MarlinFirmware/Marlin #3061: LCD print timer does not stop

jbrazio, 2016-03-01:
When I print using serial the LCD timer starts but it never stops even when motor timeout kicks in a motors are turned off. Does anyone else experience this issue ? I'm using Marling RCBugFix.

Blue-Marlin, 2016-03-02:
What timer?
At the status line?
The messages there are send by the host/slicer. Just add a final `M117 Ready` to your g-code files.

When printing over the serial line Marlin has no way to know about starting or ending files and therefore has no own timer for that.

jbrazio, 2016-03-02:
Not the status line, the timer which is activated when SD printing..

I see print_job_start_ms being started with M24, M31, M32 or with M109. When serial printing I don't explicit tell Marlin to start the print job timer on my g-code, it automatically does so when it sees the M109 heating the extruder. I see four possibilities for it to automatically detect my print has finished:

1. Disable the auto start in M109 if serial printing
2. By timeout #1 - the buffer/planner is empty for some time
3. By timeout #2 - the motors have gone into no-power mode
4. By seeing the extruder go cold for instance with the M109 S0

thinkyhead, 2016-03-02:
This is a longstanding feature. When your print finishes, the LCD will read "1 hour 3 minutes" but the little clock will continue to tick after the print has finished. I like the idea of having the timer stop when all the extruder heaters have been turned off. So, not just on any `M109 S0` but only the one that turns off the last heater.

jbrazio, 2016-03-02:
Exactly it must for sure have into account all extruders.

Maybe we should also pay some attention to M104 as a start/stop for the print timer ? Don't know if people may start their g-code with M104 but for sure people may end their g-code with M104 has the way to turn off the extruders in a non blocking way.

Blue-Marlin, 2016-03-02:
Ahh, printing from SD.
Isn't closing the file a good indicator?

jbrazio, 2016-03-02:
Not from SD.. from serial.

Blue-Marlin, 2016-03-02:
> When I print using serial the LCD timer starts but it never stops

> Not the status line, the timer which is activated when SD printing..

> Not from SD.. from serial.

???

jbrazio, 2016-03-02:
Each sentence has it's own context..
Issue: When I **print using serial** the **LCD timer starts** but it **never stops**

@Blue-Marlin
> What timer?
> The messages there are send by the host/slicer.

@jbrazio
> Not the status line, the timer which is activated when SD printing..

@Blue-Marlin
> Ahh, printing from SD.

@jbrazio
> Not from SD.. from serial.

@thinkyhead nailed with "_This is a longstanding issue. When your print finishes, the LCD will read "1 hour 3 minutes" but the little clock will continue to tick after the print has finished._"

closed 2016-03-25 by jbrazio