MarlinFirmware/Marlin #3062: Perimeter not surrounding printed item. Cuts through the part

wolfstep, 2016-03-01:
Attached is a picture of the perimeter which is set to 5mm from object that goes around part on 3 sides however on the 4th side it goes right through the middle.
I have checked all my parts and they are as they were prior to me upgrading from 1.0.2 rev. I decided to try your rc3 and it is flacky. 
I am open to suggestions. I tried different part, ran them thru netfix, cold started everything including the laptop. Help!
[image]

Blue-Marlin, 2016-03-01:
G-code generation is not a thing Marlin does. It's done by the slicer. So changing your Marlin version can't be the source of your problem.

wolfstep, 2016-03-02:
Thats the odd part, Slicer hasnt been changed. The only update was the new rev of marlin. When I was using the old version, it never happened. I use repetier and have all my various "old" preconfigs which worked fine before. I didnt change the slicer as I had them fine tuned and was deathly afraid to change them in fear of my slices going nuts. The old adage, if it isnt broke dont fix it. The only reason I went with rc3 is everyone was tough-ting its auto-leveling Proust.

Now I am not able to get my z home to work. The up and down button works great, but the home button gives me the "Home X/Y before Z" which I did. x homes, y homes, z just sits there. I can use the down arrow fine though.

brainscan, 2016-03-02:
If the perimeter cuts through the object it's nothing to do with Marlin and probably just that the object is not completely level and so only a very small area is touching the build plate (the area with a perimeter) and the rest is floating so the slicer doesn't give it a perimeter, but as has already been said its nothing to do with Marlin.

thinkyhead, 2016-03-02:
@wolfstep Make sure before you slice the object that it is flat and all the way dropped down to the build plate. To me it looks like your STL has the object tilted slightly prior to slicing.

thinkyhead, 2016-03-02:
> I am not able to get my z home to work.
> I decided to try your rc3 and it is flaky.

Yes, it is known to be flaky. I recommend testing `RCBugFix` before reporting, because it is getting patches that are ahead of RC3. If something has already been patched in `RCBugFix` we've already heard about it, and we don't need to hear about it again.

thinkyhead, 2016-03-02:
> I am not able to get my z home to work.

It's great to have you here, getting involved. All we ask is, please post any separate issues as a new issue. We like to do one topic at a time – the one specified in the title of the issue.

closed 2016-05-04 by jbrazio