MarlinFirmware/Marlin #3063: When using offset for the probe to the extruder, if there is a decimal place the compiler errors

wolfstep, 2016-03-01:
When I was compiling the new RC3 firmware and had a decimal place in the offsets the compiler wont work. It only accepts whole numbers claiming there is no place holder for it. rev 1.0.2 had no issues.
I attached my config.h file for your perusal. Really frustrating. Lucky the small decimal can be rounded up or down, but really should be able to use the tenths column.

[Configuration .txt](https://github.com/MarlinFirmware/Marlin/files/153778/Configuration.txt)

Blue-Marlin, 2016-03-01:
That's exactly the reason for having the warning directly above. 
The c preprocessor can't calculate with floats.

thinkyhead, 2016-03-02:
@wolfstep Bed leveling will not be affected in any perceptible way if the XY position that _actually_ gets probed is off by under 1mm. So just input the whole values that are closest to the actual position.

closed 2016-03-02 by thinkyhead

jbrazio, 2016-03-02:
Just one remark: when deciding which points you'll setup for the auto leveling points, you have to think not about the nozzle position but about the probe position, i.e. when you tell Marlin to probe (10,10) it's not the nozzle that will be centered at (10,10) but instead the probe, that's one of the reasons behind the offset.

thinkyhead, 2016-03-03:
Hopefully we'll have some time to write up better Marlin documentation and gather it all into a good place. The documentation that other firmware projects are producing are putting Marlin to shame.

jbrazio, 2016-03-03:
maybe getting offtopic..
@thinkyhead the wiki feature here at Github is not enough ? Your your idea is to start a MarlinDocumentation and produce real "documentation" using LaTeX and such ?

thinkyhead, 2016-03-04:
@jbrazio Well, our original concept was to put up marlinfirmware.org as a MediaWiki site with a distinctive url marlinfirmware.org. I spent dozens of hours there writing up documentation, mainly working on the individual GCodes and getting a start on tricky topics like configuration, calibration, auto bed leveling, etc. Unfortunately the individual who set up the wiki was inaccessible and unhelpful, so I was the only person who could edit it. Fortunately most of the content was saved, copied to the Marlin wiki here on Github. But it remains incomplete.

I would be happy if we were to simply continue to work on the documentation and fill it out, make it as complete as possible. I don't know how it is that other firmware projects have such an easy time getting these things together. The documentation for [Smoothie](http://smoothieware.org) is particularly nice. I would like to think that Marlin can do just as well, but there is no one really "spearheading" these efforts.

jbrazio, 2016-03-05:
Ok I spent sometime around this and I may have found a solution, I hope at end of today (I'm in UTC) to open a dedicated issue on the bug tracker so we can start the discussion there and build something.

jbrazio, 2016-03-06:
Continues at #3088