MarlinFirmware/Marlin #3068: Issue for hosts with M600 (and possible other commands that behave similar)

foosel, 2016-03-02:
`M600` is basically an "LCD-only" command in that when it is processed, it [waits for the user to interact with the LCD](https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L5364) and blocks the processing of commands in the meantime.

This is very problematic behaviour for hosts, especially since `M600` also seems to show this behaviour if invoked from a file being printed from SD (hence without any chance for the host to know that an `M600` is active). The printer simply stops responding in any way, and there's absolutely no way for the host to know if that is because it just crashed, exploded into flames, or some "LCD-only" command just happens to be active and the user simply hasn't completed the process yet. There is no way for a host to communicate in a meaningful way what is up now and why the print is not continuing (and the temperatures aren't being updated, and the serial line is pretty much dead, and "nothing works" - to quote some tickets I got in similar cases ;)).

I haven't done a deep dive into the code base to look if there might be other commands that similarly block all processing within `process_command` until the user does something (`M0` and `M1` do come to mind), but at the very least the firmware should output some meaningful message to a connected host if that is the case, so that it's clear what's up. Something like

    Waiting for user

The way the implementation behaves now, if users use `M600` in their files and print through OctoPrint, OctoPrint will assume the printer crashed after waiting some time, cancel the print if configured such and inform the user that there was a communication error, since it has no way to know if the printer stopped answering because of an error or just because `M600` can block endlessly and produce no output in the meantime. I could modify OctoPrint to set an internal flag if an `M600` is sent through it, but that would still not solve the problem in case of SD prints, where in the middle of the print stuff would just stop without OctoPrint having any chance to know what's up since it doesn't see the commands that are printed directly from the printer's SD.

As a general rule of thumb, the printer should never stop responding an arbitrary amount of time. We have no heartbeat or anything like that in place, so the only way to know if the printer is still up and the firmware still working is by sending it commands and receiving output. If both stops working out of the blue, a host can only shrug its shoulders and assume the worst.

thinkyhead, 2016-03-03:
Thank you for laying it all out in detail.

There are two problems to be solved here:

1. Allow commands in GCode to pause the printer in a manner that hosts know it. This is needed because multi-color printing with filament change is becoming very common. The only host-friendly way to do filament change today is to have the host initiate the filament change.

I recently saw that [Prusa Research has a tool](https://www.youtube.com/watch?v=vnbQ0YTZnmk) on their website to insert GCode for filament change into existing GCode. It works around the fact that not all firmware has a command like `M600`. But it also appears to be completely dependent on having a controller, so I assume it uses `M0` or `M1` to pause. So the community hasn't caught on to the existence of `M600` yet, and Prusa's tool is also going to be problematic for hosts.

2. Prevent commands like `M600` from hanging in a tight loop that simply blocks the command injector.

One thing that bugs many of us is the occasional place in Marlin where you see things like `while (!lcd_clicked()) idle();` as a way of keeping the printer alive while sitting inside of some arbitrary function. One of the more egregious instances exists in `plan_buffer_line` because you can easily fill up the planner buffer while doing Z moves on the LCD, and if this happens a stack overflow condition inevitably occurs. (We've already patched this, but will soon have a more robust solution that adds nothing to the stack.)

This way of pausing in the middle of a command was used for `M600`. Wisely, @epatel avoided this mistake when implementing the `MESH_BED_LEVELING` feature, by making `G29` a re-entrant command. `G29 S0` does step 1, `G29 S1` does step 2, and so on. In this manner the printer can continue to receive commands in between steps. Obviously, this won't work for `M600`.

So `M600` should really behave more like this:
- `M600` command is received.
- Marlin waits for all buffered commands to finish.
- Marlin sends a message like `M600 Pause` to the serial output.
- Marlin performs moves for the filament change.
- Marlin goes into `WAITING_IN_M600` state. (Yes, states! Let's add states!)
- While in this state, Marlin waits for an LCD click or some kind of RESUME command. (When SD printing, that command could simply be `M24`, but `M601` would be more universal.)
- Upon resuming, Marlin first pulls in the new filament (by the configured length) to purge the old color.
- Here the user needs some opportunity to clean the nozzle. Waiting for user interaction again.
- Marlin then moves XYZ back to the current print position.
- The printer then sends a message like `M600 Resume` to the serial output.
- If SD printing, the job continues.

When printing directly from a host, the host would have to respond to the `M600 Pause` message by pausing for user interaction and sending `M601` if the user decides to continue. It would need to allow the user to interact multiple times, if necessary. The host should only resume printing once it receives  `M600 Resume`.

Is the host is controlling SD printing, it would not need to do anything, except tolerate any (reasonable) length of pause after `M600 Pause` without timing out, and then go back to being watchful when it gets `M600 Resume`.

Obviously the `M600 Pause/Resume` messages could apply then to any command, like `M25 Pause` etc.

foosel, 2016-03-04:
There is still an issue with controllers that do not reset on connect via serial: if a pause is active then, the pause message will already have been sent (to an older session or no-one) and the printer will be unresponsive without indication as to why.

I'd actually - if that is technically feasible - would rather suggest some kind of busy signal that gets sent periodically in such situations (e.g. once per second), to avoid this situation. I'd also not make such a paused message `M600` specific, to allow re-use where necessary. E.g. something like `waiting for input` could also be used for `M0` and `M1`. 

So you'd get something like

    >>> M600
    <<< waiting for input
    <<< waiting for input
    <<< waiting for input
    ...
    <<< ok

It might even make sense to think about some more general `busy` kind of signal, to use in cases such as very long taking `G0`, `G1`, `G28`, `G29` etc situations. Basically, making it visible for the host that the firmware in fact is still alive, only currently busy with something that prevents it from listening or answering, even if the host wasn't connected yet when the problematic command in question was triggered.

Only a suggestion though.

thinkyhead, 2016-03-05:
I will implement these ideas pretty soon and post PRs for further discussion. It will be some time next week, as this weekend I'm focused on updating RC with tested patches and (fingers crossed) tagging a bugfix release.

Roxy-3DPrintBoard, 2016-03-06:
A Busy Signal would be very straight forward to add.  A simple implementation would have two versions of idle().  They both would do the same thing except the one that is trying to keep the session alive would periodically output the "Waiting for input" message.

thinkyhead, 2016-03-08:
The PR #3109 implements a very simple busy signal that posts a message every 2 seconds whenever any command blocks the command queue. In this implementation it sends one message for a long process and another when waiting for user interaction. It opens up the future possibility of having a GCode substitute for an LCD click and a GCode substitute for a value input.

jbrazio, 2016-03-25:
@foosel can this issue be closed ?

foosel, 2016-03-25:
No objections from my side.

closed 2016-03-26 by thinkyhead