MarlinFirmware/Marlin #3070: Any love left for "astyle"?

thinkyhead, 2016-03-03:
Some time ago it was suggested that using automated formatting on the code might be a great idea. But in practical terms it turned out to be a disaster, making the code unreadable, unmaintainable, inhuman. So for all intents and purposes, we have abandoned any enforcement or usage of automated formatting rules.

Nevertheless, the [Coding Standards documentation](https://github.com/MarlinFirmware/Marlin/wiki/Coding-Standards/592ec7f019113c33833f8cee1318999022f7b8ed) that we put together some time ago was replaced with a blurb asserting that [astyle is the new mojo](https://github.com/MarlinFirmware/Marlin/wiki/Coding-Standards). I would like to revert this page, restore the coding standards that encourage contributors to pay attention and put Coding Standards into the main wiki sidebar links for easier discovery.

Any thoughts, objections, additions…?

lrpirlet, 2016-03-03:
@thinkyhead.
 It took me ages to find that set of instructions to contribute... What about having a paragraph titled "contributing" in the readme file... The content of that paragraph could be just a link to the wiki, everyone could then set his preferred formatting program to match the guidelines...

Beside the coding recommendation, one could find some explanation about that, very, very, complex setup... (RC, RC1, RC2, RC3, RCBugFix but also the MarlinDev that is closely tied with Marlin even if I fail to make an easy connection...)

Today, if I find a bug (that is: something that goes against the specs) or an enhancement (we want Marlin to protect the user from burning the house) I do NOT know against what to submit the patch, WHO will apply it and WHEN. History shows that the decision to integrate a change is vastly subjective (function of a subject rather than from the object being submitted).
It would be so nice to have a common definition of what is a BUG (is it a bug when it fails MY expectations?)

In My Humble Opinion, the lack of direction for contributing is part of the explanation to the time it takes to "publish" a stable version of Marlin... The "solutions" taken to "help" with that problem only made it a lot more difficult...
In 3 questions: What are we trying to accomplish?  What change should we accept? How do we measure that the change aims to the goal?

jbrazio, 2016-03-03:
I do find a well known coding style a must have for any project.

Roxy-3DPrintBoard, 2016-03-03:
> Nevertheless, the Coding Standards documentation that we put together some time ago was replaced with a blurb asserting that astyle is the new mojo. I would like to revert this page, restore the coding standards that require contributors to pay attention and put Coding Standards into the main wiki sidebar links for easier discovery.

> Any thoughts, objections, additions…?

Please --DO-- revert this page.   

But let's talk about coding styles.   I'm OK with what ever the majority of people want to do.    

I really don't want to have to do any work to get my code to conform to the specified coding style.  If we could use Astyle without screwing up everybody's forks and branches and get to a consistent style, I would say we should do it.  But the problem is, there is too much history and if we 'clean up' the code it breaks everybody's ability to track changes and fold them into their own personal forks.

The actual 'style' of the code right now is not the problem.  Right now the biggest thing we can do to clean up the code base is split up and modulize the files.  

And just for the sake of discussion...   I added this code to do_block_move_to().   I feel very strongly it is more readable the way I have it formatted than what Astyle produced:
~~~~
    #if ENABLED(DELTA)
      feedrate = XY_TRAVEL_SPEED;
 //
 // Should we break up the move because it is too big ???
 //
      if (abs(x-current_position[X_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	||
      	  abs(y-current_position[Y_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE	||
      	  abs(z-current_position[Z_AXIS]) > MAX_MOVEMENT_WITHOUT_BREAKING_UP_MOVE ) {
  		do_blocking_move_to( current_position[X_AXIS] + (x-current_position[X_AXIS])/2.0,
  				     current_position[Y_AXIS] + (y-current_position[Y_AXIS])/2.0,
  				     current_position[Z_AXIS] + (z-current_position[Z_AXIS])/2.0);
//
// At this point, we are halfway to where we want to be.  Do the other half of the move.
//
		do_blocking_move_to(x, y, z);
		feedrate = oldFeedRate;
		return;
      } else {		// No...  It is small enough to do in one piece.
      	destination[X_AXIS] = x;
      	destination[Y_AXIS] = y;
      	destination[Z_AXIS] = z;
      	prepare_move_raw(); // this will also set_current_to_destination
      	st_synchronize();
      }
    #else
~~~~

thinkyhead, 2016-03-07:
> Today, if I find a bug (that is: something that goes against the specs) or an enhancement (we want Marlin to protect the user from burning the house) I do NOT know against what to submit the patch, [...]

We can try to explain some of this in the README.
- Today we have two code-bases going.
- **MarlinFirmware/Marlin** is where all release versions of Marlin are found. It has branches like `1.0.x`, `RC`, `RCBugFix` (and experimental branches).
 - The `RC` branch is where tagged release candidates live. Right now we're going through a series of release candidates until we feel confident we've eliminated the most serious and obvious bugs. Then we will have a formal 1.1.0 release.
 - The `1.1.x` branch will be added when the formal Marlin 1.1.0 release is ready.
 - The `HEAD` of `RC` is always the most recent release candidate.
 - `RCBugFix` is where we fix bugs and test patches bound for the next RC. The branch starts with the last published `RC` and then includes whatever patches have been deemed ready.
- **MarlinFirmware/MarlinDev** is where new development is occurring for Marlin 1.2.x and beyond.
 - There's a `master` branch there, a `dev` branch, and some others.
 - I've been making PRs against the `dev` branch for a while, just about whenever I patch something in `RCBugFix`.
 - The code between `RCBugFix` and `dev` has diverged in terms of organization, but the body of individual files is still mostly the same. I can still do side-by-side comparison between them and see where they don't match up, and we will have to keep doing this on occasion to make sure they both get all the patches.

> It would be so nice to have a common definition of what is a BUG (is it a bug when it fails MY expectations?)

Maybe. It should be discussed with others whenever you're unsure. If you have a fix in mind, it's always best to post a PR with the suggestion. Even though it can be tedium sometimes, the process of working with others to figure this stuff out can often be pretty valuable.

I find that keeping an active discussion around open PRs and active issues keeps things always moving forward. Each day I pick through issues to see what I can address. Every issue presents evidence of some flaw or gives light to some new direction. Even the naïve questions of new users reveal ways in which Marlin can be improved.

It's true that Marlin doesn't have a single "direction" and you can see some ways that Marlin occasionally fragments. If one were to start completely over today to make an AVR firmware, I feel like only a madman would make it the way Marlin is constructed. And of course when you look at other firmware like Repetier, aprinter, and Smoothie you do find more thoughtfully designed systems and cleaner looking code.

So we could just start over. But then why not just pick another firmware that has a bigger head start?

My personal approach to Marlin has not been to try to overhaul its structure in a massive way, but to just keep making incremental improvements to the existing structure – understanding that this mundane work has to come first. Bug reports are always arriving and someone must investigate them and produce patches, or they pile up. In the meantime, each bug we clear brings us closer to a final release.

As an open source communal project with a very _ad hoc_ system of contributions, Marlin has been highly resistant to any kind of top-down structure. Perhaps that's just the nature of non-organizational projects where there is literally no hierarchy at all.

There's still a lot of work to do to make Marlin as pretty and organized as it can be. But it's getting there! Yes, Marlin's roots in Sprinter and grbl are still very obvious. The acceleration code is just about the same as it's always been (recent changes notwithstanding). The C-like use of global functions and global variables is kind of primitive. But over the last 18 months the code has undergone some pretty big changes, mostly breaking it up into smaller parts — splitting up languages and pins, for example. An amazing number of bugs have been patched since Marlin 1.0.0. And coding standards have been pretty faithfully applied in new contributions.

I guess what I'm saying is that Marlin continues to move forward in spite of not having a "goal" other than to be awesome. Developers are trying things – hence the whole new organization of MarlinDev and the library format and all that. Some fragmentation has happened due to developers having a different emphasis for the "direction" of Marlin. But the main thing is the value of the work and what it shows to be possible. With Arduino 1.6.7 we can finally organize more nicely in subfolders, and I know that will help a lot — we might even go with a new folder structure in 1.1.0 just because we can!

paulusjacobus, 2016-03-07:
@thinkyhead the approach to have incremental improvements and features is a very agile approach which has worked very well for Marlin. Sofar my fork work great with very minor issues. I see marlin as firmware to drive the steppers and extruders to perform the basics. LCD and other add ons are nice but not necessary and they are secondary features. We can use the driving slicers like cura or repetier as information providers. The biggest issues that we have is a huge array of machines that use marlin. In my opinion, that is another reason to keep it basic. Next to this, i noticed a huge amount of issues that are identical like configuration of the z probe or the pins or lcds. The users are not knowledgeable enough to understand the issue or how to troubleshoot so lots of time is used to support users with explaining how to troubleshoot. Should not we put effort in a troubleshooting guide, minimum requirements (what is supported and what not i.e. Aduino ide 1.0.5) and a configuration guide.
The biggest advantages we can achieve are, not in priority order:
1. Actuator/Position feedback so marlin always knows where the printer head is (this is big and requires additional encoders and hardware, but would increase usuability)
2. Software simplification and algorithm/design documentation so we can follow up when a bug is discovered but the programmer has dissapeared (like auto bed level was broken for so long because nobody could work it out how it worked)
3. Harness programmers, testers, writers. 'We' are here but it is hard to contribute without a simple way to publish the wiki pages. May be a guide how to contribute?
4. Replace sinus/cosinus routines with new algorithms that are faster (smart ways to improve, this requires talent). May be we need an experimentation corner to try out a specific topic that we want to explore.
5. A Think tank area were we can discuss what is a bug and what is not? Bugs are undesired behaviour of a feature that is different from its specification. That is what don't have: specs hence bugs are difficult see 2. That's why we have yes no discussions about M110 etc. maybe a better specification of the g codes?

Anyway that is just my view based on 3 years of tinkering with my mendels. I already anticipate the opposite arguments... 😀 That's part of open source.

thinkyhead, 2016-03-08:
> Should we not put effort into a troubleshooting guide, minimum requirements (what is supported and what not i.e. Aduino ide 1.0.5) and a configuration guide?

I agree, and would love to write up guides on various other topics as well. Please check out #3088 and join in the effort! The documentation for Marlin has been moving around a lot over the last year, but I think we've hit on the best idea for a solution. The documentation will live in the Marlin repository but also be accessible as a website with nice navigation.

thinkyhead, 2016-03-08:
> Actuator/Position feedback

For the time being the stepper mechanism provides the high accuracy we need. But features like this can certainly be added any time. The question is, what kind of position feedback can be given, and how accurately? Could Marlin be used with DC motors that have positional feedback, for example?

> Software simplification and algorithm/design documentation

I'm actually doing a lot of note-taking around the low-level routines in Marlin, and even producing rough flowcharts. The code around stepper.cpp and planner.cpp is itself pretty thoroughly self-documenting, and it's worth reading through to understand how everything is coordinated. There are papers referring to theory around acceleration, and we can certainly write up exactly how Marlin does it (with the same core code as Sprinter and just about every other firmware out there). We need to improve the `ADVANCE` code, and I'm finding that comparing Marlin to other firmware is helpful in that effort. But anyway, yes, I think we should document Marlin down to that level – explaining just how each unit of the program works.

> Harness programmers, testers, writers. 'We' are here but it is hard to contribute without a simple way to publish the wiki pages. May be a guide how to contribute?

We are working on these as well, with plenty more to do. I posted an improved [Home page](https://github.com/MarlinFirmware/Marlin/wiki) and a more up-to-date [Contributing](https://github.com/MarlinFirmware/Marlin/wiki/Contributing) page to the wiki last night. As we work on the process around documentation in #3088 we can post more information about it.

> Replace sinus/cosinus routines with new algorithms that are faster

The Delta routines require some unavoidable trigonometry for accuracy. That's probably unavoidable. A Mega2560 can run a Delta pretty handily, but if you add a Graphical Display it starts to push beyond the processing limits and you have to start reducing `DELTA_SEGMENTS_PER_SECOND`.

The G2 / G3 plan_arc code uses `sin`/`cos` to calculate points on the arcs. While it could probably be reworked to use some form of the [Midpoint Circle Algorithm](https://en.wikipedia.org/wiki/Midpoint_circle_algorithm) there are some good reasons to keep using the trig functions — mainly to preserve high positional accuracy.

> experimentation corner

I like to see things happening in other people's forks, especially experimental things. For that I recommend starting a new branch from `MarlinDev/dev` and messing around. If you post a PR against `dev` then it makes it easy to see all the code changes. I've been posting a lot of "experimental" PRs lately against `RCBugFix` because the layout is still familiar to most people. So I can show the reference code for new ideas. If you check the Pull Request list you'll find some of these hanging around, and they could use some discussion and testing!

> undesired behaviour of a feature that is different from its specification

Well, we try to get all the commands working just as they are documented and expected to work. The GCode descriptions that I started writing up try to explain a little about what Marlin does internally in response to the commands. In general, when it comes to interfacing with hosts, running print jobs, etc., the behavior of Marlin should be very much like other firmware. The movement tuning parameters are also much the same, so you should see similar acceleration and everything.

Marlin's trying to balance a lot of things right now. There's ongoing development that introduces new code, mostly happening in the Dev branch. There's bug-fixing with patches to existing code, mostly happening in `RCBugFix` but to be transferred over to MarlinDev. There's code cleanup, which involves writing comments, fixing indentation, breaking up code into files, etc. There's code overhaul, which involves more radical changes like rewriting the GCode parser, wrapping up planner and stepper into singleton classes, updating the build process, etc. All of these tasks bring us incremental improvements, but they also point the direction for further improvement and organization. So we just have to take them bit by bit….

> Think tank area were we can discuss what is a bug

Well, we do that on a case-by-case basis using the Issue Queue, and that works out well. The Issue Queue really does make an effective "think tank" allowing us to deal with topics in their own little boxes.

paulusjacobus, 2016-03-09:
@thinkyhead that's really great news and I have seen the new Wiki on Marlin which will help to reduce the number of raised issues by users. Still a bit difficult to find but we can put some hyperlinks on the main page. I have edited some of the configuration/G-code pages to highlight the relationship between Auto bed leveling, G28, G29, and the servos. I have finally a good feeling that new entrants and less tech savvy users can actually deploy/configure Marlin and overcome the steep learning curve.
We still lack a direction in terms of Objectivs and prioritized bugs (things that break Marlin or keep user groups from using Marlin like Corexy). Maybe a short status page on where we at with (by release)?

closed 2016-05-04 by jbrazio