MarlinFirmware/Marlin #3075: M106 P1 - Support for Fan 1 missing?

bmarl, 2016-03-03:
It seems that while covered in the pinout of boards (i.e RUMBA) that have two fans, the actual code does not evaluate the P parameter and will only ever control fan 0.
`
#define FAN_PIN             7
#define FAN1_PIN            8
`

`  /**
   * M106: Set Fan Speed
   */
  inline void gcode_M106() { fanSpeed = code_seen('S') ? constrain(code_value_short(), 0, 255) : 255; }
`

bmarl, 2016-03-03:
If support for Fan 1 is not planned, the entry in the GCode table on [reprap wiki](http://reprap.org/wiki/G-code#M106:_Fan_On) the  should be adjusted

jbrazio, 2016-03-03:
Actually `FAN1_PIN` is never used on any Marlin code either than `pins_*.h` files.

Blue-Marlin, 2016-03-03:
FAN_PIN is used as the part van. Usually one of them is enough. So until now there is no P parameter in Marlin. Feel free to change the wiki - it's a wiki. Fell also free to implement P.

If you have more fan outputs you can use them as EXTRUDER_?_AUTO_FAN_PIN or CONTROLLERFAN_PIN (Configuration_adv.h)

bmarl, 2016-03-03:
@Blue-Marlin Thanks for the hints on the advanced config, sadly it does not do what i need. I would take you up on the self coding, but my coding skills are 30 years out of date ;)

jbrazio, 2016-03-04:
@thinkyhead please mark this one as a feature request.

thinkyhead, 2016-03-06:
#3086 is how we would generally need to implement multiple fans support.

jbrazio, 2016-03-25:
@thinkyhead with the release of RC4 this issue should be closed.

closed 2016-03-26 by thinkyhead