MarlinFirmware/Marlin #3076: kill() behaviour changed with USE_WATCHDOG

Blue-Marlin, 2016-03-03:
With activated USE_WATCHDOG the behaviour of kill changed. You can't recover a Arduino Mega with the reset button or the serial reset any more. It needs a power cycle to recover.
This is caused by the endless loop of kill(). The watchdog_reset() is not called any more. The watchdog bites 4 seconds after the kill. The processor resets and recovers with the watchdog flag set. The used Arduino bootloader then does not work as normal (at least for Arduino Mega) and also does not clear the watchdog bit. So the next reset again has the watchdog bit set - and so on.

To get the former behaviour of kill() i propose calling watchdog_reset() in the final loop of kill().

The watchdogs purpose is to protect us from __unexpected__ endless loops. The kill loop is expected - we did everything software can do to switch of heaters, motors, power supply, interrupts, ... . Even USE_WATCHDOG in the WATCHDOG_RESET_MANUAL flavour uses kill() instead of the hardware watchdog reset.

Roxy-3DPrintBoard, 2016-03-04:
> To get the former behavior of kill() i propose calling watchdog_reset() in the final loop of kill().

Does this change make sense for the Release Candidate?   I'm pretty convinced that the behavior you are describing is a 'bug'.  And the change (or bug fix) is very easy to understand and very limited in scope where it can even come into play.  But this is a very serious change that might warrant a longer test cycle?  

(I'm just expressing a concern.  Let's see what other people think.  It definitely needs to go into MarlinDev.  I'm just asking the question of whether this is too serious of a change to put into the Release Candidate.)

jbrazio, 2016-03-04:
This is clearly a Bugfix for a bug that will now potentially affect now all RC users with `USE_WATCHDOG` enabled by default.

closed 2016-03-07 by Blue-Marlin