MarlinFirmware/Marlin #3079: M23 with long file name

WillSmithInChina, 2016-03-04:
when i get the long file name and select the file , it occured ：
">>>M23 /quick-test.gcode
SENDING:M23 /quick-test.gcode
echo:Now fresh file: /quick-test.gcode
open failed, File: quick-test.gcode." 

when i get the short file name and select the file , it works fine as following ：
">>>M23 QUICK-~1.GCO
SENDING:M23 QUICK-~1.GCO
echo:Now fresh file: QUICK-~1.GCO
File opened: QUICK-~1.GCO Size: 2778
File selected"

so i think this point is needed to improve, is that right ?

thinkyhead, 2016-03-05:
For the convenience of humans, it would be very nice to have this option. I've worked on this section of the code before, dealing with long filenames, so it makes sense that I should add this feature also.

Implementing support for long filenames has been considered redundant from the perspective of host software, which is recommended to interface with Marlin. The normal procedure for getting a file listing is suggested as…
- Host requests a directory listing
- Marlin always returns the short names in a list – and the host is expected to cache this
- Host retrieves the long name for each short name only as needed (on first display)
This is actually documented, but its use hasn't been emphasized, so as far as I know no hosts yet do long filename listings of SD Cards connected to boards with Marlin on them.

jbrazio, 2016-03-05:
In order to be backward compatible nothing should change on the current behavior except start accepting long file names in M23 (as well as short).

thinkyhead, 2016-03-08:
Quite!

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

closed 2016-05-04 by jbrazio

xk, 2020-04-23:
Hey guys, wake up! We're well into the 21st century... 8.3 is a thing of the '70s, and MSDOS came out in 1981. Just sayin'...

http://elm-chan.org/fsw/ff/00index_e.html

thinkyhead, 2020-05-01:
@xk — As an open source project we welcome your expertise and code contributions to get things moving forward. Everyone who contributes is a volunteer doing the work for fun and education. Keep in mind that we aim to remain compatible with Arduino / AVR / RAMPS so we need to preserve the old stuff even as we adopt the new stuff.

If a new 32-bit-only platform is what you're after, you can't do much better than RepRapFirmware, and David Crocker may already be well ahead of us in this area.