MarlinFirmware/Marlin #3080: Software Endstops are moving with G92

boaschti, 2016-03-04:
Hi,
I'm using Ramps 1.4 with Marlin 1.1.0-RC3. I've got endstop switches on X- Y- Z+. The travel limits after homing are for X=0-350, Y=0-370 and Z=0-137. I use the machine also for CNC applications. If I set the workpiece zero point with G92 X0 Y0 Z0 I've got a few problems. First one is that the software endstops are shifted and its possible to crash at X+ Y+ (Z+ has a hardware endstop). The second Problem is that I cannot move less than Z0 (But it's th minor problem, it can be solved by setting travel limits to Z=-137-0).
Am I right with my theory or is there another G or M Code to move absolute coordinates?

Best Regards S.Weisser

thinkyhead, 2016-03-07:
I think the problem is that Marlin's concept of `G92` is only to set the current position. It doesn't shift the rest of the coordinate system around the new given position (unfortunately). So if your X lower limit was zero, you can still move only where X>=0. This could probably be fixed, but it would require taking some inventory of all that variables in Marlin that constitute the coordinate system, and where limits are being applied. I suspect there are both high-level limits, such as the software endstops, min and max positions, etc., but there may also be lower-level limits, such as the allowable values for current stepper positions.

thinkyhead, 2016-03-07:
If you're not using home offsets for anything, you can use `M206` to get a similar effect to the what you want. It changes the effective home positions and hence the minimum and maximum positions. After `M206 X-350` the command `G28` will set X coordinate to -350. So if you moved `X` to `350`, then set `G92 X0` to make `X = 0`, you should then be able to set `M206 X-350` and get free movement – unless `M206` fails to update the min/max limits. In which case, it probably needs a small patch.

closed 2016-05-04 by jbrazio