MarlinFirmware/Marlin #3081: Possible GPL violations by geeetech

egorse, 2016-03-05:
The geeetech makes the Prusa I3 M201 with their own controller named GTM32 based on STM32.
The FW shipped as .hex only.

Question about source code they answered:

> Hello friend, the firmware for the GTM32 board is not open sourced, so
we'll not publish it. But if you want to reinstall the firmware, you can
download the hex file here:
http://www.geeetech.com/forum/viewtopic.php?f=20&t=17046, tks!

Digging the .hex I think that is Marlin based (or at least some part of Marlin is inside).
Following question on licenses violations got no answer.

Roxy-3DPrintBoard, 2016-03-05:
Do we have any lawyer types here?   Geetech is a horrible company.   I wonder if we could sue to block their sales in the USA?

egorse, 2016-03-05:
@Roxy-3DPrintBoard  I think the Adobe has :D The fonts highly probably theirs (as those goes with u8glib)

jbrazio, 2016-03-06:
http://gpl-violations.org/

Roxy-3DPrintBoard, 2016-03-06:
Oh!   That is a good link!   We should file a report!

jbrazio, 2016-03-06:
> What can I do if I encounter a GPL violation?
> The gpl-violations.org project keeps track of reported GPL violations in its internal request tracking system. If you want to report an alleged violation, please write to license-violation@gpl-violations.org and make sure to enable --verbose mode, i.e. include as much details as possible.

Roxy-3DPrintBoard, 2016-03-06:
@egorse

Geeetech really sucks as a company...  But I think we are going to have to wait before we can bring them to task:

http://www.geeetech.com/forum/viewtopic.php?f=20&t=17046

egorse, 2016-03-06:
@Roxy-3DPrintBoard 
Sure there is no rush. 
My point is - if someone who from Marlin contributors could confirm base on .hex that there is Marlin, than we should communicate to them regarding licenses violations and see - are they willing to communicate back or not (at least they reply to me nothing more than is in issue)?
Depending on their response the following action should be planned.

ErwinRieger, 2016-03-06:
The following commands list all strings in the GTM32_V1.0.4.zip binary that are longer than 11 bytes. Attached is a file with all strings with len >= 5.
[gtm32strings.txt](https://github.com/MarlinFirmware/Marlin/files/160531/gtm32strings.txt)

```
$ unzip GTM32_V1.0.4.zip                                                                                                                                                                                                                                                                 
Archive:  GTM32_V1.0.4.zip
  inflating: GTM32_V1.0.4.hex        

$ ls -l *.hex
-rw-r--r-- 1 erwin users 293219 Feb 18 16:13 GTM32_V1.0.4.hex

$ ./hex2bin  GTM32_V1.0.4.hex GTM32_V1.0.4.bin                                                                                                                                                                                                                                           
Result: 1

$ ls -lh *.bin
-rw-r--r-- 1 erwin users 256K Mär  6 20:44 GTM32_V1.0.4 0x8000000.bin

$ strings -12 G*bin
0123456789abcdefABCDEF
0123456789abcdefghijklmnopqrstuvwxyz
x`x`|`|`~`n`o`g`g
p`p`p`p`p`p`p`p`p`p`p`p`p
p!q!r!s!t!u!v!w!x!y!z!{!|!}!~!
`!a!b!c!d!e!f!g!h!i!j!k!l!m!n!o!!
%$%a%b%V%U%c%Q%W%]%\%[%
%<%^%_%Z%T%i%f%`%P%l%g%h%d%e%Y%X%R%S%k%j%
FIRMWARE_NAME:STM32_3D V1.1 PROTOCOL_VERSION:V1.0 MACHINE_TYPE:JIETAI-PI3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
FIRMWARE_NAME:STM32_3D V1.1 PROTOCOL_VERSION:V1.0 MACHINE_TYPE:JIETAI-PI3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
Line Number is not Last Line Number+1, Last Line: 
checksum mismatch, Last Line: 
echo:Unknown command: 
no find this hotend
                    
FILE TYPE NOT MATCH!
 echo:SD init fail
echo:SD card ok
X MIN Endstop:ON
X MIN Endstop:OFF
X MAX Endstop:ON
X MAX Endstop:OFF
Y MIN Endstop:ON
Y MIN Endstop:OFF
Y MAX Endstop:ON
Y MAX Endstop:OFF
Z MIN Endstop:ON
Z MIN Endstop:OFF
Z MAX Endstop:ON
Z MAX Endstop:OFF
www.geeetech.com
Disable Steppers
Printer Settings
Restore Defaults
Preheat PLA Conf
Preheat ABS Conf
Printer settings
SD Card Removed!
Print stopping...
G1 X0.1 F4800 
G1 X-10 F4800 
G1 X-0.1 F4800 
G1 Y0.1 F4800 
G1 Y-10 F4800 
G1 Y-0.1 F4800 
G1 Z0.1 F4800 
G1 Z-10 F4800 
G1 Z-0.1 F4800 
G1 E0.1 F1800 
G1 E-0.1 F1800 
GTM32 3D Controller
Begin file list
echo:SD init fail
echo:SD card ok
%d Hours %d Minutes
SDIO->STA : %d 
    WELCOME!!!
Store Memory
Custom 1 Conf
Custom 2 Conf
Custom 3 Conf
Custom 4 Conf
Custom 5 Conf
Custom 6 Conf
Start Percent:
End Percent:
Start Height:
Home speed X:
Home speed Y:
Home speed Z:
X Motor Dir:
Y Motor Dir:
Z Motor Dir:
E Motor Dir:
E1 Motor Dir:
Logic level:
SD Loading...
OPEN DIR ERR!
Resume Print
G1 X10 F4800 
G1 X1 F4800 
G1 X-1 F4800 
G1 Y10 F4800 
G1 Y1 F4800 
G1 Y-1 F4800 
G1 Z10 4800 
G1 Z1 F4800 
G1 Z-1 F4800 
G1 E-1 F1800 
G1 E1 F1800 
Firmware 1.0.4 
End file list
OPEN FILE ERR!
GTM32 3D Controllerg
```

egorse, 2016-03-06:
The '''Line Number is not Last Line Number+1''' I've found only in Marin sources (checked bunch of others firmwares)

jbrazio, 2016-03-06:
This is not enough.. I believe it should have a proper disassemble of the binary and prove that most of the functions are behaving as Marlin does.

thinkyhead, 2016-03-07:
> Line Number is not Last Line Number + 1

That is found in [most, if not all firmware](https://github.com/search?utf8=✓&q=%22Line+Number+is+not+Last+Line+Number+%2B+1%22&type=Code&ref=searchresults). It's required by the protocol for hosts to communicate.

Roxy-3DPrintBoard, 2016-03-07:
You don't understand what I'm saying...   At that link...  The source code is posted for all of the different Marlin's they ship on their hardware.    They really are redistributing and publishing the source code they are shipping.

jbrazio, 2016-03-07:
Then we're ok ? Damn I was getting all my rusty pitch forks and stuff from the barn. :-/

egorse, 2016-03-07:
@thinkyhead  did not find in Repetier-Firmware, Sailfish (two versions), Shoothieware, Teacup_Firmware, RepRapFirmware(Duet), grbl. Your results are for Marlin based, Tonokip and Sprinter(based on Tonokip)

egorse, 2016-03-07:
@Roxy-3DPrintBoard yes they do - but not for GTM32

CONSULitAS, 2016-03-07:
@all 
It is Marlin, but with additions!
see here:
* Manual (screenshots of display, e.g. pages 30+41): http://www.geeetech.com/Documents/Quick%20starter%20manual%20for%20M201(1).pdf (found here: http://www.geeetech.com/geeetech-prusa-i3-m201-3d-printer-diy-kit-p-965.html)

It is definitely a Marlin fork. And for every Marlin fork the GPL makes sure, that it will be GPL again. So even if it is based on an other Marlin based fork, it is under the copyright of Marlin. And every addition, too.

@Roxy-3DPrintBoard I didn't find a download link for the source. Could you post it?

I think @thinkyhead is right: All the Marlin source downloads are 200k+. And the binary only is about 100k. 

BTW: How about a honeypot for them (and others)? A unique hidden String that will be compiled in every time. If they update to a newer marlin: Gotcha!

If you say "Oh, they would remove it!": Thiefs are mostly dumb. :smile:

jbrazio, 2016-03-07:
Would an hex encoded string be ASCII grep(able) when compiled ?

AnHardt, 2016-03-08:
```
:1084E000 809A90418E418F804545454949498E8F FC €šAŽA€EEEIIIŽ 
:1084F000 9092924F994F555559999A9B9C9D9E9F 4A ’’O™OUUY™š›œžŸ
:10850000 41494F55A5A5A6A7A8A9AAABAC21AEAF D6 AIOU¥¥¦§¨©ª«¬!®¯
:10851000 B0B1B2B3B4B5B6B7B8B9BABBBCBDBEBF E3 °±²³´µ¶·¸¹º»¼½¾¿
:10852000 C0C1C2C3C4C5C6C7C8C9CACBCCCDCECF D3 ÀÁÂÃÄÅÆÇÈÉÊËÌÍÎÏ
:10853000 D0D1D2D3D4D5D6D7D8D9DADBDCDDDEDF C3 ÐÑÒÓÔÕÖ×ØÙÚÛÜÝÞß
:10854000 E0E1E2E3E4E5E6E7E8E9EAEBECEDEEEF B3 àáâãäåæçèéêëìíîï
:10855000 F0F1F2F3F4F5F6F7F8F9FAFBFCFDFEFF A3 ðñòóôõö÷øùúûüýþÿ
```
That's not one of _my_  utf/display conversion tables.

Some senseless lines of assembler called/hidden in the stepper interrupt could be a hard to find signature. Changing that would hopefully be scaring enough without completely understanding what's going on.

thinkyhead, 2016-03-10:
Even if Geeetech's custom firmware is based on Sprinter or grbl it would still be in GPL violation. But it is truly hard to determine. If they've started using faster hardware they could be just working off of stock stepper libraries and LCD libraries, because any inefficiency in the code would be made up for by the faster CPU. And then you could probably turn out a "clean room" firmware pretty quickly. The hard part is the acceleration code, and things like that. If they made use of any GPL code, even for that, they ought to open up their code just to be on the safe side.

ilvadol, 2016-03-20:
I own this 3d printer (with this board). But lack programming skills. May I be of any use to solving this? Maybe there is a special g-code command that works only with Marlin FW. Or something like that to identify that it's Marlin-based firmware. I know that it's pretty difficult, but what can I do to help?

jbrazio, 2016-05-04:
Thank you for your interest making Marlin better and reporting this issue but this topic has been open for a long period of time without any further development. Marlin has been under heavy development for the past couple of months and moving to it's last mile to finish the RC cycle and release **Marlin v1.1.0**. We suggest you to try out the latest [RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) branch and reopening this issue if required.

agilliam80, 2016-09-09:
I have emails from John Zhao, the creator of the GT2560 board, which is clearly in violation alone. They refuse to release the schematics of the GT2560, yet they advertise it as a RAMPS board. Uses all the same pins and features, uses the ramps 1.4 board file to define pins, and other settings. They told me they cant give me the schematics or .brd file for the gt2560. A few of us wanted to add autolevel, and filament width sensors and other stuff. Furthermore, I also had a chat with them about the m201 and more specifically the board GTM32, they told me it was a ramps replacement, more so like the ramps FD with some improvements.

This company is horrible to deal with. I had a printer that was missing pieces, came with a defective board (gt2560 and refused to give schematics so I could repair it). They gave me a PSU rated at 150 watts that they re-labeled as a 250 watt. The printer caught fire (motherboard and psu started smoking really badly thank god I was home, and saw a flame). When I got everything squared away, I opened up the PSU and saw that it was silk screened as a 150 watt, they added extra solder to the traces to 'help' with current issues. Needless to say it took me 6 months  of shipping wrong parts to me, or incomplete parts, or even flat out defective parts to get something to actually work. 3 heated beds (resistance was severely high causing extreme current draw, then the steppers went out, had to replace the gt2560 board 2x (shipped with rev a firmware on a original 2560 board with a bad setting and nothing worked, and the heated bed melted the fuse on the motherboard (little smd fuse) and carrier.) I have filed a claim with my bank and they got my money back and I have tossed everything that had came to me except a few things (few steppers, extra controller board and lcd.) The smooth rods for the bearings were soo poor, that during the test prints and attempted prints, all the rods are scratched severely and that shouldn't happen for a hardened steel used for bearings. The linear bearings all had BB's falling out. the 450$ printer ended up costing me close to 1000 after buying new parts and quality parts.

Roxy-3D, 2016-09-09:
I have a Geeetech Delta printer.  It is very low quality.   It has numerous issues.    Probably the worst issue is it can't be fully calibrated.   And that is what motivated me to do the High Resolution Mesh leveling system that is used as a foundation for UBL.   I think my best chance of getting that low quality printer to work is to put a High Resolution Mesh under its coordinate system.

I had broken parts but I was able to get them replaced.   Probably because I started a review of the kit here:  http://3dprintboard.com/showthread.php?16024-Geeetech-G2s-Printer-Kit-Review

thinkyhead, 2016-09-10:
Porting Marlin to the Mightyboard is one of those heroic tasks that deserves a serious weekend hack session.

agilliam80, 2016-09-10:
I hope I didn't mislead anyone into thinking that they put marlin on the mightyboard, however I hope someone does someday, I have to say though Jetty (I think) did a fantastic job of sailfish firmware!

agilliam80, 2016-09-10:
The GeeeJunk printer I received was a i3 Pro C Dual with black acrylic and I agree with you Roxy, never buy any printer with acrylic unless it comes from a reputable quality company. The parts I received from the GeeeJunk were very poorly cut. The laser was not at a perfect 90 degree angle to the surface and most corners were cut at an angle, causing the side wall and the front wall to be at an odd angle. Durring assembly it cracked 4 sides. They sent 2 replacements, they came broken, and took 2 weeks to get. Then they sent another set and the tabs were broken off, but decided to try to use them as apposed to wait 2 more weeks. They too broke more tabs off and when I showed Rita at geeetech the pictures showing why it happened, she said it looks perfectly fine to her. I used a carpenters square and put into the corner where they mated and showed her that it was not ok.she said that ALL their printers are like that, and that they will not resend those pieces to make it right. After 6 months of receiving broken pieces, parts that got damaged or broke in shipment due to poor packaging. Oh I have pics that would blow your minds on how I received it. 

They took 6 pecies of individual cardboard panels and taped them edge to edge to make corners. They made an entire cardboard box without doing it the correct way, then taped the top and bottom on and laminated the walls, top and bottom with tape. Needless to say everything in there was broken. 6 months I put up with it. while trying to work with them about getting the printer to actually work, they have a "tech" (who i will not mention the name to protect them) who called me to help me out. They were not even paid to help or do tech support, and was offered slight discounts on their prodcuts. They are also an authorized reseller too. They have helped me out a lot, but i feel bad for them as they are always in the middle of quality problems that they have to sort out for geeejunk. I chocked up my losses on the geetech, got the 480$ i paid back and put towards my FFC wood printer. 

That was a bad mistake too, much better quality however, the build plate is very thin and after 5 heat cycles it warps the build plate so bad it cant be used.. I sent 2 Creator Woods back. Pulleys were severely warped or off center on the X and Y axis. First one came used and damaged. After my second wood creator, I upgraded to the Pro. So much better. I had to send a pro back due to bent Z lead screw (may not have affected prints but when you pay 1350 you want it to be perfect. I got my second pro and had it since. It just died the other day. I cant figure out what happened, but i had a power surge and it turned off for 2 seconds and back on. It did it 3x before I could get to it. I remember turning it back on and it booted up fine, and i upgraded the firmware to 7.8 that FF released. After that the printer didn't boot at all. Had no USB comminucations said unknown device instead of mightyboard. I tried to fix it, ultimately the place i bought it from i purchased a 3 year warranty, and they refunded me the 1350 I paid for it last year and I got the new 2016 model. SOOO NICE. Better than the Replicator 2 and 2x in my opinion.  I paid a lot less for it too, it sells now for 899 and i paid for the 3 yr warranty all under 1000$ so I got 350 + taxes and shipping back from the other printer. Flashforge has been hard to deal with at times but nothing like GeeeJunk. They have 1 person answering the phone, emailing, and doing sales email, and tech support email. Same guy and same voice but uses 4 different names.

I learned my lesson with GeeeTech. I think all i have left literally is parts of the frame, a gt2560 board, 2 or 3 of the 6 steppers, and a few drivers. Oh and the LCD which i also upgraded to a 128x64 LCD. I'm now fixing it and it prints 1000X better using ramps and Bowden extruders instead of their mk8 crap.

ADVICE, STAY AWAY FROM GEEETECH. Since I filed the claim and disputed it on ebay, they have blocked me. Their policy is if you file a complaint or dispute, they will block you! Even after dealing with them for 6 months with their crap. First I was told to deal with John Zhao, then was told to email Lemon, then Rita. Then their US tech support. Then suddenly the US tech support wouldn't help, and I asked Lemon, she told me that they told them to no longer help me I was on my own, then they blocked me. That's when I opened the dispute, filed a claim with my bank. They are also in violation of EBAY, which I have reported to them many times. They have over 15 different stores that they are all geeetech.. Lemon told me they have something like 17 sum stores. She runs 8 of them and another person runs like 5 and athe rest is run by another.

BEWARE, YOU HAVE BEEN WARNED. They also direct you to their stupid forum for assistance, and you post a problem, but their moderator or administrator has to "approve" your post prior to going live on their forum. NONE of my posts EVER got approved (and at that time i had not said anything bad about geeetech). I told them about no one approving the posts and they ignored that.

Sorry for the length I wanted to include as much as I can, I would love to see them get shut down.  Especially stealing marlin firmware and changing it and not leaving marlin name on it. Infrindging upon copyrights and more!

thinkyhead, 2016-09-10:
We fully understand that Marlin is not yet running on the Makerbot boards. Sorry if my call for making this happen was unclear.

thinkyhead, 2016-09-10:
> Especially stealing marlin firmware and changing it and not leaving marlin name on it.

If this is true then there are legal remedies through the GPL.
Perhaps we can talk to the Free Software Foundation about making that happen.

Roxy-3D, 2016-09-10:
Let's not take this thread too far off topic. After this post let's start a separate thread to discuss Marlin on Geeetech if we want to continue this discussion. 

> The parts I received from the GeeeJunk were very poorly cut. The laser was not at a perfect 90 degree angle to the surface and most corners were cut at an angle, causing the side wall and the front wall to be at an odd angle. During assembly it cracked 4 sides. They sent 2 replacements, they came broken, and took 2 weeks to get. 

The laser cut acrylic parts really could be good.  But they need the guy in the factory cutting the parts to make sure the cuts are square.  It seems the laser doing the cutting is doing the cuts at an angle.  And that causes parts to not fit right and crack when you try to get them into position.   It would be so simple to fix, but nobody at Geeeetech has an IQ above room temperature and can figure that out.

> I learned my lesson with GeeeTech. I think all i have left literally is parts of the frame, a gt2560 board, 2 or 3 of the 6 steppers, and a few drivers. Oh and the LCD which i also upgraded to a [...]

I know my Geeetech Delta printer is crap.   But I'm taking the high road.   If I'm going to write Delta software that really helps Delta users make full use of their machine, I need one that I'm ashamed to admit I own.   The Geeetech Delta is certainly in that category.

> ADVICE, STAY AWAY FROM GEEETECH. 

No Argument!

> ... then was told to email Lemon, then Rita. 

You are obviously for real!  Those are the same names I had to deal with.  I was told to send detailed photographs of the flawed parts so they could evaluate if replacement parts should be sent.   I said "Sure, no problem.  I've opened a new section in the RepRap forum at 3DPrintBoard.com and you can see them in the product review I'm doing at this thread."    That got their attention.

> Sorry for the length I wanted to include as much as I can, I would love to see them get shut down. 

I tell you...  This goes against everything I believe about Open Source software....   But if I could identify my code was running on Geeetech hardware, I would make it fail catastrophically.   I'm writing an Auto Bed Leveling system to work around the horribly flawed Geeetech hardware.   But I don't want it to benefit Geeetech.   That is how pissed I am at Geeetech!  I want it to work for everybody else except Geeetech.

thinkyhead, 2016-09-10:
So anyway, Geeetech might not be providing the exact firmware installed on their current machines, but their wiki does provide a download of the Marlin source and guidance on how to configure it. Here is the firmware download link:

http://www.geeetech.com/wiki/images/3/39/Marlin-Marlin_v1.zip

…which is included on their wiki at this link:

http://www.geeetech.com/wiki/index.php/Sanguinololu#Software_Resources

Obviously this is pretty old, but it gives the impression that they are in compliance.
So which of their current models are they not distributing sources for?

thinkyhead, 2016-09-10:
Taking a look at the [GTM32 firmware](http://www.geeetech.com/forum/viewtopic.php?f=20&t=17046) binary for patterns that indicate it may be Marlin…

Roxy-3D, 2016-09-10:
> Taking a look at the GTM32 firmware binary for patterns that indicate it may be Marlin…

Thanks for the suggestion.  I will do that!   Remember...  I'm taking the 'High Road'.   I will spend the time to help Geeetech get into compliance and fulfill the licence requirements.    :)

thinkyhead, 2016-09-10:
Here are all of the interesting strings from a dump of GTM32_V1.0.5:
```
FIRMWARE_NAME:STM32_3D V1.1 PROTOCOL_VERSION:V1.0 MACHINE_TYPE:JIETAI-PI3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
ok

Line Number is not Last Line Number+1, Last Line: 

checksum mismatch, Last Line: 

echo:Unknown command: 

no find this hotend

FILE TYPE NOT MATCH!

echo:SD init fail
echo:SD card ok
X MIN Endstop:ONX MIN Endstop:OFFX MAX Endstop:ONX MAX Endstop:OFFY MIN Endstop:ONY MIN Endstop:OFFY MAX Endstop:ONY MAX Endstop:OFFZ MIN Endstop:ONZ MIN Endstop:OFFZ MAX Endstop:ONZ MAX Endstop:OFFwww.geeetech.comDisable SteppersPrinter SettingsRestore DefaultsPreheat PLA ConfPreheat ABS ConfPrinter settingsSD Card Removed!Print stopping...G1 X0.1 F4800 
G1 X-10 F4800 
G1 X-0.1 F4800 
G1 Y0.1 F4800 
G1 Y-10 F4800 
G1 Y-0.1 F4800 
G1 Z0.1 F4800 
G1 Z-10 F4800 
G1 Z-0.1 F4800 
G1 E0.1 F1800 
G1 E-0.1 F1800 
GTM32 3D ControllerBegin file list
echo:SD init fail
echo:SD card ok
%d Hours %d Minutes
SDIO->STA : %d 

WELCOME!!!Store MemoryCustom 1 ConfCustom 2 ConfCustom 3 ConfCustom 4 ConfCustom 5 ConfCustom 6 ConfStart Percent:End Percent:Start Height:Home speed X:Home speed Y:Home speed Z:X Motor Dir:Y Motor Dir:Z Motor Dir:E Motor Dir:E1 Motor Dir:Logic level:SD Loading...OPEN DIR ERR!Resume PrintG1 X10 F4800 
G1 X1 F4800 
G1 X-1 F4800 
G1 Y10 F4800 
G1 Y1 F4800 
G1 Y-1 F4800 
G1 Z10 4800 
G1 Z1 F4800 
G1 Z-1 F4800 
G1 E1 1800 
G1 E-1 F1800 
G1 E1 F1800 
Firmware 1.0.5 End file list
OPEN FILE ERR!

Resend: "*:<>?|Info screenAuto HomePreheat PLAPreheat ABSCooldownMove AxisNozzle 1:Fan speed:TemperatureAutotemp:   Fact:Filament 0:Filament 1:TemplatesTemplate 1Template 2Template 3Template 4Template 5Template 6Template 7Template 8Custom 1Custom 2Custom 3Custom 4Custom 5Custom 6Filament 2:OFP Max:OFP Min:End Height:Move 10mmMove 1mmMove 0.1mmExtruder 0Extruder 1Extruder 0:Extruder 1:N-PID-P:N-PID-I:N-PID-D:B-PID-P:B-PID-I:B-PID-D:Z offset:Vx-jerk:Vy-jerk:Vz-jerk:Ve-jerk:VTrav min:A-retract:Xsteps/mm:Ysteps/mm:Zsteps/mm:Esteps/mm:XmaxEndstopXminEndstopYmaxEndstopYminEndstopZmaxEndstopZminEndstopMx        % SD---% No SD CardSD Card OK!Filament 3:Pause PrintStop PrintZ offsetÀFÀFÀFÀFÿ÷$û%0.3fError: %s
ok
M112%.3f.gcodeok B:%.1f T0: T1: T2: @:0 B@:0
+,;=[]NANDUSB1USB2USB3PrepareControlMixerSD CardMainNozzle:Bed:MotionVersion   Min:   Max:N-PIDB-PIDCustomMove XMove YMove ZAccel:Vmax X:Vmax Y:Vmax Z:Vmax E:Vmin:Amax X:Amax Y:Amax Z:Amax E:VminX Max:X Min:Y Max:Y Min:Z Max:Z Min:Home X:Home Y:Home Z:Status:Z      %0.2f% SD   %0.0fSD2:SD1:gcodeTurnTrueFalseHigh%0.1f%07.2fY MaxG91
G90
%0.01fT0
T1
ok
ok B:%.1f T0: T1: T2: @:0 B@:0
G28 X
G28
GMTRAMCFSD1SD2OFPX:Y:Z:
```

thinkyhead, 2016-09-10:
And the GTM32 Pro Delta

> %5d.Max.Min

FIRMWARE_NAME:STM32_3D V1.1 PROTOCOL_VERSION:V1.0 MACHINE_TYPE:301 EXTRUDER_COUNT:3 UUID:00000000-0000-0000-0000-000000000000.ok....FIRMWARE_NAME:STM32_3D V1.1 PROTOCOL_VERSION:V1.0 MACHINE_TYPE:301 EXTRUDER_COUNT:3 UUID:00000000-0000-0000-0000-000000000000.ok.......A.A..EEEIII.....O.OUUY.......AIOU.........!..................................................................................................................................................Line Number is not Last Line Number+1, Last Line: .....IyD.1.L|D.4...h..Q..G.F.B........0...K.......... ....K...............0...... ................ ...$...............checksum mismatch, Last Line: ..block->step_event_count:%d..echo:Unknown command: .ok........(....... .F.F..3...... pG.......F8F........no find this hotend.....                    ....FILE TYPE NOT MATCH!....8..F.F(F.....B...$ F2....I......h.. echo:SD init fail...echo:SD card ok.....X MIN Endstop:ON....X MIN Endstop:OFF...X MAX Endstop:ON....X MAX Endstop:OFF...Y MIN Endstop:ON....Y MIN Endstop:OFF...Y MAX Endstop:ON....Y MAX Endstop:OFF...Z MIN Endstop:ON....Z MIN Endstop:OFF...Z MAX Endstop:ON....Z MAX Endstop:OFF...www.geeetech.com....Disable Steppers....Printer Settings....Restore Defaults....Preheat PLA Conf....Preheat ABS Conf....Printer settings....SD Card Removed!....Print stopping......G1 X0.1 F4800 ......G1 X-10 F4800 ......G1 X-0.1 F4800 .....G1 Y0.1 F4800 ......G1 Y-10 F4800 ......G1 Y-0.1 F4800 .....G1 Z0.1 F4800 ......G1 Z-10 F4800 ......G1 Z-0.1 F4800 .....G1 E0.1 F1800 ......G1 E-0.1 F1800 .....Firmware 1.1.0 A ...GTM32 3D Controller.%d Hours %d Minutes.echo:SD init fail...echo:SD card ok.....Begin file list.....................    WELCOME!!!..Store Memory....Custom 1 Conf...Custom 2 Conf...Custom 3 Conf...Custom 4 Conf...Custom 5 Conf...Custom 6 Conf...Start Percent:..End Percent:....Start Height:...Segments/sec:...Diagonal Rod:...Smooth offs:....Effector offs:..Carriage offs:..Delta Radius:...Print Radius:...Home speed X:...Home speed Y:...Home speed Z:...X Motor Dir:....Y Motor Dir:....Z Motor Dir:....E Motor Dir:....E1 Motor Dir:...E2 Motor Dir:...Logic level:....SD Loading......OPEN DIR ERR!...Resume Print....G1 X10 F4800 ...G1 X1 F4800 ....G1 X-1 F4800 ...G1 Y10 F4800 ...G1 Y1 F4800 ....G1 Y-1 F4800 ...G1 Z10 4800 ....G1 Z1 F4800 ....G1 Z-1 F4800 ...G1 E1 1800 .....G1 E-1 F1800 ...G1 E1 F1800 ....OPEN FILE ERR!..End file list....H.G.H.G....I..."*:<>?|.....Info screen.Auto Home...Preheat PLA.Preheat ABS.Cooldown....Move Axis...Fan speed:..Temperature.Autotemp:...   Fact:....Filament 0:.Filament 1:.Filament 2:.Templates...Template 1..Template 2..Template 3..Template 4..Template 5..Template 6..Template 7..Template 8..Custom 1....Custom 2....Custom 3....Custom 4....Custom 5....Custom 6....OFP Max:....OFP Min:....End Height:.Move 10mm...Move 1mm....Move 0.1mm..Extruder 0..Extruder 1..Extruder 2..Extruder 0:.Extruder 1:.Extruder 2:.N-PID-P:....N-PID-I:....N-PID-D:....B-PID-P:....B-PID-I:....B-PID-D:....Z offset:...Vx-jerk:....Vy-jerk:....Vz-jerk:....Ve-jerk:....VTrav min:..A-retract:..Xsteps/mm:..Ysteps/mm:..Zsteps/mm:..Esteps/mm:..XmaxEndstop.XminEndstop.YmaxEndstop.YminEndstop.ZmaxEndstop.ZminEndstop.Mx        ..% SD---% ...No SD Card..SD Card OK!.Pause Print.Stop Print..Z offset....Resend: .....F.F.F.F....%0.3f...%.3f.....gcode..ok B:...%.1f.... T0:.... T1:.... T2:.... @:0.... B@:0...+,;=[]..NAND....USB1....USB2....USB3....Prepare.Control.Mixer...SD Card.Main....Nozzle:.Bed:....Delta...Motion..Version.   Min:.   Max:.N-PID...B-PID...Custom..Move X..Move Y..Move Z..Accel:..Vmax X:.Vmax Y:.Vmax Z:.Vmax E:.Vmin:...Amax X:.Amax Y:.Amax Z:.Amax E:.Vmin....X Max:..X Min:..Y Max:..Y Min:..Z Max:..Z Min:..Home X:.Home Y:.Home Z:.Status:.Z      .%0.2f...% SD   .%0.0f...gcode...Turn....True....False...High....%0.1f...%07.2f..Y Max...G91.....G90.....%0.01f..T0......T1......T2......G28.....SD2:....SD1:....G28.....ok B:...%.1f.... T0:.... T1:.... T2:.... @:0.... B@:0...ok......Error: .%s......ok......M112....GMT.RAM.CF..SD1.SD2.OFP.X:..Y:..Z:..  ..B- ...........................................................................................................................................................................................l...........b...............'................................................................Info screen.......!......SD1:.....R....3..*..$...GTM32 3D Controller

thinkyhead, 2016-09-10:
No smoking gun here. For all we can tell from this, the 32-bit software is all original work.

boelle, 2016-11-02:
since there is no smoking gun and its has been inactive for close to 2 months maybe close?

closed 2016-11-02 by Roxy-3D

5N44P, 2018-06-09:
So now they've published it:
https://github.com/Geeetech3D/Smartto

egorse, 2018-06-09:
@5N44P Thanks for update !!!

thinkyhead, 2018-06-10:
@5N44P — Grbl and Sprinter are also GPL, with the very same obligations as Marlin.

5N44P, 2018-06-10:
@thinkyhead obviously. I was saying that Marlin isn't mentioned at all, but the headers are surprisingly similar (and maybe also the code, but i don't know which parts of the code are from Sprinter and grbl and which may be from marlin)