MarlinFirmware/Marlin #3083: extruder 3 wrong rotation

Letaw, 2016-03-05:
I added ramps to the Board extruder 2 ( was 4). The motor 3 of the extruder at the beginning when filing plastic rotates in the wrong direction. If you enable relay or the flow of plastic at other motors, the motor 3 starts to rotate correctly. Something wrong in the initial task direction DIR of the motor 3.

jbrazio, 2016-03-05:
Just plug the motor connector the other way around and the direction of the motor will invert.

Letaw, 2016-03-05:
The motor buggy is only the beginning. Then after starting other motors this motor rotates correctly. This problem is not in connector.

Blue-Marlin, 2016-03-05:
Would you please be so kind to give us a g-code sequence to reproduce the problem?
A second one, where it works, would also be great.

Letaw, 2016-03-05:
In a few days

retorik, 2016-03-11:
Hello, I'Ve a similar problem with 2 Extruders. It's Necessary I rotate the second Extrude just one step (minimum) in reverse for after it's works Correctly. Without this step everytime the motor rotate in bad way

thinkyhead, 2016-03-16:
#3161 addresses this, and will be merged soon!

closed 2016-03-16 by thinkyhead