MarlinFirmware/Marlin #3085: Heating failed

mtm88001, 2016-03-05:
Using repetier host, heat hot end up to working temperature.  Start a print then select kill print.  Start print again before the extruders cool off below the cold extrusion prevention setting.  Heating fails and requires reboot of Ramps board.  This behavior does not happen in Release firmware.  I have been able to duplicate this behavior on the RC branch as well as the RCBugFix

jbrazio, 2016-03-06:
Do you have the `USE_WATCHDOG` flag set ?

jbrazio, 2016-03-06:
Btw, when you say "heating fails" what appears in the screen or in the serial console before locking up ?

mtm88001, 2016-03-06:
yes I have watch dog set.  I get the error message heading failed on lcd,
as soon as I get a chance I'll re run and copy the console output.

jbrazio, 2016-03-06:
Please have a look at #3077.

jbrazio, 2016-03-06:
But this could be another bug, when possible I'd test it in my printer. Which Marlin version are you exactly using ?

mtm88001, 2016-03-07:
I have tried this on both the RC3 and RCBugFix
[configH.zip](https://github.com/MarlinFirmware/Marlin/files/161305/configH.zip)

thinkyhead, 2016-03-11:
I think I might have a solution. In the function `disable_all_heaters` change these lines:
```cpp
target_temperature[NR] = 0; \
. . .
target_temperature[0] = 0;
```
to
```cpp
setTargetHotend(NR, 0); \
. . .
setTargetHotend(0, 0);
```

mtm88001, 2016-03-11:
Thanks, I'll give it a try and report back

jbrazio, 2016-03-13:
@mtm88001 Now that #3132 has been merge to RCBugfix. Did you had the opportunity to validated if the issue has been fixed ?

mtm88001, 2016-03-13:
not yet, I'll test Monday as soon as I get a chance.  Thanks for looking
into the issue.

mtm88001, 2016-03-14:
downloading the latest rcbugfix, I was not able to reproduce the heading
failed error.  I did a test print and at the end of the print, the z axis
reversed and sent the head into the bed. haven't had a chance to repeat yet.

thinkyhead, 2016-03-24:
Feeling pretty confident about this one. And… RC4 is out!

closed 2016-03-24 by thinkyhead