MarlinFirmware/Marlin #3100: Compilation broken if AUTO_BED_LEVELING_FEATURE is enabled - RCBugFix

WheresWaldo, 2016-03-06:
RCBugFix branch, pulled from zip file 06_MAR_2016.

Since merging pull request #2995 if AUTO_BED_LEVELING_FEATURE is uncommented (enabled) the compilation fails with the following error:

> Arduino: 1.6.7 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> sketch\Marlin_main.cpp: In function 'void run_z_probe()':
> 
> Marlin_main.cpp:1334: error: 'st_get_axis_position_mm' was not declared in this scope
> 
>        zPosition = st_get_axis_position_mm(Z_AXIS);
> 
>                                                  ^
> 
> sketch\Marlin_main.cpp: In function 'void gcode_G29()':
> 
> Marlin_main.cpp:3158: error: 'st_get_axis_position_mm' was not declared in this scope
> 
>                real_z = st_get_axis_position_mm(Z_AXIS);  //get the real Z (since plan_get_position is now correcting the plane)
> 
>                                                       ^
> 
> sketch\Marlin_main.cpp: In function 'void gcode_M48()':
> 
> Marlin_main.cpp:3605: error: 'st_get_axis_position_mm' was not declared in this scope
> 
>      double X_current = st_get_axis_position_mm(X_AXIS),
> 
>                                                       ^
> 
> Marlin_main.cpp:3615: error: 'X_probe_location' was not declared in this scope
> 
>        X_probe_location = code_value() - X_PROBE_OFFSET_FROM_EXTRUDER;
> 
>        ^
> 
> Marlin_main.cpp:3623: error: 'Y_probe_location' was not declared in this scope
> 
>        Y_probe_location = code_value() -  Y_PROBE_OFFSET_FROM_EXTRUDER;
> 
>        ^
> 
> Marlin_main.cpp:3645: error: 'Y_current' was not declared in this scope
> 
>      plan_buffer_line(X_current, Y_current, Z_start_location, E_current, homing_feedrate[Z_AXIS] / 60, active_extruder);
> 
>                                  ^
> 
> Marlin_main.cpp:3645: error: 'Z_start_location' was not declared in this scope
> 
>      plan_buffer_line(X_current, Y_current, Z_start_location, E_current, homing_feedrate[Z_AXIS] / 60, active_extruder);
> 
>                                             ^
> 
> Marlin_main.cpp:3645: error: 'E_current' was not declared in this scope
> 
>      plan_buffer_line(X_current, Y_current, Z_start_location, E_current, homing_feedrate[Z_AXIS] / 60, active_extruder);
> 
>                                                               ^
> 
> Marlin_main.cpp:3656: error: 'X_probe_location' was not declared in this scope
> 
>      plan_buffer_line(X_probe_location, Y_probe_location, Z_start_location,
> 
>                       ^
> 
> Marlin_main.cpp:3656: error: 'Y_probe_location' was not declared in this scope
> 
>      plan_buffer_line(X_probe_location, Y_probe_location, Z_start_location,
> 
>                                         ^
> 
> Marlin_main.cpp:3664: error: 'Z_current' was not declared in this scope
> 
>      current_position[Z_AXIS] = Z_current = st_get_axis_position_mm(Z_AXIS);
> 
>                                 ^
> 
> exit status 1
> 'st_get_axis_position_mm' was not declared in this scope
> 
>   This report would have more information with
>   "Show verbose output during compilation"
>   enabled in File > Preferences.

No other changes made to _Configuration.h_

thinkyhead, 2016-03-06:
Try it now with #3101 merged. The error should be fixed. Not sure if this is due to a regression. The code tested ok in the past.

closed 2016-03-06 by thinkyhead

jbrazio, 2016-03-06:
#3101 fixed the issue, it compiles flawlessly.