MarlinFirmware/Marlin #3120: Bed leveling is going up

arzhurp, 2016-03-10:
When I send G29 after G28, the effector of my Kossel XL is going UP and is blocked against the MAX end stops. 
I'm using a Hall probe. 
What might be the problem ? 
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/166467/Configuration.txt)

Blue-Marlin, 2016-03-10:
Test your endstops with M119 in mid air.
No unused endstop shall be triggered.
Is the probe triggered? If yes invert it.

arzhurp, 2016-03-10:
Unfortunatly I already tried ! :-/

brainscan, 2016-03-10:
Do deltas use the Z_raise_before_probing? If it does then surely it's only trying to do the 20mm move its been asked to. Sorry this setting on anything that homes to max has confused me. Like Blue-Marlin said 99%of the time it's because one of the end stops is triggered.

jbrazio, 2016-03-10:
I don't think so, I had a quick look at the code and DELTA seems to be except from `Z_RAISE_BEFORE_HOMING` flag, could be wrong didn't went too deep.

arzhurp, 2016-03-10:
I found the problem ! I think I wasn't on the same firmware when i tried M119.
Y and X Min was triggering with X and Y max ! So I had to invert them even if they weren't used :
It was like Min and Max were linked ! 
const bool X_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
const bool Y_MIN_ENDSTOP_INVERTING = true; // set to true to invert the logic of the endstop.
You should do something about that ;-)

Thank you so much for all your help !
All the best,
A.P

Blue-Marlin, 2016-03-10:
@arzhurp 
Would the configuration changes proposed in https://github.com/AnHardt/Marlin/pull/15/files have helped you?

arzhurp, 2016-03-10:
Now the endstop are inverted, the effector is going down after G29, touching the bed juste one time and then going up like Z+20 for a Z-5 !

Blue-Marlin, 2016-03-10:
Touching at 0,0 or at the first point to probe?

arzhurp, 2016-03-10:
First point to probe

arzhurp, 2016-03-10:
After a fresh new install, everything works except that the probe is not going down enough...

arzhurp, 2016-03-10:
So, after hours of making adjustments, I finally succeed to do a nice bed leveling but it doesn't seem to be applied ! I've at least 0.5mm of difference when printing, exactly like before, without bed leveling...
My head is gonna explode ... hahahaha
Here is my new config.h
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/167826/Configuration.txt)

paulusjacobus, 2016-03-11:
@arzhurp wonder if you have tried to modify the z probe offset?
`Z_PROBE_OFFSET_FROM_EXTRUDER -15.90  
Not sure if your custom M code M851 is enabled but that might help you modify the offset on the fly?

arzhurp, 2016-03-13:
Hello,
My Z_Probe_Offset is the good one ;-) 
I tried several things but the leveling is still not accurate or not applying.

arzhurp, 2016-03-13:
Edit : I tried a larger bed leveling with larger parts.... During the printing the nozzle can go 2mm above the bed at X90 Y90 and touch it at X90 Y-90

closed 2016-04-28 by jbrazio