MarlinFirmware/Marlin #3125: Development Repository

tamarintech, 2016-03-10:
I have a PR open for 70+ days now with no recent interactions.  It passes tests.  I use the branch on my printer at home.  It hasn't been tagged.  It hasn't been merged.  It seems to have been completely ignored for over two months.

Could I ask for some guidance on what is happening here?

Roxy-3DPrintBoard, 2016-03-10:
Can you provide a link?   Can you tell us what it contained?

jbrazio, 2016-03-10:
It's not only yours, there are a lot of PR piling at dev, but there is a reason behind it, the effort has been all directed to the release of RC4, only today @thinkyhead started today syncing back RCBugFix and dev, PR should start to go in soon.

tamarintech, 2016-03-10:
> It's not only yours, there are a lot of PR piling at dev, but there is a reason behind it, the effort has been all directed to the release of RC4, only today @thinkyhead started today syncing back RCBugFix and dev, PR should start to go in soon.

Excellent!  

If you review the PRs over on MarlinDev you'll see only a few with no tags or any specific attention.  A couple seem to even have a comment trail suggesting needing more work.  Mine is just... sitting there.  No comment suggesting it's being reviewed, when it might be addressed, or that the current work focus of the last nearly 3 months does not include reviewing PRs.

How else can contributors help to avoid code hanging for 70+ days?  That's only going to lead to future frustration as the commits that the team are doing start to obsolete or conflict with other commits.  What's worse is that I've already technically forked this RC release after this initial work went wholly ignored for the first month.  Having a large number of unrelated forks of a project isn't ideal or healthy for an open source project.

jbrazio, 2016-03-10:
I don't want to argue with you about this, just be patient the PR will eventually get in.

tamarintech, 2016-03-10:
I'm really not arguing.  I'm asking how I can contribute back to an accessible open source project that I use with other open source projects to enable creators to create.  What can I do for Marlin?

Roxy-3DPrintBoard, 2016-03-10:
The bulk of the focus right now is on Release Candidate 4.  If you have a Pull Request that fixes a verified bug in RC-4, I can assure you it will get attention within a few days.  But the development pull requests are still visible and can be grabbed by anybody that wants to use them.  If somebody is operating off of the development branch, they know how to grab and merge them into their branch if they need the fix.

tamarintech, 2016-03-10:
> The bulk of the focus right now is on Release Candidate 4.

Excellent!

How do contributors help with that?

jbrazio, 2016-03-11:
Have a look [here](https://github.com/MarlinFirmware/Marlin/compare/1.1.0-RC3...RCBugFix).

We have to put things into perspective, Marlin is nothing without every little contribution each one of it's user base can donate back, so you, mine or any other user PR is really valuable for the project.. even if it does not get approved and merged in (we all can have different opinions).

But here is not a question of either the PR deserves to be merged or not, is a question of timing. Well you must understand that 99% of the user base is currently either using the latest stable release or any sort of RC branch (Marlin has a big fat warning at Dev saying it will break your printer, proceed with caution) this means **bug reporting** will go into RCBugFix, and this also means that the bugs affecting **more people** will have more priority.. specially when @thinkyhead is rushing for a major RC release.

tamarintech, 2016-03-11:
@jbrazio I think we have a misunderstanding here.  The concern is not that it wasn't merged - the concern is that it was simply ignored (inadvertently, I'm sure).  Someone could have tagged it or *something*.  Surely, "having patience" does not extend into 70+ days.

More importantly, rather than defending why something doesn't get merged I am looking for a way to help Marlin.  What can be done during the RC rush?  How do we help?  If contributions are important then certainly losing 70 days of them is a crisis worth addressing the question I have repeated: **How do we help this project?**

thinkyhead, 2016-03-11:
Sorry about the long wait. There were a few contributors, including me, who used to spend the requisite time following up on issues, but they have had busy lives so we've had a shortage of help in that area.

My focus for the past few weeks has been on clearing the scores of issues that piled up while I was on hiatus. Mostly I've been focusing on clearing the `RCBugFix` queue to get `RC4` out for testing, putting in about 25 hours a week to understand issues, do research, and produce patches for `RCBugFix` (and companion patches for `dev`). We're close, oh so close, but there are a couple of lingering issues holding it up.

So I'm just beginning to pay more attention to the Dev branch, mainly trying to sync it up with recent bug fixes in the `RCBugFix` branch so it doesn't get too far behind. Development on `dev` is being pulled in a couple of directions, as it has to keep up with bug fixes and other changes, while at the same time the file structure is being reorganized and the code being perhaps split up into components.

I'm glad when people bring their issues up to remind us. I'd rather we random participants get a tap on the shoulder than the punch in the nose we sometimes receive. I will incorporate MarlinFirmware/MarlinDev#318 in the stream of catching up with old PRs, and I would like to have it in RCBugFix too, so I may roll that PR on your behalf - or if you prefer to do it for the GitHub credit points, please do!

tamarintech, 2016-03-11:
@thinkyhead Please see https://github.com/MarlinFirmware/Marlin/pull/2893

I don't normally care about the github cred, but this would be the forth time someone rolled a change on my behalf (I have accounts as I'm sure most githubbers do - work, professional, consulting and, of course, the tamarintech brand).  Thanks.

If you're busy, on hiatus, or putting in extra hours what can we do to offload some of that?

thinkyhead, 2016-03-11:
Take a look at #3131 and it should be discussed whether this feature is needed. The `pins_MINIRAMBO.h` file specifies the PWM values to set motor currents. Other pins files are expected to do the same. So when would we need to override the pins files to set alternative PWM values? Is it possible for these to change by using different stepper drivers? I think most boards that use this feature have their drivers soldered on.

thinkyhead, 2016-03-11:
> If you're busy, on hiatus, or putting in extra hours what can we do to offload some of that?

Well this project proceeds only by volunteers giving time to the parts they care about. Those who can effectively help others with issues –especially verifiable bugs– are few and far between. And volunteers who live to do support are quite rare. Since I know the code more intimately than anybody else here (except maybe Andreas) I'm one of the few who can quickly resolve bug reports and PRs. So I feel a certain responsibility to keep up with issues.

Anyway, my hiatus from Marlin is over, and you will find me scrubbing issues much more frequently. While my priority is still on `RCBugFix` issues, I will try to deal with a few issues a day in the MarlinDev issue queue, starting from the oldest ones.

Meanwhile, if you truly feel like helping out, we could really use more assistance in the documentation department! #3088

tamarintech, 2016-03-11:
Of course it's driven by volunteers giving time to the parts they care about! :P

Have you considered opening bounties, creating a call for action, and even propositioning high profile Marlin dependent projects/orgs?  There are so many forks of Marlin around that *clearly there are capable contributors* that aren't being leveraged.  There's no shame is begging or even gently reminding companies that they're leveraging Marlin for their success and it's time to contribute back (again in many instances).  Marlin permeates the whole industry.  If it's continued success is completely dependent on a single entity then we need to find that entity some well deserved help! :)

These are a few ways open source projects *just work*.  If docs are what you want then point me at an empty page.

paulusjacobus, 2016-03-13:
@tamarintech you can have a look at the wiki. I went through them and added additional info around the features, they dependencies i.e. G28, G29, M851 and servos etc and how to config them. It is still very thin in terms of knowledge, especially for people who just start with Marlin, they may find the config too complex. So your help would be really valuable.

tamarintech, 2016-03-13:
@paulusjacobus Awesome of you to add that information to the wiki! :+1:

You've seen #3088, yea?  It's new generation docs with some awesome work presented by @jbrazio and it looks great compared to the old wiki.

Whenever this team is ready to receive contributions I'm certainly ready to start writing.

jbrazio, 2016-03-13:
By all means please update the wiki, unless the page as a warning at the top stating "DO NOT EDIT". No changes will be lost that's a guarantee. There is just one caveat, when editing the wiki the preferred idiom  is markdown.

tamarintech, 2016-03-13:
> By all means please update the wiki [...]

See #3088.

You'll notice @thinkyhead greenlighted it and @jbrazio suggested they were going to add additional features and a 'contributing' section.

Let's do this thing!

thinkyhead, 2016-03-13:
The weaknesses of the Github wiki, and the unreliability of hosting Marlin documentation anywhere else has led us inevitably to the concept of #3088, where we host all the documentation in the Marlin repository, but it is web-accessible through the tools that make that possible. The wiki can still be used to develop the documentation, because it's convenient and editable from any browser. But in the long run, we should think of it as writing the official user manual for Marlin.

tamarintech, 2016-03-13:
So is the wiki the source of authority and Proof-of-Concept from @jbrazio not being added?  I don't quite follow the decision made by @thinkyhead's last post.

If the wiki is the authority then there's a significant time suck happening on #3088 and if we want to have the rich features of something like Jekyll then we're wasting time not having a project repository.  I think jbrazio's current work looks great.  It would be cool to see something like that emerge as official.

> But in the long run, we should think of it as writing the official user manual for Marlin.

There's no better time to start than now. :)  Can we get it in an official repository?

paulusjacobus, 2016-03-13:
@tamarintech The proof-of-Concept from @jbrazio
<https://github.com/jbrazio> looks
very modern and enticing, people will immediately know where to go. I'm in!
Just use it as the front page navigation for the existing wiki and enhance
it. How do you eat an elephant? Just bite for bite.

jbrazio, 2016-03-13:
I didn't discuss this with @thinkyhead but my assumption, when everything is ready to be presented to the world as the official documentation source for Marlin, to click on the transfer repository to the MarlinFirmware organization.

Everything official about Marlin must be owned by the project an not a specific member.

But this transferral process is not mandatory nor a current blocking point.

Roxy-3DPrintBoard, 2016-03-13:
> Everything official about Marlin must be owned by the project an not a specific member.
> But this transferral process is not mandatory nor a current blocking point.

The only reason for 'Everything official about Marlin' is we got locked out of our own Wiki.   Other than that, I don't think anybody cares.    If a team can get a good start at better documentation, the project needs that!   And it can easily become part of the project!   Please don't worry about the final home for the documentation.   That problem can easily be solved with everybody nodding their head up and down to the final solution.

Thank You for taking this on!

tamarintech, 2016-03-13:
It sounds like @jbrazio @thinkyhead @tamarintech and @Roxy-3DPrintBoard all like the new documentation.  Tons of newer projects use similar strategies for very rich documentation and I think everyone is agreeing that this would be a very valuable addition.  There seems to be absolutely no reason not to bring it into Marlin officially.

Offshoot and disconnected repositories are a terrible idea.  The documentation should not be a separate product to the wiki and code.  Additionally, the Jekyll approach has quite a few advantages over the github wiki and we should be able to work it with a higher velocity than the wiki.

..... so let's get started! :)

jbrazio, 2016-03-14:
This is so off-topic..
I propose to switch the documentation conversation to #3088.

Roxy-3DPrintBoard, 2016-03-14:
> This is so off-topic..
> I propose to switch the documentation conversation to #3088.

It is good to try to stay closely focused on the thread's topic.   But...   I'm just really glad to see a team forming to help with documentation.   You  can continue to pollute this thread with documentation issues all you want!

tamarintech, 2016-03-14:
> You can continue to pollute this thread with documentation issues all you want!

The topic here was really "I pushed stuff and it's nearly through its first trimester of human life - help."  It hasn't gone much off-topic. :P

The effort going into this thread is certainly greater than the effort going into new docs, so it seems we're still very much on-topic.

Let's go!

thinkyhead, 2016-03-25:
The original PR discussed has been merged, and if no one minds I will retire this thread.

closed 2016-03-25 by thinkyhead