MarlinFirmware/Marlin #3128: Modify firmware to use ADC function

thachdovan, 2016-03-11:
Hi everyone,
I use marlin for ultimaker, I want to modify Marlin firmware to use ADC function, my idea is detect a voltage level and case a pause action to printer. where I can start with my idea.
Thank in advance!

jbrazio, 2016-03-11:
You have to be a bit more explicit with your request..

Blue-Marlin, 2016-03-11:
Define an other extruder. Wait with M109. Maybe you need an alternative temperature conversion table.

thinkyhead, 2016-03-11:
@thachdovan Thanks for asking!

In `temperature.cpp` you will find an interrupt routine around line 1250 called `ISR(TIMER0_COMPB_vect)`. Within that routine you will find a block starting `switch (temp_state)`. This is where we do all of Marlin's ADC reading. It currently operates up to 4 temperature sensors and a filament-width sensor. You should be able to figure out from these examples how to add your own ADC device, which would go into the same place.

thachdovan, 2016-03-12:
@thinkyhead Thank for your support !, I will try and response later.

thachdovan, 2016-03-13:
Dear @thinkyhead ,
I want to make a ADC to detect two levels 0V and 3.3V, if value = 3.3V will pause the machine, the user need adjust rotate the button manually to resume (I use LCD with encoder button). Can you give me in detail about my case. The source so huge, I hope your help to make it easier.
Thank you!

thinkyhead, 2016-03-27:
I wish I could help more — I don't know much beyond what I pointed out. There is the code I pointed out that reads sensors and stores the value, and there is code elsewhere that checks some of those values. I recommend using an editor where you can search many files at once, and follow the breadcrumbs to your solution. Searching on variable names, function names, etc., is the best way to follow the logic. Don't look at "all the code."

Roxy-3DPrintBoard, 2016-03-27:
@thachdovan    Windows is much more limited than Linux for this kind of stuff.   But I pretty much have to be on a Windows machine.   So...  I use:  http://www.wingrep.com/   to search for different text in all of the different files.   That will let you isolate where a given variable or function is being used pretty easily.

thachdovan, 2016-03-27:
Thank you guys!
I will try, this function is so vital with my system.

thinkyhead, 2016-03-27:
I use OS X, so if you need any tips on good tools for that system, I can recommend a few.

closed 2016-05-04 by jbrazio