MarlinFirmware/Marlin #3134: Overshoots Z at bottom

Roo123, 2016-03-12:
1.1 RC3

Spin the dial a few times to set the bed moving down and it will keep on going and going.

Even though the number displayed is less than total Z height and less than distance from current position.

Apparently there's no way to stop it.

In addition the bed doesn't respect the defined Z height but crashes into the floor. Both in the above situation and when under fine manual control.

Tx.

thinkyhead, 2016-03-13:
> the number displayed is less than total Z height

The number displayed should always be less than the total Z height.

> Spin the dial a few times to set the bed moving down

Are you using the **Move Axis > Move 1mm > Move Z** command on the LCD, or the **Babystep** command?

I'm not seeing this behavior in `RCBugFix` so perhaps it's been fixed already…?

Roo123, 2016-03-13:
BigBox Pro - yes, using 1mm intervals as you say.

The number displayed IS always less than the total Z height.

However if you spin the dial a few times then the bed will move further than number indicated would suggest. Sometimes seeming to never stop, leading to nasty crashes.

Moreover, when it is moving like that, turning the dial in the OPPOSITE direction will do nothing to stop it. Only thing is to hit the power.

thinkyhead, 2016-03-13:
We have some fixes already in `RCBugFix` to fix a stack overflow problem, and one more patch is pending to do the same for the E axis. I suggest you try `RCBugFix` and see if the problem still exists. If the problem is fixed, then we know our patches are working.

thinkyhead, 2016-03-25:
Sorry we didn't get to dig into this more before releasing [RC4](https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.0-RC4). We know we have one more round of bug fixes before we release 1.1.0 officially, so if this is still broken we have time to sleuth it out.

Roo123, 2016-03-25:
My fault too: I'd like to have tried RCBugFix, but having only just built my printer (BigBox) two weeks ago am still catching up on many things and it doesn't feel right to start experimenting in firmware yet. Downstream hope to contribute more.

thinkyhead, 2016-03-26:
Re-open if you get this behavior again. We'll be here!

closed 2016-03-26 by thinkyhead