MarlinFirmware/Marlin #3135: Access PID Autotune from LCD

Spoonunit, 2016-03-12:
It would be great if there was an option, perhaps from the "Prepare" menu, to run PID Autotune and read off the results from that on screen. While not absolutely vital to success, having PID tuned for your printer and environment leads to much tighter control of the hot end temperature and more successful first time prints. Once you have these numbers, you can choose to push them into your own firmware build or run them as part of your GCODE starting script. This would be nice to do as part of the commissioning process of a printer, prior to actually hooking up other tools that give you access to a GCODE terminal.

Blue-Marlin, 2016-03-12:
Step a) would be to extend M303 with a parameter to actually use the new values instead only displaying them. That's not too complicated.
Step b) would be a, because of the at least 2 needed parameters, relative complicated/big menu.
Storing in the EEPROM is already there.

An extra cool feature could be temperature dependent PID tables, using different parameters for different target temperatures

Spoonunit, 2016-03-14:
Also nice would be for PID values to be loaded for the tool in operation instead of using a single set of PID values for all tools. These values could be loaded during tool change, something which you can actually do using most slicers already, but might be nice to have the firmware handle it. Also, if the inactive tool drops out of range while different PID values are in operation for the active tool, that probably shouldn't trigger a halt situation (not sure if that happens right now, I'm a little new to this)

WPBack, 2016-03-20:
I am thinking about trying to implement this, but I wonder what extra parameter you think I should use in step A above. U maybe for Use? And also, should it save it to EEPROM or only use it temporarily?

duncanford, 2016-03-20:
I would say temporary, it can always be stored with Store Memory. U seems like a sensible choice; intuitive. As long as this isn't used for other, conflicting meanings. Another idea would be perhaps to make it M3031? Elsewhere it seems like alternate numbers are used to distinguish between **do** and **save**.

WPBack, 2016-03-20:
Temporary is probably right. Another code would probably be the best potion, but can you really have codes higher than 999? I'm not sure since I'm pretty new to this.

AnHardt, 2016-03-20:
`codenum` is currently a int (16bit signed), bu could be enlarged.
But new g-codes are much more problematic than added parameters.

thinkyhead, 2016-03-27:
This feature has been merged into `RCBugFix`. Just enable the `PID_AUTOTUNE_MENU` option. It works, but needs a small tweak – basically an option to cancel or proceed, since it's a long procedure involving heating. For now, just wait for the LCD to time-out back to the Info Screen.

closed 2016-03-27 by thinkyhead