MarlinFirmware/Marlin #3136: why not use endstop as kill switch?

Roo123, 2016-03-12:
Presently the endstop seems to only serve for homing calibration.

I've had the situation where the printhead hits one end of the Y axis such that the belt slips and Marlin loses track of position.

I've then sent the printhead off towards the other end of the Y where the endstop switch is and seen - to my great dismay - the printhead go straight to the end, smash into the kill switch and continue to generally bounce of the wall.

Wouldn't it make sense for all manoeuvres to contain a conditional of the form 

If endstop not reached: THEN X = X + 1 

In normal operation the "if" statement would be redundant, but in odd situations like the one described above it would prevent blind overshoot.

ntoff, 2016-03-13:
it already does stop at the end stops if you enable the option to do so, however I'm not entirely sure how it handles it (whether it knows it's stopped or whether it just stops the motors and the position keeps counting).

https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration_adv.h#L124

It's usually left off so that false triggers from electrical interference don't set it off mid print but I've never had an issue with that in the couple of years of messing about with printers even though my end stop wires and motor wires are bundled together.

jbrazio, 2016-03-13:
@Roo123 did @ntoff suggestion solved your issue ?
If it has been solved, please close this support request, thanks.

closed 2016-03-13 by Roo123