MarlinFirmware/Marlin #3137: Z probe offset issues

UltiFix, 2016-03-12:
No matter what I do, I can't get my printer to listen to the z probe offset parameter.

Say I set it to -0.6mm

After I run G29 it should be -0.6mm lower right? Nope.

EEPROM disabled
marlin
prusa i3

Look here for my configuration.h
http://forums.reprap.org/read.php?406,638503

jbrazio, 2016-03-12:
Z probe offset has nothing to do with G29.. that's auto bed leveling.

Let's imagine you have compiled Marlin with Z probe offset at -0.6 on the config file.
Issue a G28 to home all your axis, when the operation finishes issue a M114 and your Z height shall be 0.6.

UltiFix, 2016-03-12:
Click the home all button, then run M114. It says z -9.40 which is correct right?
But when I print it will start printing .6mm below what the probe says is zero? Do I need to run any special gcode to get it to do that?

jbrazio, 2016-03-12:
No.. something is not right. Please open a serial connection to the printer and copy paste the output of the following commands in order:
```
M501
G28
M114
```

With my printer which as a Z-offset of -0.90 here is the output:
```
>>> M501
SENDING:M501
echo:V21 stored settings retrieved (380 bytes)
echo:Steps per unit:
echo:  M92 X160.00 Y160.00 Z8000.00 E204.15
echo:Maximum feedrates (mm/s):
echo:  M203 X200.00 Y200.00 Z3.30 E200.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X1000 Y1000 Z100 E3000
echo:Accelerations: P=printing, R=retract and T=travel
echo:  M204 P1000.00 R3000.00 T1000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X15.00 Z0.40 E2.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H210 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P19.18 I1.36 D67.42 C100.00 L20
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Z-Probe Offset (mm):
echo:  M851 Z-0.90
>>> G28
SENDING:G28
>>> M114
SENDING:M114
X:71.00 Y:133.00 Z:0.90 E:0.00 Count X: 11360 Y:21280 Z:7200
```

The relevant parts are:
- From M501: `M851 Z-0.90` which should have the same offset and defined at compile time
- From M114: `Z:0.90` which should be the inverse of the z-offset right after the `G28`

UltiFix, 2016-03-12:
Here ya go... I have tried everything. Im totally stumped.
http://pastebin.com/zSTZZKd5

jbrazio, 2016-03-12:
OK you're using Marling 1.0.2
Could you please download the [latest RCBugfix](https://github.com/MarlinFirmware/Marlin/archive/RCBugFix.zip) set it up for your printer and try again ?

UltiFix, 2016-03-12:
I just noticed I had #define Z_MIN_POS -10
Im changing that to 0 now and re uploading, I will then test and see if that fixes it.
...waiting...
Alright, know m114 spit's out the right thing. (Z:0.60)

So am I good now? Should everything be ok?

jbrazio, 2016-03-12:
If now by doing:
```
G28
M114
```
You see your Z height at Z:0.60 all is hunky-dory.
G29 is automatic bed leveling.. it's another thing, what exactly are you trying to accomplish ?

UltiFix, 2016-03-13:
I assumed that doing G29 would take the z probe offset into account, aka lowering itself lower than what the sensor thought should be zero. But thinking about it there is no reason it should.

My goal is when I print it lowers itself by .8mm (I changed my offset from -0.6 to -0.8)

Boom! G28 followed by M114 shows Z:0.80

Wahoo! Thanks!

jbrazio, 2016-03-13:
Great !
`G28` will do exactly what you want. First Marlin find the Z-zero using the inductive sensor (so this Z-zero is referenced to sensor tip) then applies the Z offset to get the Z-zero referenced to the nozzle, as the nozzle is sitting lower than the sensor tip, Z-offset is negative.

If you want to test the offset in order to fine tune try:

```
G28 ; Z height will be the module of z-probe offset
G1 Z0 ; move the nozzle to it's zero by apply the offset
```

On any print job you only have to do `G28`, Marlin does the rest.

closed 2016-03-14 by UltiFix