MarlinFirmware/Marlin #3139: Rotary encoder moves modifies the HotEnd0 temp and freeze Märlin + RAMPS 1.4 + MAX6675 + Thermocouple + FULL GRAPHIC SMART DISPLAY

zeeproduction, 2016-03-13:
Hi everybody. I'm stuck and I really need help. I just signed up on github so if this is not the right place to post this please excuse me and do the appropriate modifications.

I have no idea what else to test to make this thing work.
I use:
Arduino IDE 1.6.5
Arduino Mega 2580
RAMPS 1.4
REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
ADC MAX6675
Thermocouple Hot end (-2)
Thermistor 100K NTC Hot bed (1)

I tested:
Märlin 1.0.2
Märlin Marlin-1.1.0-RC2
Märlin Marlin-1.1.0-RC3
Märlin RCBugFix latest versions
AnHardt Marlin latest versions

MAX6675 is connected on RAMPS pins:
SCK/CLK D52
SO/DO (MISO) D50
CS/SS D53
VDC AUX2 pin +
GND AUX2 pin GND

like in this video: https://www.youtube.com/watch?v=6JciilNAgA8

The best I could get is at start up, without moving anything I get temp readings from both sensors. If I click the encoder the menu swaps forth and back and keep have temps readings. BUT as soon as I turn the encoder any direction the software freezes and it displays weird temps on hotend0 of thermocouple. And this happens in any point any time.

Even if first move I do is to rotate the encoder, the software freezes and displays temp error.

This ERROR, for some reason (what has one to do with the other?) is dependent on ENCODER_PULSES_PER_STEP and ENCODER_STEPS_PER_MENU_ITEM setting in Configuration.h.

If ENCODER_PULSES_PER_STEP has a greater than 4 value the MAXTEMP error is triggered and the Hotend shows temperatures above 300ºC.

If ENCODER_PULSES_PER_STEP is under 4 the MINTEMP error is triggered and Hotend (Thermocouple + MAX6675) displays around -100 - 0ºC.

With values 4 and 1 for these variables, behaves as it should (IMO) but even if I don't do anything after a while it freezes...

I tried even with disabled vars and with -1 and 0 combinations but it didn't worked.

It seems to be important the last position where the encoder last remained because sometimes the software doesn't start until I rotate a bit the encoder.

I'm stuck. I have a thermocouple attached to the hot end and I need to use this. The temp reading with thermocouple 100K are not reliable. I checked with a very accurate infra red gun that reads up to 900ºC and the values read are far from reliable.

And by the way the 100K thermistor is rated up to 125ºC. How it is supposed to measure temperatures above without firing it?
I also realized that pin SCK from RAMPS doesn't have a function on the module in any point. It only reaches the other end of ribbon cable but no point responds to it as a contact. So here is another question: How is it possible for the LCD module and SD card module to work without the clock? I wonder.

With the RCBugFix Märlin versions, the software keeps resetting itself and it doesn't let me do anything but with temp readings or, if it doesn't freezes it behaves as described above.

So, where else should I look? What else can I try to start the thing?
I am working on this printer for a year now. I finally got the ADC MAX a week ago and since then... oh boy...

I think the encoder rotation corrupts the temp readings and therefore a move of the rotary gets put into temp variables and modifies the values of the temp but even if it is so I have no idea how to fix this.

Please give me a hand or at list some directions where to look for.
Let me know if you need me to put some code in here or to send it somewhere.

Thank you so much for what you are doing.

AnHardt, 2016-03-13:
Its working for a lot of machines. The first thing i'd check is, if the MAX6675 is connected the right way. It does not work without SCK. We __have__ a working SCK - somewhere.

AnHardt, 2016-03-13:
When i tested it i used this kind of adapter to split the signals out, without soldering.
[image]
But i don't remember the exact holes.
@Grogyan can you describe how to find SCK?

zeeproduction, 2016-03-13:
I just found them with a countinuity check of a multimeter. I have them splitted correctly now.
On EXP2 (p18 prr BOB) the pins are the first two. Only MISO reacehs the display on pin 3 around SD card.The SCK has no correspondent pin on the siplay board !!!It's weird.
Not the pins are the issue. The weird thing is when I move the rotary encoder it actaully displays weird temps on HE0 and freezes.
I had a terrible week reading and reading and uploading sketches in big volume.I don't know what else to do.
Thanks for your answer.

Roxy-3DPrintBoard, 2016-03-13:
> Have to re-flash a borrowed Mega/RAMPS with RepetierFirmware and i'll have a long day at MakeFair Ruhr in Dortmund Germany after getting up. (Sucht nach dem Marlin Docktor)

makerfaire.com  <----<<<   Very Cool!!!

AnHardt, 2016-03-13:
@zeeproduction 
I'm not sure if you are familiar with.
http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
http://reprap.org/mediawiki/images/7/79/LCD_connect_SCHDOC.pdf
http://reprap.org/mediawiki/images/5/51/RRD_FULL_GRAPHIC_SMART_CONTROLER_SCHEMATIC.pdf

No need to mess with a multi-meter. 

The display itself has it's own SCLK, CS, SID on EXP1

Is there a tiny possibility you could have messed up the numbering of the plugs?
```
1  2 this is the right one
3  4
5  6
7  8
9 10

vs 

1  6
2  7
3  8
4  9
5 10

or

1 10
2  9
3  8
4  7
5  6
```

zeeproduction, 2016-03-13:
@AnHardt - Thanks again. My point was that CLK is on pin 2 on EXP2 but was no point to connect to on the back of LCD module. Anyway this is NOT the issue. I checked 100 times. It's correctly connected.

===
Meanwhile I discovered that I had not enabled SDSUPPORT. I enabled this option and like a wonder I am able to navigate the menu now and get temp readings. I even did a small heating hotend test today. A bit weird the readings but I'll figure it out later.
===

Now I changed to other problem: when I go to Prepare > Move Axis > whatever > Move Y, the movement starts, ends before finishing and it returns to the main screen and I have a MINTEMP error. Things are pretty messed up in my head right now.

Any possible help at this new stage?

Thanks a lot.

Grogyan, 2016-03-13:
@zeeproduction yesterday PR #3112 was merged into th RCBugfix repo
I've tested this extensively over the weekend with the MAX31855, which should be considered the preferred default over the MAX6675 for all applications.
The code is more stable and clearly identifies the key differences between the two.
Pleaae try that

zeeproduction, 2016-03-13:
I was thinking we are going this path but I just bought a 6675 :( I'll have to buy a 31855 to test it and when I'll have it I'll report back, of course. Meanwhile if somebody can tell me where is this interraction between the Y axis and the temp reading... I would really appreaciate it.

You know... I don't have yet the printer so I don't know how it will work and now to buy another module it's not making me happy. I'll take it into consideration but still if you can provide some guide whre to look at...

For now I tested with 1.1.0 RC3 and RCBugFix last versions and it does the same thing. When manually jogging the Y axis, with RC3 it goes to  MAXTEMP error and with RCBugFix goes into MINTEMP error.

Thank you so much for your help.

Grogyan, 2016-03-13:
@zeeproduction please ensure that the thermocouple is electrically isolated from where it is measuring.
I typically use a piece of kapton tape.
A thermocouple produces a very tiny voltage proportional to the temperature based on 2 metals, when a 3rd or an alloy is added this produces some error, or pick up PWM signal from the heater cartridge, or some other AC signal, however I find that electrically isolating the thermocouple produces much more stable results, and is also the simplest.

zeeproduction, 2016-03-13:
Yes, about the temperature, I deeped the thermocouple head in heatsink compund Evercool-420. It reports a temp higher with 3 degrees. It's ok.

Anything about the interraction between the Y axis movement and MINTEMP error? I am clueless of what to do with this.

Thanks a lot for your help.

Grogyan, 2016-03-13:
It is most likely not a problem with the Y axis, but rather the stability of the thermocouple reading.
As mentioned, a thermocouple produces a very tiny voltage and is susceptible to AC signals, especially where the thermocouple is attached to the hot end.
Please try kapton tape, as the thermocouple still sounds like it is shorting out to the heater block, which has a small amount of PWM from the coupled heater cartridge.

Grogyan, 2016-03-13:
@zeeproduction if you are still having issues, please msg me on Hangouts

zeeproduction, 2016-03-14:
I'm sorry. I don't have Hangouts. So I'll follow here with good news.

Well, the head of the thermocouple is fixed to the hot end head with a screw so it cannot be completely isolated electrically.
Yes the casing of the thermocouple is in contact with the head - heater. But then it has a metalic mesh that prevents EM interference. Does it?

I need to share with you two important things.
1. This is how I solved the issue and it appears everything runs without any problems. I used a decoupling capacitor directly mounted on the MAX6675 board at the entry point of the terminal of the thermocouple exactly as the 31855 recommends. Working! I tested polarized 1uF and not polarized 223, 224 and 682. All gave the same POSITIVE result. Everything works. But then I discovered my HotEnd Stepper is not working at all anymore !!! and here comes the second important thing
2. About ten days ago I checked the mechanical functioning and I declared myself satisfied. Everything was working smoothly at a turn of a knob. After connecting the decoupling capacitor I realized three pins from the HotEnd stepper were missing on my connection board with the ribbon cable !!! And the stepper was working 10 day ago in this manner! Yes. Read well what I say. Only because the signals were perfectly aligned and at a slight distance, the stepper was working. But now with the decoupling capacitor the motor ceased to work so YES, Grogyan you are right there is a lot of interference.

Thanks a lot for your help all. Is very much appreciated.

P.S. I did a HotEnd Preheat and the temperatures read are different from the real ones.
1. I used a thermistor placed in front of the thermocouple and was displaying the temperature on the bed temp on screen.
2. I used a temp meter gun (very accurate) and the temperature diplayed by the HotEnd is about 20º more than the double of the real temperature. Examples:
Real        HotEnd on screen                Thermistor
temp       from thermocouple                on screen (bed)
19.8ºC      23º                                     19º
                 71º                                     58º
               100º                                     80º
               150º                                   120º
               161º                                   129º(132º)
               170º                                   140º
83.3ºC    180º                                   148º
103ºC     227º
112ºC     239º
112.8ºC  240º

Is thre some place where to correct the temperature?

Grogyan, 2016-03-14:
I'm glad you have it working.

I'm a bit confused, perhaps it's because i'm basing my assumptions about your hot end with either a J-head or E3D V5/6, not sure what you mean by a metallic mesh.

As for temperatures, what thermocouple are you using,  a typical K type?
A thermocouple shouldn't ever need to have offsets as it already has a natural linear curve, vs a thermistor, hence why there are tables for the various common thermistors.

Perhaps a photo of your setup?

On my home setup I don't use an additional decoulping capacitor for the MAX31855
Vs my old, very old Sells Mendel which i'm sure doesn't have one either and has been running fine for over 4 years, MAX6675.

Sounds like you have other EMI issues to deal with as well now.

zeeproduction, 2016-03-14:
Oh, I'm not ready yet to present pictures :). The configuration is in the title plus the print head that I didn't mention.

The head is an MK8 bought from China and truly I have no idea what thermocouple comes with but I teste with another one that came with the MAX6675 and that is a 1M K type. I tested it separated from the circuit, meaning  away from any other wire and it displays the same temperature. By the way this one behaved the same way the the one that came with the MK8 printing head.

Metalic mesh means the jacket of the thermocouple. 

About the EMI I might have a doubt but I'll make some more test before looking into it. When I splitted the CLK and MISO pins I noticed that the EXP2 port has some chees.metal alloy used in the wires so it might be something in there. I just moved the display and Märlin jumped into MINTEMP error...

Thanks a lot for your help. Maybe tomorrow I'll come with the final result.

All the best.

Roxy-3DPrintBoard, 2016-03-14:
>Well, the head of the thermo couple is fixed to the hot end head with a
screw so it cannot be completely isolated electrically. Yes the casing of
the thermo couple is in contact with the head - heater. But then it has a
metalic mesh that prevents EM interference. Does it?

If it is electrically coupled...  The hot end will be acting as an antenna
and inducing electrical noise.   If you can ground the hot end, it probably
will act as a Faraday cage and shield the thermo couple from electrical
noise.  But we need a EE type to weigh in here.  (I was really bad at
Maxwell's equations.)

>I used a decoupling capacitor directly mounted on the MAX6675 board at the
entry point of the terminal of the thermo couple exactly as the 31855
recommends. Working! I tested polarized 1uF and not polarized 223, 224 and
682. All gave the same POSITIVE result. Everything works.

Once again, we need a EE to weigh in.  But on this kind of stuff, a
tantalum capacitor is usually what is picked.   They have low leakage and
very high capacitance.  I think there are other good things they do, but
things like low inductance don't really matter here.

>Is there some place where to correct the temperature?

If you have to go this route, you can edit the thermal tables.  Take the
table entry you are currently using, and clone it.   Then edit it and give
it a new thermal couple number.  You will be able to change it to what ever
you need to get accurate temperature response.

zeeproduction, 2016-03-14:
@Roxy - I am aware of the fact thatthermal tables can be modified for the thermistors.
What about the thermocouple together with MAX6675? As far as I understand the curve of this is calculated through a function somewhere (maybe temperature.cpp ?!?).

Is this correct or where else could be done?

Thanks for your answer.

Roxy-3DPrintBoard, 2016-03-14:
> As far as I understand the curve of this is calculated through a function somewhere (maybe temperature.cpp ?!?).

Yes:

~~~~
#if ENABLED(HEATER_0_USES_MAX6675)
  #define MAX6675_HEAT_INTERVAL 250u
  static millis_t next_max6675_ms = 0;
  int max6675_temp = 2000;

  static int read_max6675() {

    millis_t ms = millis();

    if (ms < next_max6675_ms)
      return max6675_temp;
    
    next_max6675_ms = ms + MAX6675_HEAT_INTERVAL;
    max6675_temp = 0;
    #ifdef PRR
      PRR &= ~BIT(PRSPI);
    #elif defined(PRR0)
      PRR0 &= ~BIT(PRSPI);
    #endif
    SPCR = BIT(MSTR) | BIT(SPE) | BIT(SPR0);

    // enable TT_MAX6675
    WRITE(MAX6675_SS, 0);

    // ensure 100ns delay - a bit extra is fine
    asm("nop");//50ns on 20Mhz, 62.5ns on 16Mhz
    asm("nop");//50ns on 20Mhz, 62.5ns on 16Mhz
    // read MSB
    SPDR = 0;
    for (;(SPSR & BIT(SPIF)) == 0;);
    max6675_temp = SPDR;
    max6675_temp <<= 8;
    // read LSB
    SPDR = 0;
    for (;(SPSR & BIT(SPIF)) == 0;);
    max6675_temp |= SPDR;
    // disable TT_MAX6675
    WRITE(MAX6675_SS, 1);
    if (max6675_temp & 4) {
      // thermocouple open
      max6675_temp = 4000;
    }
    else {
      max6675_temp = max6675_temp >> 3;
    }
    return max6675_temp;
  }
#endif //HEATER_0_USES_MAX6675
~~~~

You could do define a function to shape the curve and at the end do:
~~~~
    return my_warming_over_of_the_results(max6675_temp);
~~~~

zeeproduction, 2016-03-14:
Wow!!! Thanks a lot. This will give me some long period to crunch it. :)
I have no idea about C++ but I'll dive as much as I can into.
(Yes I am aware of decompression issues...)
Thanks a lot everybody. I get back when I'll have something.

Grogyan, 2016-03-14:
@Roxy-3DPrintBoard that code has changed in RCBugfix

@zeeproduction a thermocouple has a linear curve which is managed by the reference cold juction, that is why you often hear thermocouple chips referred to as cold jucntion which maintains the linearity of the curve, the MAX6675 and MAX31855 both have internal high precision ADCs.

@Roxy-3DPrintBoard @zeeproduction the AD595/AD587 produce an analog signal out and therefore have no internal ADC, and do have an internal cold junction reference, therefore precision of the thermocouple reading is determined by the the ADC in the micro.  
In the case of the Atmel 2560 used on the Arduino Mega, which has 10bit ADCs. 
The MAX6675 has an internal 12bit ADC
The MAX31855 has an internal 14bit ADC

Roxy-3DPrintBoard, 2016-03-15:
@zeeproduction  I think what I'm hearing @Grogyan say is:   There is no reason to shape the curve the MAX6675 puts out.    The output of the chip is accurate.    

Grogyan, can you verify my interpretation of what you said?

zeeproduction, 2016-03-16:
Oh boy. Long story short, yes, the curve does not need to be modified. At least with my two thermocouples tested.
They are very accurate. And poor thermocouple was working flawless. It was my over confidence in my temp meter gun that made me forget the meter has a setup for the emissivity of material measured.
It measures in infra red so each material has it's own emissivity. While must of materials can use a set up of 0.95 on this meter, for the Aluminium it is 0.30 which is a huge difference. As soon as I set up this value, the thermocouple was hand in hand with the meter. Actually the meter came hand in hand with the thermocouple.

I still have to figure out how to modify one of the tables for the thermistor I have.

Even if is not necessary, if somebody decides to apply a correction factor for the MAX6675, I did my homework and can be placed in temperature.cpp like this

max6675_temp = max6675_temp * factor;
return max6675_temp;

Overunity factor gives smaller temperatures in linearly decreasing and under unity gives bigger temperatures display. But there is no need to do this.

To summarise this issue,
- the biggest error I had is that I haven't the SDSUPPORT enabled in Configuration.h and from there MINTEMP oR MAXTEMP error and software crash at every rotary encoder turn.
- another error I had it was three wires not connected but still sitting very close to each other for the extruder stepper.
- using a sure meter that was badly set up and it was the meter showing wrong temperatures and not the thermocouple readings.

Thanks a lot everybody for helping me with this one. Infinite thanks and thanks a lot for what you are doing.

P.S. As soon as I will have a stable system I will try to disable back the SDSUPPORT to see what happens and also I will post pictures.

Grogyan, 2016-03-16:
@Roxy-3DPrintBoard That is correct

@zeeproduction There should never be a scale factor or offset if using the MAX6675 and a K type thermocouple.
The MAX31855 comes in various varieties depending on the thermocouple used.  The Adafruit and others can be used with a K type,  Which are very common, as you can go to any electronics shop and buy one.

Keen to see your setup :-)

Roxy-3DPrintBoard, 2016-03-16:
@zeeproduction   If everything is OK and understood...  Closing the issue helps us by reducing the number of things we are watching.

zeeproduction, 2016-03-17:
If you think it can be closed it's ok from my side. The issue in the title is solved.
Thank you so much.

closed 2016-03-17 by Roxy-3DPrintBoard