MarlinFirmware/Marlin #3143: Dual extruder Filament_Runout_Sensor

sailorpete12, 2016-03-13:
Hi Everyone!

This is my first post on here and I'm not too familiar with how Github works. be nice :)

My issue - I have been working on a IR gate filament sensor which is Incorporated into the extruder housing (see link for my extruder http://www.thingiverse.com/thing:1064390) I have not updated the thing with the electronics yet. But 

-How would this work with a dual extruder printer?
-Is the sensor I/O only corresponds to the active tool or is dedicated to only T0? for instance both the filament hardware sensor share the same I/O and one runs out, this will trigger the active tool to initiate the M600 Command. Or is this only for Tool_0. I have not completed the HW build for this to test it.

Secondly, I am using Octroprint which works very well, but going through the code the Filament_runout_sensor is only activated when printing via an SD card. how would this be tweaked to activated when printing via the serial port? would a 'M' Command need to be incorporated to set marlin to monitor the sensor which processing the serial commands but also disable when not printing. say this can be defined in the starting and ending Gcode?  

Thanks for any help.

jbrazio, 2016-03-13:
@sailorpete12 I would suggest to close the issue here and open a new one at [MarlinDev](https://github.com/MarlinFirmware/MarlinDev).
PS: I've added your thing to my watch list, is looking good. :-)

sailorpete12, 2016-03-13:
@jbrazio Sounds like a better place for it. Thanks for the advice. =]

closed 2016-03-13 by sailorpete12