MarlinFirmware/Marlin #3145: Printing client

NervenCid, 2016-03-14:
Hello:

I'm new on github, and i making my thesis project on 3d printing, i want to understand how marlin comunicates, send a receive to the clients...

Where can i find documentation about this?

Thank you

jbrazio, 2016-03-14:
Marlin uses serial protocol and [g-code](http://reprap.org/wiki/G-code).

Roxy-3DPrintBoard, 2016-03-14:
> Where can i find documentation about this?

We don't have as good of documentation as we should (or want to) have.   But that issue aside....

The source code is the ultimate documentation.  It defines EXACTLY what the behavior is going to be.  Can you read C code?  If so,  you can dig in.

NervenCid, 2016-03-14:
Hello again:

Thank you for your answers, i have been reading about G-code, and i know that this code is what the arduino really receives from the client (i guess), i have been exploring some source files of the firmware but i couldn't find the file which is responsible of the serial comunication.

Thank you

Blue-Marlin, 2016-03-14:
loop() https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L736
get_command() https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L794
process_next_command() https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Marlin_main.cpp#L5662

thinkyhead, 2016-03-18:
The protocol between a host and Marlin (or any firmware) is pretty straightforward. The host sends GCode as plain text, each command a single line. Marlin replies "ok" as soon as it processes the command. Hosts may send extra information on each command line during a print session so that Marlin can do error checking and the host can re-send on error. See the bottom part of https://github.com/MarlinFirmware/Marlin/wiki/DNE-G-Code-in-Marlin for a description of the "N" and "*" parameters. The best place to find documentation on GCode is at http://reprap.org/gcode

jbrazio, 2016-04-05:
@NervenCid if you feel your question has been answered please close this ticket.

NervenCid, 2016-04-05:
Hello:

Thank you for your answers and sorry for my delays, i see that you don' t use delays in your code...

But for example in pseudocode if i send for example this:

serial.send("G1 X50 Y50;")

marlyn is capable for understand? or i need to send each character

and what is the format or data type in wich marlyn receives/send data?

Thanks and sorry for my english

jbrazio, 2016-04-05:
Marlin will understand because ";" is the start of a comment. A standard "move nozzle to XY" is just "G1 X10.5 Y5.3".

jbrazio, 2016-04-05:
I understand your question, the final char is usually a `\n` for each command.

NervenCid, 2016-04-07:
Thnak you for your answer...

I will try to work in that

I will close this issue, but: can i reopen it? or i must open a new one

jbrazio, 2016-04-07:
If the topic is the same I would to keep this one open.

closed 2016-05-04 by jbrazio