MarlinFirmware/Marlin #3154: Marlin RC-4 (RCBugFix) almost stable!  Going final soon!

Roxy-3DPrintBoard, 2016-03-15:
This post is just a head up for everybody using Marlin on their 3D-Printers.    Release Candidate #4 is getting very stable and the bug reports have dropped off to mostly being support questions.   Very soon, we expect to promote RCBugFix to be the new, official, stable release of Marlin.   If you know of any issues with RC-4, please let us know about them ASAP!

I've also alerted the users over at RepRap.org and 3DPrintBoard of the status:

http://forums.reprap.org/read.php?415,639920

http://3dprintboard.com/showthread.php?20851-Marlin-RC-4-(RCBugFix)-almost-stable!-Going-final-soon!&p=83325#post83325

PS.  A really big Thank You! to ThinkyHead for his help in taking point on this effort.

AnHardt, 2016-03-15:
To finalize fighting the "ok"/buffer problems we need ether #2984 or #3012. #3012 is going better and better now, but my tests are not finished yet (hopefully tonight).

nickbardom, 2016-03-15:
Thanks for the heads up. Any news about the Full Graphics LCD slowness fix?

Roxy-3DPrintBoard, 2016-03-15:
> To finalize fighting the "ok"/buffer problems we need ether #2984 or #3012. #3012 is going better and better now, but my tests are not finished yet (hopefully tonight).

We will wait until we hear from you on this topic.  Everybody wants this one exactly correct!

jbrazio, 2016-03-15:
@Anhardt made a good proposal for the LCD speed up but IMHO it's ready to go into this release.

paulusjacobus, 2016-03-15:
@Roxy-3DPrintBoard will try the new RC4 in the weekend on my Mendel and
Corexy printers. I'm curious!

AnHardt, 2016-03-16:
@jbrazio 
Sorry. My lcd_speedup branch is far from ready. For now more a test of concepts. It's good for testing but not for integration. The number_to_string conversions should not be in there (discovered a further conversion in the Serial code - and 4 `sprintf_P()` to print numbers - never benchmarked  `sprintf_P()` ). The "clock" should be displaied even when SDSUPPORT is not defined, ...
I'll be ready for RC-5.

Roxy-3DPrintBoard, 2016-03-16:
> I'll be ready for ~~RC-5~~ the version after RC4.

I'm starting to think there will be an RC-5 after all.   There are more outstanding Pull Requests (with important bug fixes) than I realized.

jbrazio, 2016-03-17:
Do we have a changelog for this RC version ? I'm not talking about `git log` more like a high level overview of it.

Roxy-3DPrintBoard, 2016-03-17:
No.   It's kind of like the documentation issue.   We need some help trying to dot all the i's and cross the t's.  We don't seem to be very good at documenting things.

thinkyhead, 2016-03-17:
Here's my priority to-do list for release. Basically, critical bug-fixes and obvious bug-fixes should go into `RC4`, items needing more testing, or experimental PRs can wait. So there are perhaps 4 PRs that I feel should definitely be merged and go into RC4, rather than making users wait for RC5 (_definitely_ the _real_ and _final_ RC!).

The obvious ones are:
- #3149 - suppress some compiler warnings
- #3127 `BABYSTEPPING` fixed for `COREXY` / `COREXZ`

These should go into `RC4`, but need more testing:
- #3012 Prevent queue corruption and don't send "ok" to serial for internal GCode. I've done a lot of testing and reworking to get it solid, as this is a vital fix.
- #3069 Delta-compatible `M48`. We need more Delta owners to test this feature.

This would be great to have in `RC4` but maybe it's too soon? I'm working on clarifying the documentation in `Configuration.h`:
- #3082 Check probes _only_ when deployed. Adds a `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` setting. In addition to testing, this would also benefit from simplified endstop and probe settings. #3065 is the concept I'm proposing for discussion.

I'll do as much testing and integration as I can tonight. Over the next day or two we can take care of any remaining polish needed for RC4. It's not utter perfection, but I feel a lot better about it than `RC3`!

jbrazio, 2016-03-17:
If I could add to RC5 wish list it would be automatic mesh leveling as it seems to be a often requested feature. But maybe this can turn out to be a project on its own, I never tried to evaluate the required effort.

We see some questions about the thermal protection which should also be addressed, maybe try out releasing RC4 with the more relaxed default set proposed by @Blue-Marlin as see how it goes on the wild ?

thinkyhead, 2016-03-17:
### The Release Process
- Merge all new commits in RCBugFix onto RC.
- Tag RC and RCBugFix with the new version.
- Make a release, listing all the changes since RC3
- Go tell two people to tell two people…

thinkyhead, 2016-03-17:
> automatic mesh leveling … seems to be a often requested feature

I keep flowcharting things as they are, and this has led to some good insights. Also, collecting all the code in one place. I think instead of just focusing on automatic mesh leveling (in fact, I kind of want to take a shot at this soon) we should apply the manual leveling system to the other two leveling equations, following the same model as the current Manual Bed Leveling (that comes with Mesh Bed Leveling), which is very clean. The process of splitting these up conceptually should yield a better system for future expansion.

(We should also investigate how far this kind of work has already progressed in the MarlinDev branches.)

Roxy-3DPrintBoard, 2016-03-17:
I get confused easily.  But I think I'm hearing you say that Automatic Mesh Bed Leveling that is not even fully written yet is way too big, complicated and untested to sneak into a Release Candidate.  If that is what you are saying, I'm in agreement.

thinkyhead, 2016-03-19:
Change Log for Marlin 1.1.0-RC4 – [as compared to RC3](https://github.com/MarlinFirmware/Marlin/compare/1.1.0-RC3...RCBugFix)

CoreXY and CoreXZ:
- #3127: XY BABYSTEPPING for CoreXY and CoreXZ
- #2995: Add CORE support to st_get_position_mm (and elsewhere)

Hosts, GCode:
- #3109: Add "Host Keepalive" (enabled by default) to inform hosts when Marlin is busy
- #2983: Remove "ok" from the output of M280, M301, M304, M851
- #3052: Fix "M145" output in M503 report
- #3108: Improved handling of `M110` for hosts
- #2985, #2987, #3051, #3044, #3027, #3111: Many more fixes for GCode handling

LCD Controllers and SD cards:
- #3156: Fix CardReader::lsDive()
- #3157: Use variable in lcd_erase_line()
- #3151: Update the pt-br translation
- #3116: Fix value edit click bug
- #2989: Blink XYZ on display when positions are uncertain
- #2942: Better number format for editing Delta Jerk and Z Steps
- #2973: Update splash screen with current no-domain
- [f18d3f4](https://github.com/thinkyhead/Marlin/commit/f18d3f451a5468adf046a4bc9e8e978a7a2ca753): LCD-based movement better-constrained for Deltas (#1869)

Minor bugs:
- #2902: Fix Z min endstop stepper.cpp
- #3161: Fix initial extruder direction for second/third/fourth extruder if they run inverted to the first
- #3050, #3126: Fix stack overflow issue with many LCD movements
- #3132: Fix a bad "heating failed" error when starting to reheat a warm extruder
- #3092: Fix Printrboard with LCD (redo)
- #3077: Prevent a `kill()` from rebooting with `USE_WATCHDOG` enabled
- #3066: Fix disable_all_heaters() for MAX6675
- #2848: Fix Rigidbot LCD Panel Support
- #2821: Fix status LEDs update when reinitializing the display

Homing, Bed Leveling:
- #3164: Fix #3160 - `MESH_BED_LEVELING` broke correct `G1` movement in Z
- #3107: `MIN_Z_HEIGHT_FOR_HOMING` replaces `Z_RAISE_BEFORE_HOMING`
- #3107: Adjustments to Z Raise options for probes and homing
- #3102: Bugfix: G28 was lifting Z-axis multiple times
- #3013: Enhanced #3010 - formal Mesh/Manual dependency
- #3057: Fix `Z_RAISE_AFTER_PROBING` for non DELTA printers
- #3001: Fix Z-raise order with `Z_SAFE_HOMING`
- #2978: Apply Mesh Bed Leveling matrix when switching extruders
- #3089: Fix `raise_z_for_servo` for Z offsets below the nozzle
- #3124: Update the bed topography output in `G29`

New and Experimental Features:
- #3069: Very New support for DELTA with `M48` Probe Repeatability Test
- #3112: Formal support for the MAX31855 thermocouple
- #3114: Option to reverse the click-wheel direction in menus
- #3043: Example configuration for Kossel XL
- #2976: Move the experimental Configurator to its own repository

Configurations:
- #2988: Automatically assign the next open E connector for Y2/Z2
- #3131: Allow override of `PWM_MOTOR_CURRENT` in configuration
- #3105: Much cleanup of configuration descriptions and code comments
- #3053: Improve description for nozzle-to-probe offsets
- #2986: Enable USE_WATCHDOG by default
- #3031: Fix board type for Hephestos: RAMPS 1.4
- #2868: Fix AUTOTEMP documentation in configs
- #3018: K8200: sample config back in sync

And...
- Many cleanups to the code: #2974, #2992, #2994, #3005, #3024, #3067, #3090, #3101, #3119, #3121, #3123, #3140, #3146, #3147, #3149, #3150, #3153

Roxy-3DPrintBoard, 2016-03-19:
That is an impressive list of changes!   For RC-5, let's commit to only making changes that fix bugs.  We need to get things to stabilize.

paulusjacobus, 2016-03-19:
@thinkyhead RC3 seems to work extremely well for me. So I'm happy to move to RC-4 and start testing.

Roxy-3DPrintBoard, 2016-03-20:
> @thinkyhead RC3 seems to work extremely well for me. So I'm happy to move to RC-4 and start testing.

We need 20x of you.

paulusjacobus, 2016-03-20:
@Roxy-3DPrintBoard > We need 20x of you.>
I have 3 printers... 2 Prusa I2s and one CoreXY (I'm building it, 90% complete)
Does that mean I count for 3 testers...hummm probably not

AnHardt, 2016-03-20:
Todays RCBugFix(https://github.com/MarlinFirmware/Marlin/commit/32f75749e49409065504293185c1fb59396f3d5d) feels very well. It only needs #3186, #3187 to be good enough for a release.

For #3182 (Add support for Printrboard RevF) i see low potential to break something else.
#3113 (Stop print timer with M105/M109) looks ok, but i don't see all consequences. I'd not feel well with merging right now before release - or i need some more days to test.
#3082 (Check probes only when deployed) was part of a bigger endstop-master-plan i made 5 month ago. Right now i have no clue about the consequences when merged alone.
#3065 (Simplified endstop configuration) is also a candidate for the next round- like some other endstop stuff.
#3069 (Delta-compatible extensions to M48) needs confirmation to work, from at least one additional tester with a DELTA.

Roxy-3DPrintBoard, 2016-03-20:
> #3069 (Delta-compatible extensions to M48) needs confirmation to work, from at least one additional tester with a DELTA.

It actually needed changes to handle Cartesian Machines also.   We need a couple of Cartesian Machines to load it and give a Thumbs Up.

> #3082 (Check probes only when deployed) was part of a bigger endstop-master-plan i made 5 month ago. Right now i have no clue about the consequences when merged alone.

This one feels important to get included just because so many machines are now having probes that are triggered when they are stowed.   Can we get this one checked out to see if it can be included?   Ideally, we would want to verify that neither the triggered or not triggered state does not have any impact when the probe is stowed.

epatel, 2016-03-20:
@thinkyhead Note about the list. I see #2978: Apply Mesh Bed Leveling matrix when switching extruders. But when looking at #2978 its about ABL.

AnHardt, 2016-03-20:
@Roxy-3DPrintBoard 
The importance of #3082, or what ever number it had 5 month ago has not changed, for me, size then. If it had been important enough for you, or anyone else, it would have been tested and merged then. :-(

thinkyhead, 2016-03-21:
@epatel I will double-check the titles I gave to the PRs. I generated the original list using some command piping and regex in BASH and it may have munged some of the number-title correspondences.

thinkyhead, 2016-03-21:
#3082 was one of those I was on the fence about. The description it included convinced me it was well thought out, pretty well constrained, and relatively straightforward. I couldn't find any fault with the logic in the code itself. And there are a couple of issues it would seem to address even beyond its immediate target,

For me the question was whether it was better to get it into the hands of users or to leave the issue of unwanted probe readings unaddressed in RC4. My confidence level is about 80% that the idea will work like a charm. The 20% I hold in reserve is for spots where it may yet need to be applied. But we can take a closer methodical look at it and see how we feel in toto.

Anyway, I'm giving another day or two to see what shakes out of `RCBugFix`, and if we feel that nothing is too egregious, I will aim to release RC4 on March 23.

jbrazio, 2016-03-21:
@thinkyhead will this be the last RC or do you plan for another round ?

thinkyhead, 2016-03-21:
@jbrazio If RC4 can survive a week without getting flagged with any "serious" bugs, it might end up being the final release. But, if we end up fixing any "tricky" issues, we may want to do just one more RC# with a very short and constrained patch cycle, Either RC4 or the RC5 will definitely be the official release, and I would like to aim for April 1st, no fooling.

Roxy-3DPrintBoard, 2016-03-21:
> But let me ask @AnHardt … How do you feel about #3082 and adding the new Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and FIX_MOUNTED_PROBE settings? Do you also think it must be possible to simplify this mess of multiplying probe and endstop options? Should we hold off and think it through more before committing to the techniques from #3082?

I think there is a lot of room for debate on this.  All opinions are welcome and deserve to be considered.   But the way I'm coming at it, if a probe is not deployed, the values it returns should not be looked at or used.   That is kind of independent of how we make it easier for the user.   And really...  This seems like a good place for somebody to introduce a Probe Object.   We don't even need one object that handles all cases.   We just need to be able to select which probe object (fixed, kicked down with a servo, Z-Min EndStop, etc.) that is going to be used.

> @thinkyhead will this be the last RC or do you plan for another round ?

With this many changes...   My guess is the odds of us not seeing some bad behavior in RC-4 is low.

AnHardt, 2016-03-21:
> But let me ask @AnHardt … How do you feel about #3082 and adding the new Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN and FIX_MOUNTED_PROBE settings? Do you also think it must be possible to simplify this [...]

As far i can read my own messages above i said, 'part of a bigger plan' and 'not right now'.
However - it works for me like a charm. But i know what to configure for my needs.

Of course. That area needs simplification. Especially the old descriptions are more confusing than enlightening the matter. Some of the options can be avoided at all. The right grouping seems to be important

To get an impression of my plan:

The logic has to be reversed from
No ABL -> No probe
to
No probe -> No ABL

Do we have a probe?
If not we use one of the Z endstop-pins for normal homing and can probe only manually.
If yes, we have to define where it is connected.
Ether at a dedicated probe pin.
PROBE_USES_PROBE_PIN
// that implies we also have a Z endstop that will be used for homing
or
PROBE_USES_Z_MIN_ENDSTOP_PIN
// implies we use the probe also for homing, if homing to z-min, and use Z_SAFE_HOMING

Than we can define one kind if probe.

+ Manual
+ Fix mounted
+ Servo
+ Sled
+ Allen Key
+ Tilt (Not for homing - needs to know z for deploying - or luck)
+ User defined

Per probe type there could be predefined selections of probe actions - maybe in separate files.
They build a sequence. Each of them can be omitted.
//// deploy()
// store x,y
+ Sequence of moves to be save to deploy (MIN_HEIGHT_FOR_PROBE_HANDLING)
+ Test if already deployed. (triggered when stowed) - skip
+ Sequence of moves to actually deploy the probe (Allen Key, sled, tilt)
+ Set servo angle.
+ Set solenoid pin.
+ Manual deploy (wait for click).
+ Test if deployed now (triggered when stowed) - repeat - error
// return to x,y
// activate pin testing

//// probe()
+ go down until the probe triggers
+ touch touch
+ Manual probe (adjust height - wait for click)

//// refresh for 'E'()
+ complete stow
+ complete deploy
+ Set servo angle.
+ Set solenoid pin.

//// stow()
// deactivate pin testing
// store x,y
+ Sequence of moves to be save to deploy (MIN_HEIGHT_FOR_PROBE_HANDLING)
+ Test if already stowed. (triggered when stowed) - skip
+ Sequence of moves to actually stow the probe (Allen Key, sled, tilt)
+ Set servo angle.
+ Set solenoid pin.
+ Manual stow (wait for click).
+ Test if deployed now (triggered when stowed) - repeat - error    

// return to x,y

If the probe is an object or not, does not matter. We need 4 functions/methods (deploy, probe, refresh, stow). All ending at the same x/y they started. Than probe handling for the calling function is trivial.

Targets:
+ Simplification by generalization.
+ Printing without ENDSTOPS_ONLY_FOR_HOMING

Roxy-3DPrintBoard, 2016-03-21:
Thank You for taking the time to organize and type all of that out.   I am going to think about the ramifications of changing the logic:
~~~~
The logic has to be reversed from
No ABL -> No probe
to
No probe -> No ABL
Do we have a probe?
~~~~
I think you may be on to something.   Changing this logic might help simplify things.   However, (and I'm not trying to be argumentative!!!) I'm not sure that   No probe-> No ABL is valid any more.   I think the Mesh Bed Leveling can function very nicely without a probe.   The user would have to 'Dial In the Nozzle' at each point in the mesh and let the firmware record that value.  But, I'm not sure the logic is as rigid as "No probe-> No ABL".

AnHardt, 2016-03-21:
@Roxy-3DPrintBoard 
Manual probing could be seen as just an other kind of probe. Deployment is calling the menu to adjust the height. Probing is adjusting the height and waiting for a click. Stowing is closing the menu or empty, when done with the click. Refreshing is calling the menu.

Roxy-3DPrintBoard, 2016-03-21:
OK!  Good Point!   That helps keep things simple.

WheresWaldo, 2016-03-23:
I will be very happy to move some of my braver compatriots onto RC4 in the Robo3D community. We have had a lot of nice things said about RC3 and everyone who has tried it there especially love MESH leveling. 

I was also wondering when feature creep would stop so this can move forward to a gold build. It is good to see this progress.

Roxy-3DPrintBoard, 2016-03-23:
> I was also wondering when feature creep would stop so this can move forward to a gold build. It is good to see this progress.

Because @thinkyhead is doing most of the work, it is hard to be too critical.   But I do think everybody is very seriously trying to only do bug fixes right now.

WheresWaldo, 2016-03-23:
I wasn't trying to be judgmental just wondering out loud. I have been a very big supporter of RC3 on the Robo3D forums and would love to push another RC to the brave few who are using it there.

jbrazio, 2016-03-23:
RC4 is rocking hard.
:facepunch:

thinkyhead, 2016-03-24:
Guess what… We have lift-off!

:rocket: [Marlin 1.1.0-RC4](https://github.com/MarlinFirmware/Marlin/releases/tag/1.1.0-RC4)

Please tell two people to tell two people. And get ready for the bug reports to start pouring in. Regardless of the many issues yet to go, we are in a better place now than we were with RC3.

esenapaj, 2016-03-24:
I feel admiration for your achievement.
By the way, Readme.md is still RC3.

emartinez167, 2016-03-24:
Just posted it on RigidBot's group on Google+

paulusjacobus, 2016-03-25:
I have tested the RCBugFix release on a Printrbot Simple with a Rev F board. Works absolutely great but what an absolute torture to flash the board. I installed Teensyduino but couldn't that to work despite a working install with Arduino PDE 1.6.9
Used an old programmer tool called Flip 3.4.7 from Atmel (see http://www.atmel.com/tools/FLIP.aspx) and imported with that tool the generated marlin.ino.hex file from the Arduino PDE (users/yourname/appdata/local/temp/xxxbuild/marlin.ion.hex).

Note for people who want to get the firmware installed: 
you need to choose AT90USB1286, HEX format 286 and check all the checkboxes in the run column. On the Rev. F board you need to jumper the boot pins near the processor and press the reste button.

Roxy-3DPrintBoard, 2016-03-26:
@WheresWaldo
>I wasn't trying to be judgmental just wondering out loud. I have been a very big supporter of RC3 on the Robo3D forums and would love to push another RC to the brave few who are using it there.

That comment wasn't directed at you.  I was saying "I'm trying to keep it to be only bug fixes...  But I'm not in a position to complain."

thawkins, 2016-03-26:
There is a fairy nasty bug with MANUAL_BED_LEVELING if you turn it on without MESH leveling enabled, then it does not compile,  I will pull down the latest RC tonight and run it up and capture the errors.

thinkyhead, 2016-03-26:
@thawkins Sorry about that. Since `MANUAL_BED_LEVELING` depends on `MESH_BED_LEVELING` the configuration was altered to reflect it. I can add a check for this condition in SanityCheck.h and throw a more descriptive error. Meanwhile, that section of `Configuration.h` now looks like this:
```cpp
//===========================================================================
//============================ Mesh Bed Leveling ============================
//===========================================================================

//#define MESH_BED_LEVELING    // Enable mesh bed leveling.

#if ENABLED(MESH_BED_LEVELING)
  #define MESH_MIN_X 10
  #define MESH_MAX_X (X_MAX_POS - (MESH_MIN_X))
  #define MESH_MIN_Y 10
  #define MESH_MAX_Y (Y_MAX_POS - (MESH_MIN_Y))
  #define MESH_NUM_X_POINTS 3  // Don't use more than 7 points per axis, implementation limited.
  #define MESH_NUM_Y_POINTS 3
  #define MESH_HOME_SEARCH_Z 4  // Z after Home, bed somewhere below but above 0.0.

  //#define MANUAL_BED_LEVELING  // Add display menu option for bed leveling.

  #if ENABLED(MANUAL_BED_LEVELING)
    #define MBL_Z_STEP 0.025  // Step size while manually probing Z axis.
  #endif  // MANUAL_BED_LEVELING

#endif  // MESH_BED_LEVELING
```

thinkyhead, 2016-03-26:
> timing on the thermal protection needs to be backed of by default

Did you increase the `WATCH_TEMP_PERIOD`? And what ended up being a good number for that? Did you also play around with the `WATCH_TEMP_INCREASE`? Smaller values would be more permissive there.

Default is to require a 4 degree rise in 16 seconds. So we might try something like 2 degrees in 20 seconds… (Giving the machine a full 30 minutes of space to rise from 20°C to 200°C), or 3 degrees in 15 seconds, giving about 15 minutes to rise from 20-200...

Roxy-3DPrintBoard, 2016-03-26:
> Default is to require a 4 degree rise in 16 seconds. So we might try something like 2 degrees in 20 seconds… 

I've been at 2 degrees in 20 seconds for a long time.   Anything tighter trips the protection.

thawkins, 2016-03-26:
I will build out a new version tonight/tomorrow and report back.

I was hoping that manual leveling would do the following.

1. Probe for the bed at the center of the bed. With either a proximity or
manual switch on z axis.
2. Put enable hold on z axis so it won't move
3. Free motors on x, y so you can push the axis around.

You can then use a bit of paper as a gauge,  push the head until it is over
each adjustement point, put the paper under the nozzle, and adjust the bed
leveling screws until you feel a "pinch" on the paper as you slide it back
and forth. Move the carriage to each other adjuster point and adjust as
required.

You don't really need to drive the head through a sequence of positions, as
long as it can be pushed around without changing z POS.  It also means you
don't need to know the adjuster geometry in the software.

jbrazio, 2016-03-26:
Makes sense your suggestion, what we have now might be named "assisted manual leveling" as Marlin is assisting you to go trough points.

Roxy-3DPrintBoard, 2016-03-26:
> You don't really need to drive the head through a sequence of positions, as
long as it can be pushed around without changing z POS.  It also means you
don't need to know the adjuster geometry in the software.

I think it is very common to have a 'Problem Area' on your bed and/or Mesh.   Being able to adjust and edit the Mesh for these areas is important.   One thing I'm looking at is allowing the adjustment of any random point as opposed to the actual Mesh Points.   By doing a Bi-linear correction to the four points around the random position being corrected, it may be possible to push and pull the Mesh into the correct shape easier.

thawkins, 2016-03-26:
Sounds like something better solved with a bit of glass.

Roxy-3DPrintBoard, 2016-03-26:
> Sounds like something better solved with a bit of glass.

This might be true on a Cartesian printer.   But for a Delta, a perfectly flat piece of glass won't fix the issues.   And besides...  It's a fun piece of code to write!

thinkyhead, 2016-03-27:
> what an absolute torture to flash the board

@paulusjacobus Sounds like something we should add to the new documentation. We need a page for each board that we support, probably, just to explain the quirks. PB RevF sounds like one of the more quirky ones.

thinkyhead, 2016-03-27:
> I've been at 2 degrees in 20 seconds for a long time. Anything tighter trips the protection.

There probably needs to be a longer initial period, when the hot-end is more cold, then after that, the heating speed may tend to increase. Ideally it would be nice to automate this tuning, like `M303`, so the machine knows what its heating profile "should" look like, and can give some leeway, but not too much.

Anyway, I will relax it to 2 degrees in 20 seconds, which should be more than enough time even initially.

thawkins, 2016-03-27:
I have noticed that the PID as it approaches the set temp, can
significantly slow down heating and even reverse before picking back up
again. Would that not potentially trigger the protection.

thinkyhead, 2016-03-27:
A well-tuned PID shouldn't fall back too far. Anyway, as soon as the heating is within the "hysteresis" range, the `WATCH_TEMP` test turns off.

StephS, 2016-03-28:
@paulusjacobus The Printrboard does not use the PJRC bootloader, thus the necessity for Atmel FLIP to be installed to flash. The teensyduino libraries are required for the MCU to function since Arduino never added official support for the AT90USB1286. The Reprap wiki has full instructions on how to flash the Printrboard ( Linux and Mac support is limited at the moment).
The remaining issue for Printrboard support is a nuisance more than anything, it utilizing marlin pin mapping for everything except the SPI pins, which it reverts to Teensyduino pin mapping. Part of this issue is that the Teensyduino library uses non-standard pin mapping.

Roxy-3DPrintBoard, 2016-03-28:
@StephS   How much work would it be to get the PrintRboard cleaned up and working well with Marlin?

StephS, 2016-03-28:
You mean changes to the physical board? or to Marlin? Support is added and thinkyhead merged. The issues were minor, like pin configuration and support for the DAC. The real issue is that Arduino does not natively support the AT90USB1286. You would need to develop and push a new library to Arduino, and even then you wouldn't have native programming. The AT90USB uses native USB instead of FTDI or another MCU, so special software is required to program, or programming a new bootloader (experimental). Since there is a lot of risk in programming the bootloader, it's better to use Atmel FLIP.
The options are limited and complicated. Nobody wants to write and maintain a library from scratch, write a DFU loader from scratch, or risk bricking with a new bootloader.

thawkins, 2016-03-28:
My understanding is that the official boards from Printrbot use the DFU
bootloader, whereas the clones from places like geeetech use the CDC
bootloader.

Both bootloaders require a jumper to be installed and reset to initiate
bootloading.  While the use of FLIP is one way of flashing the boards, it
is not the most common way of doing it.

thawkins, 2016-03-28:
Can you not use PlatformIO,  which does support the chipset with the
Arduino framework?

paulusjacobus, 2016-03-29:
I had a look at PlatformIO which supports the Arduino (Mega) and Teensy
(AT90USB) hardware and libraries (i.e. U8Glib). So, It looks very promising
from a first browse on the website. I will download and install it and see
if it works (compiles) with Marlin

[12 more replies]

closed 2016-04-10 by thinkyhead