MarlinFirmware/Marlin #3155: Axis "hopping" after high-speed move

jrouvier, 2016-03-15:
At the end of a high-speed movement, after decelerating and coming to a stop, the axis may "hop" to the final position.  [Video of the movement, slowed down to 1/2 speed](https://www.youtube.com/watch?v=V5RgjnvMxuk).  [Oscilliscope capture of the step pin](http://i.imgur.com/6kzFiQG.png)

The above video and graph where captured when complied with a `DEFAULT_AXIS_STEPS_PER_UNIT` of `160` and `G1 Y200 F17000`, resulting in a requested steps/sec of 45,334.  Below `F16000`, the motion seems smooth.  As the steps/sec increases, the effect becomes worse.  The board is an atmega2560-based board (16MHz clock).  It would appear that beyond a certain number of steps/sec, the math of how long to stay at top speed is being under calculated.

Obviously, there are limits to how may steps/sec are possible given hardware constraints.  It wouldn't likely be possible to have a 250kHz movement given the current architecture.  Therefore, I propose that `DEFAULT_MAX_FEEDRATE` and `M204` be capped to `32,768 / DEFAULT_AXIS_STEPS_PER_UNIT`, to prevent a movement being requested that is beyond the range that is possible to accomplish smoothly.

Blue-Marlin, 2016-03-15:
On what version/branch are you on? Somehow this error sounds familiar.

jrouvier, 2016-03-15:
This happened on both 1.1.0-RC2 and a checkout of master as of last night.

CONSULitAS, 2016-03-15:
@jrouvier 
Are you sure you used "master"?

Cutting edge is RCBugFix (https://github.com/MarlinFirmware/Marlin/tree/RCBugFix)

You should test with this branch.

Blue-Marlin, 2016-03-15:
What i remembered was '1888. But we have not had a better solution than reducing the speed.
@Wurstnase came up with #1997, but that was never merged. Have not digged in enough, to say if that is relevant or not, or if that has fond its way on an other path.

Wurstnase, 2016-03-15:
That not a bug. An AVR can't manage much more than 40kHz. This is what you see. If you need more, take a look at 32bit-hardware.

Blue-Marlin, 2016-03-15:
Maybe we should indeed do a plausibility check. 
@Wurstnase 
Can you suggest a formula?
DEFAULT_MAX_FEEDRATE <= factor * F_CPU / DEFAULT_AXIS_STEPS_PER_UNIT ??

Blue-Marlin, 2016-03-15:
```
DEFAULT_MAX_FEEDRATE [mm/s] * DEFAULT_AXIS_STEPS_PER_UNIT [step/mm] * stepper_interrupt_duration [ticks/step] + overhead [ticks/s] <= F_CPU [ticks/s]

DEFAULT_MAX_FEEDRATE [mm/s] <= (F_CPU [ticks/s] - overhead [ticks/s]) / ( stepper_interrupt_duration [ticks/step] * DEFAULT_AXIS_STEPS_PER_UNIT [step/mm])
```

jrouvier, 2016-03-15:
@CONSULitAS 
Indeed, you are right, it wasn't master, it was very likely RC.  I just copied the git url that was presented to me by default.

jrouvier, 2016-03-16:
Same behavior on RCBugFix

thinkyhead, 2016-03-16:
If we can forewarn users about bad settings (such as a too-high max feedrate) that would be one approach. Or, if it can be capped automatically, that is another. I would rather we warn users so that they will be discouraged from flashing a firmware that can misbehave given an unrealistic feedrate.

Wurstnase, 2016-03-16:
simple warn if max_speed of an axis * step_rate of this axis is > 40.000. Maybe add a sqrt(2) factor for coreXY.

jrouvier, 2016-03-19:
A check in SanityCheck.h wouldn't prevent someone from increasing the speed to unreasonable rates via M203...

Wurstnase, 2016-03-19:
Yes, you can't prevent any stupids.
Maybe you could, but this will also increase the size incredible.

thinkyhead, 2016-03-21:
Well. Rather than totally prevent it, maybe we can just issue a warning message for rates above the recommended threshold. Allow users to experiment, but let them know they are in sketchy territory.

paulusjacobus, 2016-03-21:
@Thinkyhead That might be a good idea or we can expand the user
documentation. I guess we need to start making a list of topics to be
covered so contributors (including of me) can start writing the wiki.

closed 2016-05-04 by jbrazio