MarlinFirmware/Marlin #3160: MESH_BED_LEVELING broke correct G1 movement in Z

eberttbi, 2016-03-15:
If MESH_BED_LEVELING is active you can't move in a linear way with all axis (G1 command). F.i. lets say your current axis positions are `X10 Y10 Z10` and if you execute `G1 X190 Y190 Z30` then only X and Y axis will move in a linear way but not the Z axis. The movement looks like a `G1 X190 Y190` and a `G0 Z30` running simultaniously.

Roxy-3DPrintBoard, 2016-03-15:
Did you do a G28 prior to this?

eberttbi, 2016-03-15:
Of course I homed all axis with G28. I can reproduce this behaviour each and every time. With MESH_BED_LEVELING deactivated in Configuration.h the G1 linear motion works exactly the way it should. If it's activated then it won't.

Roxy-3DPrintBoard, 2016-03-15:
Which branch are you running?   If you are not running RCBugFix, It would be very helpful if you could verify the problem is in that branch.

epatel99, 2016-03-15:
@eberttbi Interesting, this might be happen because lines need to be split on mesh boundaries. I have to look into it. Though I don't think any printing is actually made with "slopes", yet.

epatel99, 2016-03-15:
@eberttbi Yeah, you are right :) when I wrote that I left out the Z part. It's in the `mesh_plan_buffer_line(...)` function 

https://github.com/MarlinFirmware/Marlin/blob/5a54204f723f7fd110b57d19f0de01cecd8d47e7/Marlin/Marlin_main.cpp#L6365

I think my reasoning was that printing is actually only carried out in layers (=same Z). It should be fairly easy to add (but will result in math that adds up to "0" and actually waste clock cycles).

Roxy-3DPrintBoard, 2016-03-15:
@epatel99  Will it take you very long to create a patch?   Or edit and post the lines of the source code that need to change?    I'm thinking we want to get this checked out for correct behavior and get the changes into RCBugFix as soon as possible.

And I guess these changes will have to make it into the code I'm doing for Delta Mesh Leveling.  I'm leveraging your stuff pretty heavily and most likely this same bug exists for me too.

eberttbi, 2016-03-15:
@epatel99 If you use the marlin firmware for CNC milling then a correct linear motion is required for all axis. The question then arises why to use mesh bed leveling in a CNC setup, but still linear motion should always be linear.

epatel99, 2016-03-15:
@Roxy-3DPrintBoard No, should be easy fix.

epatel, 2016-03-16:
Think this will add the wanted behaviour. Though, can't test this right now but am pretty confident this adds same handling of Z as E.

epatel, 2016-03-16:
@eberttbi Care to verify? :)

epatel, 2016-03-16:
@Roxy-3DPrintBoard Oh, should I have added it to RCBugFix branch? ok, lets do that too then...

Roxy-3DPrintBoard, 2016-03-16:
> @Roxy-3DPrintBoard Oh, should I have added it to RCBugFix branch? ok, lets do that too then...

Yes,  but it turns out the Marlin Team would have noticed it and done it very quickly!   But THANK YOU again!    It is going to take a little while before this gets crossed over to MarlinDev, and that is where I'm doing my work.  But I'll cross it over manually before I continue on....

@epatel   (no 99 an more???)  Do you have a Delta Printer?

eberttbi, 2016-03-16:
@epatel I've tested the changes you made. Excellent! Now linear motion is working for all axis as expected. Thanks.
I thought of an example where you 3d print with linear motion in all axis: spiral vase prints.

epatel, 2016-03-16:
@Roxy-3DPrintBoard "99" is my "job account", and this my private. No Delta printer, yet ;) might get one someday.

closed 2016-03-16 by thinkyhead