MarlinFirmware/Marlin #3162: GPL Violation by "Dagoma" a french company

Alex4J, 2016-03-15:
Hello,
Dagoma (http://www.dagoma.fr/) a french company is selling printer with a preflashed Marlon on their Melzi, they say it's a "customized Marlin".
But they give no source code of it and no way to install it by your own.

thinkyhead, 2016-03-16:
Their only obligation is to supply the source code _upon request_. Have you written to request it? If you have and they have refused, you actually have the right to write back and _demand_ the source code. If they still refuse, tell them that the Marlin project team would like to talk to them!

Roxy-3DPrintBoard, 2016-03-16:
> I don't know what it is, but it seems like all these violators are using Melzi boards. Maybe this is actually a Chinese company…?

I wish there was a way for the firmware to easily ask "What kind of board am I running on?"    That would make it real easy for us to 'Encourage' the non-compliant manufacturers to honor their obligations.

Alex4J, 2016-03-18:
Yes i sended an email but they looks like to just "ignore" it, they don't answer. I am not the first one to ask, some people on the french RepRap IRC also asked but they also ignored them.

This is not a Chinese company, it's a french company, they just use most cheap and crappy materials from china...

Roxy-3DPrintBoard, 2016-03-18:
Send a Registered Letter to their Legal Department.   My guess is they will respond.

jfpion, 2016-03-18:
no need to ask [https://github.com/dagomafr/dago_FW]()

Roxy-3DPrintBoard, 2016-03-18:
> no need to ask https://github.com/dagomafr/dago_FW

Thank You!

Alex4J, 2016-03-18:
Oh yes sorry it's not listed on their website and they ignored all my emails, really sorry

CONSULitAS, 2016-03-25:
@Alex4J 
If the Marlin posted on https://github.com/dagomafr/dago_FW solves your problem, please close this issue for housekeeping. Thanks.

But this Marlin is at least one year old. :disappointed: 

I would recommend to use their `Configuration.h` and `Configuration_adv.h` to manually build a configuration for the brand new `RC4` branch. A huge amount of bugs have been fixed here thanks to @thinkyhead !

BTW: I have seen the Dagoma printer on the Make Faire in Dortmund two weeks ago and thought about building one for and with my son, because it is really simple - specially if you have a printer to print the plastic parts. :smile:

thinkyhead, 2016-03-26:
Now I'm curious to find out more about this Dagoma.

closed 2016-03-26 by thinkyhead

CONSULitAS, 2016-03-27:
@thinkyhead
It is a rather simple design and the kit is about 300€/$:
http://www.dagoma.fr/en/produit/le-kit-discovery200/

Key to the simplicity is that the frame is printed and you only need a few standard metal parts you get at eBay or so, a E3D with Bowden and a Mega/Ramps. The kit contains an electronics that I would not choose, because you can't change the stepper drivers and seems not to be expandable like a Ramps.

But i would change the fan layout, because the fan cools the hotend and the print.

But if you download the parts it seems to be rally easy to build...

epatel, 2016-03-27:
@CONSULitAS The card looks like a Melzi http://reprap.org/wiki/Melzi (think its an original RepRap design)

thinkyhead, 2016-03-27:
Too many of these Melzi machines popping up – but I guess they are inexpensive.

thinkyhead, 2016-03-27:
Changed the title because Internet.