MarlinFirmware/Marlin #3165: Do we have the Sanity Checks set too tightly for Auto Bed Leveling?

Roxy-3DPrintBoard, 2016-03-16:
Please pull up:  http://3dprintboard.com/showthread.php?20871-Question-about-AUTO_BED_LEVELING_GRID_POINTS&p=83412#post83412

I'm wondering if we should relax this check a little bit:

~~~~
#if X_PROBE_OFFSET_FROM_EXTRUDER < 0 #if (-(X_PROBE_OFFSET_FROM_EXTRUDER * AUTO_BED_LEVELING_GRID_POINTS) >= (RIGHT_PROBE_BED_POSITION - LEFT_PROBE_BED_POSITION))
#error "The X axis probing range is not enough to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS"
#endif
~~~~

Blue-Marlin, 2016-03-16:
```
"/Configuration.h:457:8: error: #error "The X axis probing range is not enough to fit all the points defined in AUTO_BED_LEVELING_GRID_POINTS".
```
That's not code from a current Marlin.
I think the complete check has been dropped.

This is replaced by a runtime check. Search for MIN_PROBE_EDGE in Marlin_Main.cpp.
Even that restriction does not make much sense to me. (hidden limit to restrict AUTO_BED_LEVELING_GRID_POINTS ? Limit number of mesh points, to not subdivide moves too often?)

Roxy-3DPrintBoard, 2016-03-16:
> That's not code from a current Marlin.   I think the complete check has been dropped.

Oooops!  I guess I should have checked before I made a 'New Issue'.    Sorry about that!

closed 2016-03-16 by Roxy-3DPrintBoard