MarlinFirmware/Marlin #3166: Improved Documentation

Roxy-3DPrintBoard, 2016-03-16:
@jbrazio  I think GitHub might have deleted your previous thread because of some questionable content in it.  (Certainly, not by you!!!)    I think what you are trying to accomplish is valuable and I would like to restart your thread.   I took the liberty of re-posting the message that started your thread:

Continuing the discussion which went a bit off-topic at #3063.

Wikis are basically garbage, the effort required to make one look good and easy to navigate is not productive because in the end of the day users will not update them, they are over complicated. After been told by @thinkyhead what happened in the past and due to the good experience we have in overall with Github I decided to build a mock up based on Jekyll and Github Pages.

    All documentation is stored on Github
    The repository is property of the MarlinFirmware Github organization
    Everyone can fork and contribute in a standard manner like they already do to Marlin and MarlinDev
    We can use our own domains which means the previous idea can have continuity
    Documentation can be build either in html or in markdown

The mock up I've build can be found at jbrazio/MarlinDocumentation and it can be live previewed here.

I'm demoing the following concepts:
1. Intro page with key Marlin features
2. Quick start guides - shall be guide a newb user from start to finish to have Marlin up and running
3. Documentation - Where the current "wiki" will go

Only three links works currently:

    "Learn more" which goes to a intro page
    "Rich G-Code Support > View details" which goes into the G-Code listing
    "Documentation" which generates a list of pages under the "docs" category

Please have a look and let's take this as a starting point to build something great.

jbrazio, 2016-03-16:
@Roxy-3DPrintBoard #3088 is back ! Github staff unlocked my account.

Roxy-3DPrintBoard, 2016-03-16:
OK...  I'll close this thread!

closed 2016-03-16 by Roxy-3DPrintBoard

thinkyhead, 2016-03-16:
I wrote Github to ask about #3088 but it was apparently fixed by the time they read my report.

Meanwhile, please ignore @ t a m a r i n t e c h. Don't even bother to read what he writes. We are trying to get Github to take some action, but if we cannot block him or get him off of Github, we will simply delete his posts as soon as they appear.

jbrazio, 2016-03-17:
Not far from today some one send me this advice:
http://opensoul.org/99ways/#113

:-)