MarlinFirmware/Marlin #3169: lcd_goto_menu not working ?

CoderSquirrel, 2016-03-17:


Roxy-3DPrintBoard, 2016-03-17:
Is the reason because the lcd_goto_menu(prevMenu,) never happens?   It can't happen until enough time has gone by that the the next_button_update_ms is bigger than the current time.    In other words, you might have a race condition.   Because next_button_update_ms keeps getting reset to a bigger value by the interrupt routines that refresh the LCD display.

CoderSquirrel, 2016-03-17:
So, if i change
```
if (millis() > next_button_update_ms && READ(JXE) > 0 ) {
 lcd_goto_menu(prevMenu, prevEncoderPosition);
  }
```

to 
```
if ( READ(JXE) > 0 ) {
 lcd_goto_menu(prevMenu, prevEncoderPosition);
  }
```
it may work?

Roxy-3DPrintBoard, 2016-03-17:
I don't know what your code is trying to accomplish or the logic you have going.  But the second version doesn't care how much time has happened.  It is just checking to see if READ(JXE) has a value greater than 0 and if so will do the lcd_goto_menu().

CoderSquirrel, 2016-03-17:
Yeah, but that's the thing, the lcd_goto_menu(prevMenu) is doing nothing

jbrazio, 2016-03-17:
You're exactly using this block of code where in Marlin ?

CoderSquirrel, 2016-03-17:
@jbrazio at ultralcd.cpp in lcd_buttons_update

Roxy-3DPrintBoard, 2016-03-17:
One of the problems is you are operating at interrupt time.  So even though AVR's have limited debug capability, you are getting even less of it than you would normally have.  In this kind of situation, if I wasn't at interrupt level...  I would do something like:

```
if ( READ(JXE) > 0 ) {
 SERIAL_ECHO("My Push Button Got Pressed!!!\n");
 lcd_goto_menu(prevMenu, prevEncoderPosition);
  }
```

With that said...  The transmission of serial characters is currently polled.  So you probably can do this at the expense of losing interrupts during the transmission.  In other words, if you have the stepper motors moving, you most likely will lose position.  If you have the heaters on, they may lose temperature control.   But you probably can figure out if your switch is really getting pressed.

jbrazio, 2016-03-17:
Test if you getting inside the condition by doing something like:

```
if (millis() > next_button_update_ms && READ(JXE) > 0 ) {
  SERIAL_ECHOLN("we are in");
  lcd_goto_menu(prevMenu, prevEncoderPosition);
}
```

or..

```
if ( READ(JXE) > 0 ) {
  SERIAL_ECHOLN("we are in");
  lcd_goto_menu(prevMenu, prevEncoderPosition);
}
```

You should have a serial terminal connected to the printer.. it can even be Arduino IDE serial monitor.

Roxy-3DPrintBoard, 2016-03-17:
Snicker Snicker!!!   I type faster than you!!!    :)

jbrazio, 2016-03-17:
lol

CoderSquirrel, 2016-03-17:
RACE 2
GO!
ahahah sorry
Yeah guys, i tested and its pressed. :S the thing that is pissing me off is that i used it at the same way at the old 1.0 and worked and now in 1.1 its not

thinkyhead, 2016-03-18:
@CoderSquirrel You should be using `lcd_save_previous_menu()` and `lcd_goto_previous_menu()` in the latest code. Also, the `lcd_goto_menu()` takes 3 arguments, not 2:
```cpp
lcd_goto_menu(menuFunc_t menu, const bool feedback = false, const uint32_t encoder = 0);
```

CoderSquirrel, 2016-03-19:
@thinkyhead thanks for the answear.
As i said, i used it only with  lcd_goto_menu(prevMenu) at the 1.0 and it worked, maybe lucky of mine or it was this way in the 1.0?
And another thing i must have lcd_save_previous_menu() in each menu?
Because the reason i wanted to use prevMenu to go to the last menu is because it is save in the menu macros, so i wouldn't have to implement the thing of saving the previous menu.
i don't have the things to test with me this weekend but monday morning i will test cd_goto_previous_menu() thanks.

PS .i post a lot here and i think someday you'll probably will get piss of about me (Please don't be)

Roxy-3DPrintBoard, 2016-03-20:
> PS .i post a lot here and i think someday you'll probably will get piss of about me (Please don't be)

If you look back at all of the responses you got from the beginning when you started asking questions...   If you are willing to spend the time to learn and try things...  The hard core developers here will answer questions.   Everybody with code checked in here knows what you are going through.

CoderSquirrel, 2016-03-21:
@thinkyhead by using lcd_save_previous_menu() and lcd_goto_previous_menu() you mean in 1.0?

thinkyhead, 2016-03-21:
@CoderSquirrel Are you able to do a search on all the code at once from your editor? If so, you can answer your own question about what functions exist in 1.0 code (which we don't support) versus current code. I think I was pretty clear that the functions specifically concerning "previous menu" are new. I think I was also clear about their usage. Just look at the places where they are used in the code, and you will see it is only in specific cases - for edit items that are "leaves" in the menu tree.

closed 2016-03-21 by CoderSquirrel