MarlinFirmware/Marlin #3171: Current rcbugfix strange Z_DUAL_STEPPER_DRIVERS behaviour from panel REPRAP_DISCOUNT_SMART_CONTROLLER

samwyze, 2016-03-18:
Just trialing the latest bugfix as at 18/3/16 and have some odd behaviour with my dual z stepper driver setup.

If I try and use the panel to move Z-axis at 1mm increments only one motor is turned. 
Same thing happens if I try send a g28 from the console, or use autohome from the lcd.
If I use it to move z-axis at 0.1mm increments all is normal, as is if I send movement commands via a console or pronterface (just not a home command).
Also, I've noticed X axis struggles to engage on any homing command, but moving either via lcd or console it is fine.
It looks very similar to an under powered stepper driver, however if I revert to my previous rcbugfix release (a least a month old) all movement returns to normal.

All relevant settings in my configs appear to be the same. 
Am I missing something obvious?

Keep up the awesome work, thanks for making Marlin great!

Regards,
Sam

thinkyhead, 2016-03-18:
I will have a look at the relevant code and see what's been merged in the last month or so that could be affecting this. I know there have been some changes in this area lately, so you might have caught a typo or something more off. I'll post again when I know more.

samwyze, 2016-03-20:
Thank you, the previous release I was using was the current RCBugFix as at 4/3/16 if that helps.

thinkyhead, 2016-03-21:
What are your `HOMING_FEEDRATE` and `MANUAL_FEEDRATE` settings, for reference?

thinkyhead, 2016-03-21:
After a detailed comparison between March 4 and March 16 (about 4 days ago) the only interesting change I can see that might be in any way affecting movement at certain scales is this:
In `planner.cpp` at line 856 I changed this from `long` to `unsigned long`:
```cpp
  unsigned long bsx = block->steps[X_AXIS], bsy = block->steps[Y_AXIS], bsz = block->steps[Z_AXIS], bse = block->steps[E_AXIS];
```
…and I added code to limit acceleration differently. The old code was (line 866…):
```cpp
  // Limit acceleration per axis
  unsigned long acc_st = block->acceleration_st,
                xsteps = axis_steps_per_sqr_second[X_AXIS],
                ysteps = axis_steps_per_sqr_second[Y_AXIS],
                zsteps = axis_steps_per_sqr_second[Z_AXIS],
                esteps = axis_steps_per_sqr_second[E_AXIS];
  if ((float)acc_st * bsx / block->step_event_count > xsteps) acc_st = xsteps;
  if ((float)acc_st * bsy / block->step_event_count > ysteps) acc_st = ysteps;
  if ((float)acc_st * bsz / block->step_event_count > zsteps) acc_st = zsteps;
  if ((float)acc_st * bse / block->step_event_count > esteps) acc_st = esteps;
```
…and the new code does it this way:
```cpp
  // Limit acceleration per axis
  unsigned long acc_st = block->acceleration_st,
                xsteps = axis_steps_per_sqr_second[X_AXIS],
                ysteps = axis_steps_per_sqr_second[Y_AXIS],
                zsteps = axis_steps_per_sqr_second[Z_AXIS],
                esteps = axis_steps_per_sqr_second[E_AXIS],
                allsteps = block->step_event_count;
  if (xsteps < (acc_st * bsx) / allsteps) acc_st = (xsteps * allsteps) / bsx;
  if (ysteps < (acc_st * bsy) / allsteps) acc_st = (ysteps * allsteps) / bsy;
  if (zsteps < (acc_st * bsz) / allsteps) acc_st = (zsteps * allsteps) / bsz;
  if (esteps < (acc_st * bse) / allsteps) acc_st = (esteps * allsteps) / bse;
```
So you might try reverting this part of the code and seeing if it has any effect.

If we need to narrow it down more, perhaps you can try going back in time and figure out the exact day when the behavior went awry. For example:
```
git checkout 'RCBugFix@{2016-03-11 00:00:00}' -b rcbugfix_march_11_2016
```

Roxy-3DPrintBoard, 2016-03-21:
> …and I added code to limit acceleration differently. The old code was (line 866…):

Only Bug Fixes!!!!

thinkyhead, 2016-03-21:
Quite. The acceleration fix comes from the Ultimaker fork, as I recall.

samwyze, 2016-03-21:
Will try out reverts (and note requested feedrates) and post an update tomorrow, printer is busy melting plastic tonight...

samwyze, 2016-03-22:
Just a quick update, all feedrates and jerk settings are at default. Haven't had a chance to upload yet as my lad has been busy printing toys...hopefully this evening?

thinkyhead, 2016-03-22:
I don't advise the default rates, necessarily. In my own case, I actually lowered Z from defaults even though I raised XY:
```cpp
#define DEFAULT_MAX_FEEDRATE { 500, 500, 2.25, 45 }
```

samwyze, 2016-03-22:
Thanks Scott, lowering the default max feedrate on the z axis fixed all issues on Z. Why did this not affect homing etc... in RC3 (or was more forgiving at least)? The x axis issue was mechanical, I hadn't noticed that x was no longer level after the z issues.
I'll finish optimizing printer speeds now, I'm guilty of not tuning these, as soon as the machine was printing my kids queued a bunch of prints and I never got around to any fine tuning...**my bad**!

closed 2016-03-22 by samwyze