MarlinFirmware/Marlin #3174: auto bed level grid activate and deactivate the probe beween measurments

hexxter, 2016-03-18:
what is the best way to integrate (PIN, Servo, ...) to activate an Solenoid befor and after each measurment on auto bed level grid.

http://community.robo3d.com/index.php?threads/mike-kellys-solenoid-auto-level-for-stock-r1.3864/

this is what i want to integrated.

Thx

Roxy-3DPrintBoard, 2016-03-18:
You would just not use the servo code (or activate an servos).   And you would power up the pin that controls the solenoid when the function run_z_probe() is called.

hexxter, 2016-03-18:
yes the solenoid can only used for a short time <5 sev i will activate a pin short before a z_probe will startet and disable the pin after the probe.

G29 E but with a pin and not a servo? Is this correct or i'm wrong?

Roxy-3DPrintBoard, 2016-03-18:
Yes.  You would have to modify the deploy_z_probe() and retract_z_probe() function.  You would not declare any servo's and these functions will key off of HAS_SERVO_ENDSTOP to control the generation of code for servo's.   G29 would have no idea about what you are doing to get the probe ready for use.   The suggested changes are at a lower level and should be transparent to G29.

jbrazio, 2016-03-18:
@Blue-Marlin, @thinkyhead can somehow that unused solenoid pin/code be hacked for this ?

jbrazio, 2016-03-18:
nvm I didn't read clearly the request.

closed 2016-05-04 by jbrazio