MarlinFirmware/Marlin #3178: Controller board fan stays on forever

ads61, 2016-03-19:
After the first connect to the board and homing the axis the fan switches off after the time set in the configuration.
After moving one of the axes the fan goes on and stay on forever.
(In both RC3 and RCbugfix of today)

thinkyhead, 2016-03-21:
> After the first connect to the board and homing the axis the fan switches off after the time set in the configuration.

Which fan? The controller fan or the variable-speed fan on the extruder?

> After moving one of the axes the fan goes on and stay on forever.

Just moving an axis with an ordinary `G1` command will do this?

thinkyhead, 2016-03-21:
The controller fan should stay on for as long as any stepper motor is active, so it's possible this is a side-effect of one of your stepper motors not being turned off. Do you have your stepper motors configured to disable after a period of inactivity? What is your setting for `DEFAULT_STEPPER_DEACTIVE_TIME`? There are new settings that are "true" by default, but if they aren't in your configuration it may also be causing your steppers never to disable.
```cpp
#define DISABLE_INACTIVE_X true
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z true  // set to false if the nozzle will fall down on your printed part when print has finished.
#define DISABLE_INACTIVE_E true
```

ads61, 2016-03-21:
DISABLE_INACTIVE_X and DISABLE_INACTIVE_Y are false so in that sense the behaviour is correct and fan goes off after the printjob if motors are disabled.

I was confused because in version 1 the fan stopped always after the last move and the DEFAULT_STEPPER_DEACTIVE_TIME expired.

For now switched back to v1.0 because i need the printer and test later again.

thinkyhead, 2016-03-26:
We sorted out a lot of stuff for RC4, and should have an RC5 in a few days. Meanwhile, RCBugFix is the safest bet to have the most patched version until the final release. A public release has a way of exposing bugs that we miss… but so far they have not been _too_ hard to fix.

closed 2016-04-28 by jbrazio